tum-vision/lsd_slam #195: Failure compiling with rosmake

YonatanSimson, 2016-05-04:
I am using Ubuntu 14.04 with ROS Indigo. I tried to used rosmake both on ubuntu guest in a virtual box and on AWS. In both cases

The command:
rosmake lsd_slam

Produces the following error:

[ rosmake ] rosmake starting...
[ rosmake ] Packages requested are: ['lsd_slam']
[ rosmake ] Logging to directory /home/ubuntu/.ros/rosmake/rosmake_output-20160504-182012
[ rosmake ] Expanded args ['lsd_slam'] to:
[]
[ rosmake ] WARNING: The following args could not be parsed as stacks or packages: ['lsd_slam']
[ rosmake ] ERROR: No arguments could be parsed into valid package or stack names.

I followed the installation instruction to the letter. 

Any ideas?

gogobd, 2016-05-04:
Are you sure you executed (sourced) the shell script that sets the environment? Gg

☆

YonatanSimson, 2016-05-05:
Yes

Are you referring to:

echo "source ~/rosbuild_ws/setup.bash" >> ~/.bashrc
bash

or when installing ros there is of course:

source /opt/ros/jade/setup.bash

antonays, 2017-02-22:
Hi, 
I suggest to try and install using catkin instead of rosmake.

This git is not set up correctly for ROS indigo (missing dependencies), so i suggest you to clone the git i forked and applied some fixes to work on indigo.

ndinarvand, 2018-09-19:
Hi all,
I ran "source ~/Code/rgbdslam_catkin_ws/devel/setup.bash" in terminal and issue solved.

hipforth, 2019-11-23:
> Hi all,
> I ran "source ~/Code/rgbdslam_catkin_ws/devel/setup.bash" in terminal and issue solved.

That's work, thanks!