tum-vision/lsd_slam #197: lsd_slam on mobile phone

johnny871227, 2016-05-12:
On [https://www.youtube.com/watch?v=GnuQzP3gty4](url), lsd_slam is running on a mobile phone (from 22sec). Now I'm trying to use Samsung Galaxy S6 as a webcam (lsd_slam is still on a PC).  I use the DroidCam app on Ubuntu 14.04, and its works fine for capturing video on cheese and guvcview viewer. 
However I cannot use it as a livecam for lsd_slam. When running `rosrun uvc_camera uvc_camera_node device:=/dev/video0` I got errors shown below:

```
NODES
  /
    uvc_camera (uvc_camera/uvc_camera_node)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[uvc_camera-1]: started with pid [91926]
[ INFO] [1463019260.426784438]: camera calibration URL: file:///home/xudi/rosbuild/package_dir/camera_umd/uvc_camera/example.yaml
opening /dev/video0
pixfmt 0 = 'YU12' desc = 'Planar YUV 4:2:0'
  discrete: 640x480:   1/30 
terminate called after throwing an instance of 'std::runtime_error'
  what():  couldn't set format
[uvc_camera-1] process has died [pid 91926, exit code -6, cmd /opt/ros/indigo/lib/uvc_camera/uvc_camera_node __name:=uvc_camera __log:=/home/xudi/.ros/log/2bb3ab2c-17e0-11e6-ab24-f4f26d06cd4b/uvc_camera-1.log].
log file: /home/xudi/.ros/log/2bb3ab2c-17e0-11e6-ab24-f4f26d06cd4b/uvc_camera-1*.log
all processes on machine have died, roslaunch will exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Anyone has a clue? I guess it's kinda related to #46 , but here I'm using a mobile phone to replace the webcam.

johnny871227, 2016-05-12:
Btw, this may be a general problem for all webcam.

johnny871227, 2016-05-13:
I just commented the line `throw std::runtime_error("couldn't set format");` and  problem solved.

closed 2016-05-13 by johnny871227

caap95, 2017-11-18:
@johnny871227 , I can't find that file, uvc_cam.cpp. Can you tell me where it is