tum-vision/lsd_slam #201: no binaries resulting from rosmake lsd_slam

gunshi, 2016-05-27:
OS X El Capitan (10.11.12)
Opencv 3.1 built from source
ROS Indigo installed beforehand

output of rosmake led_slam:

> [ rosmake ] rosmake starting...                                                 
> [ rosmake ] Packages requested are: ['lsd_slam']                                
> [ rosmake ] Logging to directory /Users/gunshigupta/.ros/rosmake/rosmake_output-20160527-213907
> [ rosmake ] Expanded args ['lsd_slam'] to:
> ['lsd_slam_core', 'lsd_slam_viewer'] 
> [rosmake-0] Starting >>> catkin [ make ]                                        
> [rosmake-0] Finished <<< catkin ROS_NOBUILD in package catkin
>  No Makefile in package catkin
> [rosmake-0] Starting >>> genmsg [ make ]                                        
> [rosmake-2] Starting >>> cpp_common [ make ]                                    
> [rosmake-3] Starting >>> cmake_modules [ make ]                                 
> [rosmake-1] Starting >>> rosgraph [ make ]                                      
> [rosmake-3] Finished <<< cmake_modules ROS_NOBUILD in package make_modules
>  No Makefile in package make_modules
> [rosmake-0] Finished <<< genmsg ROS_NOBUILD in package genmsg
>  No Makefile in package genmsg
> [rosmake-3] Starting >>> rospack [ make ]                                       
> [rosmake-0] Starting >>> genlisp [ make ]                                       
> [rosmake-1] Finished <<< rosgraph ROS_NOBUILD in package rosgraph
>  No Makefile in package rosgraph
> [rosmake-1] Starting >>> genpy [ make ]                                         
> [rosmake-0] Finished <<< genlisp ROS_NOBUILD in package genlisp
>  No Makefile in package genlisp
> [rosmake-0] Starting >>> gencpp [ make ]                                        
> [rosmake-3] Finished <<< rospack ROS_NOBUILD in package rospack
>  No Makefile in package rospack
> [rosmake-3] Starting >>> roslib [ make ]                                        
> [rosmake-0] Finished <<< gencpp ROS_NOBUILD in package gencpp
>  No Makefile in package gencpp
> [rosmake-0] Starting >>> rosparam [ make ]                                      
> [rosmake-1] Finished <<< genpy ROS_NOBUILD in package genie
>  No Makefile in package genie
> [rosmake-3] Finished <<< roslib ROS_NOBUILD in package rosily
>  No Makefile in package rosily
> [rosmake-1] Starting >>> message_generation [ make ]                            
> [rosmake-3] Starting >>> rosunit [ make ]                                       
> [rosmake-0] Finished <<< rosparam ROS_NOBUILD in package rosparam
>  No Makefile in package rosparam
> [rosmake-0] Starting >>> rosmaster [ make ]                                     
> [rosmake-3] Finished <<< rosunit ROS_NOBUILD in package rosunit
>  No Makefile in package rosunit
> [rosmake-3] Starting >>> roslang [ make ]                                       
> [rosmake-0] Finished <<< rosmaster ROS_NOBUILD in package rosmaster
>  No Makefile in package rosmaster
> [rosmake-0] Starting >>> rosclean [ make ]                                      
> [rosmake-3] Finished <<< roslang ROS_NOBUILD in package roslang
>  No Makefile in package roslang
> [rosmake-3] Starting >>> roslz4 [ make ]                                        
> [rosmake-0] Finished <<< rosclean ROS_NOBUILD in package rosclean
>  No Makefile in package rosclean
> [rosmake-3] Finished <<< roslz4 ROS_NOBUILD in package roslz4
>  No Makefile in package roslz4
> [rosmake-1] Finished <<< message_generation ROS_NOBUILD in package message_generation
>  No Makefile in package message_generation
> [rosmake-2] Finished <<< cpp_common ROS_NOBUILD in package cpp_common
>  No Makefile in package cpp_common
> [rosmake-2] Starting >>> rostime [ make ]                                       
> [rosmake-1] Starting >>> xmlrpcpp [ make ]                                      
> [rosmake-2] Finished <<< rostime ROS_NOBUILD in package restive
>  No Makefile in package restive
> [rosmake-2] Starting >>> roscpp_traits [ make ]                                 
> [rosmake-2] Finished <<< roscpp_traits ROS_NOBUILD in package roscpp_traits
>  No Makefile in package roscpp_traits
> [rosmake-2] Starting >>> roscpp_serialization [ make ]                          
[... 64 lines cut ...]
> [rosmake-3] Finished <<< lsd_slam_core [SKIP] No rule to make target None       
> [ rosmake ] Results:                                                            
> [ rosmake ] Built 42 packages with 0 failures.                                  
> [ rosmake ] Summary output to directory                                         
> [ rosmake ] /Users/gunshigupta/.ros/rosmake/rosmake_output-20160527-213907      
> 

However there are no compiled binaries in any of the folders under lsd_slam.
so when I run:

> rosrun lsd_slam_viewer viewer

I get:
[rosrun] Couldn't find executable named viewer below /Users/gunshigupta/Desktop/4sfm/lsd_slam/lsd_slam_viewer

Also, buildfailures.txt and buildfailures-with-context.txt are both blank.
buildoutput.log has the following message:
dyld: Library not loaded: librospack.dylib
  Referenced from: /Users/gunshigupta/ros_catkin_ws/install_isolated/bin/rospack
  Reason: image not found
Makefile:1: /cmake.mk: No such file or directory
make: *** No rule to make target `/cmake.mk'.  Stop.

Can anyone help me with this?
Thanks.

TungTNguyen, 2016-10-15:
I got into the same issue. Can anyone help us or shed a light

DL7AD, 2016-11-14:
Got the same issue too. It says there are 2 packages build but there are none available thereafter.