tum-vision/lsd_slam #202: building lsd_slam_viewer

blankered, 2016-06-09:
Hi,

I am trying to build lsd_slam from https://github.com/mhkabir/lsd_slam. I have successfully built this package on two ubuntu machines but I having trouble building it on a Raspberry Pi 3. 

This is the output:
```
**Finished <== cv_bridge       [ 3.3 seconds ]                                                                                                       
Starting ==> lsd_slam_viewer                                                                                                                       
                                                                                                                                                   
[lsd_slam_viewer] ==> '/home/pi/lsd_ws/build/lsd_slam_viewer/build_env.sh /usr/bin/make cmake_check_build_system' in '/home/pi/lsd_ws/build/lsd_slam_viewer'
[lsd_slam_viewer] <== '/home/pi/lsd_ws/build/lsd_slam_viewer/build_env.sh /usr/bin/make cmake_check_build_system' finished with return code '0'
                                                                                                                                                    
*** Error in `/usr/bin/c++': double free or corruption (top): 0x006f2bb8 ***                               [4/4 Jobs | 2/4 Active | 2/5 Completed] 
                                                                                                                                                   
[lsd_slam_viewer] ==> '/home/pi/lsd_ws/build/lsd_slam_viewer/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' in '/home/pi/lsd_ws/build/lsd_slam_viewer'
[  7%] Built target lsd_slam_viewer_gencfg
[  7%] Built target _lsd_slam_viewer_generate_messages_check_deps_keyframeGraphMsg
[  7%] Built target _lsd_slam_viewer_generate_messages_check_deps_keyframeMsg
[ 15%] Building CXX object CMakeFiles/viewer.dir/src/main_viewer.cpp.o
CMakeFiles/viewer.dir/build.make:54: recipe for target 'CMakeFiles/viewer.dir/src/main_viewer.cpp.o' failed
make[2]: *** [CMakeFiles/viewer.dir/src/main_viewer.cpp.o] Aborted
CMakeFiles/Makefile2:1220: recipe for target 'CMakeFiles/viewer.dir/all' failed
make[1]: *** [CMakeFiles/viewer.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
[lsd_slam_viewer] <== '/home/pi/lsd_ws/build/lsd_slam_viewer/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' failed with return code '2'
```
I am unable to determine what is the cause of the failure, is it a missing package or dependancy?

I also tried running it as -j1 but had no success.
@mhkabir

blankered, 2016-06-10:
This is the error again 

```
Starting ==> cv_bridge
Starting ==> libg2o
Starting ==> lsd_slam
Finished <== cv_bridge     [ 2 minutes and 0.7 seconds ]
Finished <== lsd_slam      [ 2 minutes and 11.3 seconds ]
Finished <== libg2o        [ 1 hour 18 minutes and 50.1 seconds ]        Starting ==> lsd_slam_core
*** Error in `/usr/bin/c++': double free or corruption (top): 0x01f24048 ***

[lsd_slam_core] ==> '/home/pi/lsd_ws/build/lsd_slam_core/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' in '/home/pi/lsd_ws/build/lsd_slam_core'
Scanning dependencies of target lsdslam
[  3%] Building CXX object CMakeFiles/lsdslam.dir/src/DataStructures/Frame.cpp.o
CMakeFiles/lsdslam.dir/build.make:54: recipe for target 'CMakeFiles/lsdslam.dir/src/DataStructures/Frame.cpp.o' failed
make[2]: *** [CMakeFiles/lsdslam.dir/src/DataStructures/Frame.cpp.o] Aborted
CMakeFiles/Makefile2:475: recipe for target 'CMakeFiles/lsdslam.dir/all' failed
make[1]: *** [CMakeFiles/lsdslam.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
[lsd_slam_core] <== '/home/pi/lsd_ws/build/lsd_slam_core/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' failed with return code '2'

Failed   <== lsd_slam_core [ 35.6 seconds ]
[build] There were '1' errors:

Failed to build package 'lsd_slam_core' because the following command:

# Command to reproduce:
cd /home/pi/lsd_ws/build/lsd_slam_core && /home/pi/lsd_ws/build/lsd_slam_core/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j; cd -

# Path to log:
cat /home/pi/lsd_ws/build/build_logs/lsd_slam_core.log

Exited with return code: 2

[build] Runtime: 1 hour 19 minutes and 26.1 seconds
```
We tried compiling just this file alone without flags and it seems to work through it slowly but does not time out right away as with flags.

Any ideas on where to troubleshoot would be greatly appreciated.

smitherson, 2016-06-14:
I had similar problem - it was because i had installed too new version of Eigen. Install The latest stable version of Eigen and recompile.

oblivione, 2016-07-07:
@blankered i think the errors in the first comment may be because of missing dependencies  for header file generation from msg files.

"CMakeFiles.txt" in the folder "lsd_slam/lsd_slam_core" add:
add_dependencies(lsdslam lsd_slam_viewer_generate_messages_cpp)
add_dependencies(live_slam lsd_slam_viewer_generate_messages_cpp)
add_dependencies(dataset lsd_slam_viewer_generate_messages_cpp)

"CMakeFiles.txt" in the folder "lsd_slam/lsd_slam_viewer" add:
add_dependencies(viewer lsd_slam_viewer_generate_messages_cpp)