tum-vision/lsd_slam #203: how use variable baseline to  calculate  inverseDepth

tsingjinyun, 2016-06-16:
hi,everyone:
in the  .../src/DepthEstimation/Depth.cpp, in doLineStereo() function,
https://github.com/tum-vision/lsd_slam/blob/master/lsd_slam_core/src/DepthEstimation/DepthMap.cpp#L1874
the author use 	idnew_best_match = (dot0 - oldX*dot2) / nominator; 
and i think this used 1/z=d/fB (where f is the focal length,B is the baseline,z is the pixel depth),then the question come here:
i find it hard to match this formula. anybody come across this problem, thanks for your explanation

tsingjinyun, 2016-06-23:
@JakobEngel，the alpha is  the length of the searched inverse depth interval and  the length of the searched epipolar line segment ratio，
https://github.com/tum-vision/lsd_slam/blob/master/lsd_slam_core/src/DepthEstimation/DepthMap.cpp#L1896
and those lines codes is very important to calculate depth and variance, any mathematics inference behind it?

closed 2016-06-24 by tsingjinyun

HeYijia, 2016-08-18:
Hi @tsingjinyun ,  I think I find it out.  It can be derived from the formula of  the projection of a 3D  world point into a camera image.  If have any question, let me know, GOOD LUCK.
[image]

tsingjinyun, 2016-08-23:
Thanks, I have figure it out. Thanks,anyway

发自 网易邮箱大师
On 08/18/2016 09:52, HeYijia wrote:

Hi @tsingjinyun , I think I find it out. It can be derived from the formula of the projection of a 3D world point into a camera image. If have any question, let me know, GOOD LUCK.