tum-vision/lsd_slam #205: load_module did not return. Unable to determine dependencies for file listed above

shareeff, 2016-07-04:
Hi
I have following issues... I tried rosmake lsd_slam for ros kinetic in ubuntu 16.04 
Thank you.

catkin 0.7.1
  -- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
  [rosbuild] Including /opt/ros/kinetic/share/roslisp/rosbuild/roslisp.cmake
  [rosbuild] Including /opt/ros/kinetic/share/roscpp/rosbuild/roscpp.cmake
  [rosbuild] Including /opt/ros/kinetic/share/rospy/rosbuild/rospy.cmake
  MSG: gencfg_cpp on:LSDSLAMViewerParams.cfg
  ERROR [gendeps] 1 Finding dependencies for /home/shareef/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/cfg/LSDSLAMViewerParams.cfg
  ***********************************************************************************
  load_module did not return. Unable to determine dependencies for file listed above.
  ***********************************************************************************
  Traceback (most recent call last):
    File "/opt/ros/kinetic/share/dynamic_reconfigure/cmake/gendeps", line 66, in <module>
      imp.load_module("__main__", f, srcfile, ('.cfg', 'U', 1))
    File "/home/shareef/anaconda3/lib/python3.5/imp.py", line 234, in load_module
      return load_source(name, filename, file)
    File "/home/shareef/anaconda3/lib/python3.5/imp.py", line 170, in load_source
      module = _exec(spec, sys.modules[name])
    File "<frozen importlib._bootstrap>", line 626, in _exec
    File "<frozen importlib._bootstrap_external>", line 662, in exec_module
    File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
    File "/home/shareef/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/cfg/LSDSLAMViewerParams.cfg", line 7, in <module>
      from dynamic_reconfigure.parameter_generator import *
    File "/opt/ros/kinetic/lib/python2.7/dist-packages/dynamic_reconfigure/parameter_generator.py", line 307
      print Template("Generating reconfiguration files for $name in $pkgname").\
                   ^
  SyntaxError: invalid syntax
  
  CMake Error at /opt/ros/kinetic/share/dynamic_reconfigure/cmake/cfgbuild.cmake:78 (string):
    string sub-command REPLACE requires at least four arguments.
  Call Stack (most recent call first):
    /opt/ros/kinetic/share/dynamic_reconfigure/cmake/cfgbuild.cmake:99 (gencfg_cpp)
    CMakeLists.txt:29 (include)
  
  
  -- [rosbuild] Found SSE3 extensions, using flags: -msse3 -mfpmath=sse
  -- Configuring incomplete, errors occurred!
  See also "/home/shareef/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build/CMakeFiles/CMakeOutput.log".
  See also "/home/shareef/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build/CMakeFiles/CMakeError.log".
  /opt/ros/kinetic/share/mk/cmake.mk:7: recipe for target 'all' failed

heimagithub, 2017-05-12:
Hi did you solve this problem?? I meet same problem. I didn't know how to deal it.

dungtam73, 2017-06-07:
Me too.  How to solve it?

zouyajing, 2017-06-08:
Me too. Anyone help solve this problem? Thanks.

dungtam73, 2017-06-08:
sudo gedit /opt/ros/kinetic/lib/python2.7/dist-packages/dynamic_reconfigure/parameter_generator.py

**--> and edit line 64~70 as bellow:** 

def check_description(description):
    quotes = ['"', "'"]
    for quote in quotes:
        while description.find(quote) != -1:
              description = description.replace(quote, "");
              #raise Exception(r"""quotes not allowed in description string `%s`""" % description)