tum-vision/lsd_slam #206: rosmake lsd_slam fail

hwb0314, 2016-07-04:


oblivione, 2016-07-07:
so u can change it to Catkin branch by 
**cd lsd_slam/   
git checkout catkin** 
you run " ls"   
 response should  :
### LICENSE lsd_slam lsd_slam_core lsd_slam_viewer README.md

 Then add 
**<build_depend>cmake_modules</build_depend>
<run_depend>cmake_modules</run_depend>** 
in "package.xml" both in the folder "lsd_slam/lsd_slam_viewer" and in the folder "lsd_slam/lsd_slam_core".

In "CMakeFiles.txt" both in the folder "lsd_slam/lsd_slam_viewer" and in the folder "lsd_slam/lsd_slam_core"  
add:
**find_package(cmake_modules REQUIRED)**  
check the line below  (X11 may be mising ) 
**target_link_libraries(lsdslam ${FABMAP_LIB} ${G2O_LIBRARIES} ${catkin_LIBRARIES} csparse cxsparse X11)** 

Then u can build your workspace by 
**cd <path_to_catkin_ws> 
catkin_make**   

Then again if  got some error of header file missing then 
"CMakeFiles.txt"  in the folder "lsd_slam/lsd_slam_core"  add:
**add_dependencies(lsdslam lsd_slam_viewer_generate_messages_cpp)
add_dependencies(live_slam  lsd_slam_viewer_generate_messages_cpp)
add_dependencies(dataset lsd_slam_viewer_generate_messages_cpp)**

"CMakeFiles.txt"  in the folder "lsd_slam/lsd_slam_viewer" add:
 **add_dependencies(viewer lsd_slam_viewer_generate_messages_cpp)**

QinZiwen, 2016-08-30:
@oblivione, can compile lsd_slam using catkin_make?

oblivione, 2016-08-30:
yes , u can , just follow the steps in my previous comment . then u can able to use catkin_make .

TungTNguyen, 2016-10-16:
@oblivione : catkin make get stuck in this:
The specified source space "/home/me/lsd_slam/src" does not exist
thoughts?

vacky11, 2017-04-05:
@hwb0314 
I am getting the same error. Were you able to solve this problem?

thatfreefish, 2018-01-04:
a very helpful link: http://visbot.blogspot.hk/2014/11/tutorial-building-of-lsd-slam-on-ros.html
good luck to you !

johnchars, 2019-03-21:
May it's Eigen3 version  or Sophus version too low caused. I change my Eigen library version & Sophus(0.95a), rosmake complied.