tum-vision/lsd_slam #208: double free or corruption (out) from 'rosrun lsd_slam_core live_slam' command

st117752, 2016-07-08:
Please give me some suggestion, I'm really new to Ubuntu, C++ and LSD-SLAM. I am using Opencv 2.4.11 on ubuntu 14.04. I got the problem about 'double free or corruption' when runing live_slam by using command below. 

name@Name:~/LSD_SLAM$ rosrun lsd_slam_core live_slam image:=/image_raw camera_info:=/camera_info
*** Error in `/home/name/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/bin/live_slam': double free or corruption (out): 0x08f11900 ***
Aborted (core dumped)

Thanks in advance for every help.

bchu, 2016-07-08:
I get the same issue, on Ubuntu 15.10.

st117752, 2016-07-08:
Thank you so much @bchu.  I use the same version of Eigen as you use. but it automatically install with ros-indigo-libg2o, if i run ' sudo apt-get remove libeigen3-dev', it'll automatically remove 'ros-indigo-libg2o' also. then It'll cause new error cannot find libg2o instead. Did you get this error?

> 'sudo apt-get install ros-indigo-libg2o'
> ...
> The following extra packages will be installed:
>   libeigen3-dev
> Suggested packages:
>   libeigen3-doc libmrpt-dev
> The following NEW packages will be installed:
>   libeigen3-dev ros-indigo-libg2o
> 0 upgraded, 2 newly installed, 0 to remove and 3 not upgraded.
> Need to get 0 B/1,344 kB of archives

This is an error after remove libeigen3-dev

> name@Name:~$ rosrun lsd_slam_core dataset_slam  _files:/LSD_room/images _hz:=0 /LSD_room/cameraCalibration.cfg  
> /home/name/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/bin/dataset_slam: error while loading shared libraries: libg2o_core.so: cannot open shared object file: No such file or directory

bchu, 2016-07-08:
Oh, I was saying that you _want_ to use `libeigen3-dev`. So it looks like that isn't your problem then.

st117752, 2016-07-08:
Sorry, I was confuse myself.

closed 2016-07-09 by st117752

20chix, 2019-03-04:
Still getting this error , should I downgrade too?