tum-vision/lsd_slam #214: Sophus exception: Scale factor is not positive

af-silva, 2016-07-22:
Hi,

First of all thanks to the LSD-SLAM team for providing the code.
Second, a little information about what i'm doing:
I'm trying to using LSD-SLAM to obtain three different things, (1) the camera pose, (2) the 3D information 
of surroundings and (3) visual odometry of the camera (more specific if it's moving or not). The LSD-SLAM is running on a video stream of out-door environment, more specific a trail and has only forward movement. I'm able to access the 3D-Pointcloud whit some information and the camera pose from the Viewer code, but the lsd_slam_core some times "explode" giving me the following error output:

```
Cholesky failure, writing debug.txt (Hessian loadable by Octave)
int g2o::csparse_extension::cs_cholsolsymb(const cs_di*, double*, const cs_dis*, double*, int*): cholesky failed!

Cholesky failure, writing debug.txt (Hessian loadable by Octave)
terminate called after throwing an instance of 'Sophus::ScaleNotPositive'

what():  Sophus exception: Scale factor is not positive
```

The last one "explodes" the lsd_slam_core but i'm allways getting the **Cholesky failure** error...
What can I do to mitigate this error or even better to solve this and the error never happen again (the correct question is, what i'm doing wrong here and why this exception happens :p )???

If i haven't explained myself very well sorry :) if more information is needed please ask.
Thanks in advance :)

romdos, 2017-04-05:
Did you solve this problem?

af-silva, 2017-04-06:
Which part? If you refer to the exception, this one it's never show up again from recente usage of the LSD-SLAM (better video??) but if you refer to the `Cholesky failure` part this one keeps showing up but does not 
interfere with my usage and so I don't care much about.

romdos, 2017-04-06:
I mean Cholesky failure. I've partially solved this problem sending another format of calibration file  (where fx and fy are divided by width and height). Before I put Opencv calibration format and it abused. What is interesting if I run with  camera_info ros parameter of my streaming camera, it works with opencv fx, fy values (not divided). But if I run it offline publishing from avi video and sending opencv calibration format file, it throws me the above error. I'm really stucked. Is it illegal to divide opencv calibration parameters or not? Also I am curious at "e1 e2 e3" parameters after "crop", "full", "none" in. cfg examples. What is the sense of it?