tum-vision/lsd_slam #217: how to use lsd_slam with a usb camera?

zuomaidi52, 2016-08-09:
Hi,I clone the source code and ran it with dataset.Now,I want to run it with my **logitech usb camera**, but in README.md just tell me use 
`rosrun lsd_slam_core live_slam /image:=<yourstreamtopic> /camera_info:=<yourcamera_infotopic>`
what's mean in **<yourstreamtopic>** and **<yourcamera_infotopic>**?
Thank you in advance!

af-silva, 2016-08-09:
@zuomaidi52 Have you used ROS?? If not, my suggestion is to get familiarized with it 
[ROS](http://www.ros.org/). It's the easiest way to get started using LSD-SLAM.

the `<yourstreamtopic>` and `<yourcamera_infotopic>` are ROS topics from where the LSD-SLAM gets the live stream from the camera and information about the camera.

If you want to use the webcam to acquire the video to use in the LSD-SLAM using ROS you need to have the ROS installed in your computer... Next you need to install the following ROS [package](http://wiki.ros.org/usb_cam) and then you are ready to go. The image topic will be /camera/image or /camera/image_raw and the camera information you don't need, if you pass the values from the calibration file. To calibrate the camera you can use this ROS [package](http://wiki.ros.org/camera_calibration).

Hope this helps :)

zuomaidi52, 2016-08-10:
Hi af-silva,
    I have used ROS several months, I am so sry for my question is not clear.I just want to ask what's node publish the `<yourstreamtopic>`.Now I used `rosrun uvc_camera uvc_camera_node device:=/dev/video0` to publish camera image.
    Thank you for your help! :+1:

af-silva, 2016-08-10:
@zuomaidi52 No worries :)  If you are using the uvc_camera package the stream is being publish to the `camera/image_raw` ROS topic (by default), so you need to replace the `<yourstreamtopic>` with the ROS topic. If you need to see which are the topics being published run the `rostopic list` command and you will be able to see them. Another thing that is handy is the ROS `rqt` package, you can run it like `rosrun rqt_gui rqt_gui` and from there you can use any of the available plugins, but the one that is more useful for you perhaps is the `rosrun rqt_image_view rqt_image_view`, here you can see all the topics that are streaming videos/images and the corresponding path...

Hope this helps :)

zuomaidi52, 2016-08-19:
@af-silva I am very happy to got your help. Thanks!
And your icon is very cute :)!

af-silva, 2016-08-19:
No problem :) 
If your question is solved, please mark this issue as closed to keep the issues list clean.

closed 2016-08-22 by zuomaidi52