tum-vision/lsd_slam #219: rosmake failed on TK1

scottybit, 2016-08-22:
I have compiled LSD on my desktop. Now I am trying to compile the code on Jetson TK1(ubuntu14.04+ROS indigo), but failed when rosmake.

Romoving `add_definitions("-DENABLE_SSE")` or replacing it to `add_definitions("-DENABLE_NEON")` in CMakelists.txt did not work.

> mkdir -p bin
cd build && cmake -Wdev -DCMAKE_TOOLCHAIN_FILE=/opt/ros/indigo/share/ros/core/rosbuild/rostoolchain.cmake  ..
[rosbuild] Building package lsd_slam_viewer
[rosbuild] using multiarch 'arm-linux-gnueabihf' for finding Boost
-- Using CATKIN_DEVEL_PREFIX: /home/ubuntu/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build/devel
-- Using CMAKE_PREFIX_PATH: /home/ubuntu/catkin_ws/devel;/opt/ros/indigo
-- This workspace overlays: /home/ubuntu/catkin_ws/devel;/opt/ros/indigo
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Skip enable_testing() for dry packages
-- Using CATKIN_TEST_RESULTS_DIR: /home/ubuntu/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.6.16
-- Using these message generators: gencpp;genlisp;genpy
[rosbuild] Including /opt/ros/indigo/share/roslisp/rosbuild/roslisp.cmake
[rosbuild] Including /opt/ros/indigo/share/roscpp/rosbuild/roscpp.cmake
[rosbuild] Including /opt/ros/indigo/share/rospy/rosbuild/rospy.cmake
MSG: gencfg_cpp on:LSDSLAMViewerParams.cfg
[gendeps] Finding dependencies for /home/ubuntu/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/cfg/LSDSLAMViewerParams.cfg

> 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build
cd build && make -j1 -l1
make[1]: Entering directory `/home/ubuntu/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
make[2]: Entering directory `/home/ubuntu/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
make[3]: Entering directory `/home/ubuntu/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
make[3]: Leaving directory `/home/ubuntu/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
make[3]: Entering directory `/home/ubuntu/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
[  7%] Generating ../cfg/cpp/lsd_slam_viewer/LSDSLAMViewerParamsConfig.h, ../docs/LSDSLAMViewerParamsConfig.dox, ../docs/LSDSLAMViewerParamsConfig-usage.dox, ../src/lsd_slam_viewer/cfg/LSDSLAMViewerParamsConfig.py, ../docs/LSDSLAMViewerParamsConfig.wikidoc
make[3]: execvp: ../cfg/LSDSLAMViewerParams.cfg: Permission denied
make[3]: *** [../cfg/cpp/lsd_slam_viewer/LSDSLAMViewerParamsConfig.h] Error 127
make[3]: Leaving directory `/home/ubuntu/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
make[2]: *** [CMakeFiles/ROSBUILD_gencfg_cpp.dir/all] Error 2
make[2]: Leaving directory `/home/ubuntu/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/home/ubuntu/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
make: *** [all] Error 

In some cases, no failure appears but also no excutable is made. says:"No rule to make target None"

Any advice?

nxcxl88, 2016-11-13:
A solution is: find LSDSLAMViewerParams.cfg and use the command "python LSDSLAMViewerParams.cfg " to generate the header file.

heimagithub, 2017-05-12:
@XackC8  Hi~ i try "python LSDSLAMViewerParams.cfg " but it have error 

 ERROR [gendeps] 1 Finding dependencies for /home/heima/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/cfg/LSDSLAMViewerParams.cfg
  ***********************************************************************************
  load_module did not return. Unable to determine dependencies for file listed above.
  ***********************************************************************************
  Traceback (most recent call last):
    File "/opt/ros/indigo/share/dynamic_reconfigure/cmake/gendeps", line 66, in <module>
      imp.load_module("__main__", f, srcfile, ('.cfg', 'U', 1))
    File "/home/heima/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/cfg/LSDSLAMViewerParams.cfg", line 20, in <module>
      gen.add("scaledDepthVarTH", double_t, 0, "log10 of threshold on point's variance, in the respective keyframe's scale. ", -3, -10, 1)
    File "/opt/ros/indigo/lib/python2.7/dist-packages/dynamic_reconfigure/parameter_generator.py", line 273, in add
      self.group.add(name, paramtype, level, description, default, min, max, edit_method) 
    File "/opt/ros/indigo/lib/python2.7/dist-packages/dynamic_reconfigure/parameter_generator.py", line 143, in add
      check_description(description)
    File "/opt/ros/indigo/lib/python2.7/dist-packages/dynamic_reconfigure/parameter_generator.py", line 68, in check_description
      raise Exception(r"""quotes not allowed in description string `%s`""" % description)
  Exception: quotes not allowed in description string `log10 of threshold on point's variance, in the respective keyframe's scale. `
  
  CMake Error at /opt/ros/indigo/share/dynamic_reconfigure/cmake/cfgbuild.cmake:78 (string):
    string sub-command REPLACE requires at least four arguments.
  Call Stack (most recent call first):
    /opt/ros/indigo/share/dynamic_reconfigure/cmake/cfgbuild.cmake:99 (gencfg_cpp)
    CMakeLists.txt:29 (include)

do you know how to Debug this error ??

MC215, 2017-05-21:
I have same problem, how to solve

MC215, 2017-05-21:
我解决了这个问题，可能是python新版本特性，在
gen.add("scaledDepthVarTH", double_t, 0, "log10 of threshold on point's variance, in the respective keyframe's scale. ", -3, -10, 1)
中的说明部分不允许使用引号，应该删除引号，改为
gen.add("scaledDepthVarTH", double_t, 0, "log10 of threshold on point's variance, in the respective keyframe s scale. ", -3, -10, 1)

heimagithub, 2017-05-21:
Hi MC215, Thanks for your help first!!!
I  Change the code of "parameter_generator.py". Cancel "check_description(description)" and re rosmake. it also can PASS.

andyli, 2017-05-24:
For the record, it seems to be a bug in [dynamic_reconfigure](http://wiki.ros.org/dynamic_reconfigure), and it has been fixed there:
https://github.com/ros/dynamic_reconfigure/issues/75

raktimkashyap, 2018-04-03:
I am using ROS-kinetic and Ubuntu 16.04 and getting this error while compiling the packages using "rosmake" command as described in the read-me.md file of the repository.
____________________________________
raktim@raktim-HP-Pavilion-Notebook:~$ rosmake lsd_slam-master
[ rosmake ] rosmake starting...                                                 
[ rosmake ] Packages requested are: ['lsd_slam-master']                         
[ rosmake ] Logging to directory /home/raktim/.ros/rosmake/rosmake_output-20180404-003153
[ rosmake ] Expanded args ['lsd_slam-master'] to:
['lsd_slam_viewer', 'lsd_slam_core']
[rosmake-0] Starting >>> catkin [ make ]                                        
[rosmake-0] Finished <<< catkin ROS_NOBUILD in package catkin
 No Makefile in package catkin
[rosmake-0] Starting >>> genmsg [ make ]                                        
[rosmake-0] Finished <<< genmsg ROS_NOBUILD in package genmsg
 No Makefile in package genmsg
[rosmake-0] Starting >>> genlisp [ make ]                                       
[rosmake-3] Starting >>> gencpp [ make ]                                        
[rosmake-1] Starting >>> genpy [ make ]                                         
[rosmake-0] Finished <<< genlisp ROS_NOBUILD in package genlisp
 No Makefile in package genlisp
[rosmake-2] Starting >>> geneus [ make ]                                        
[rosmake-0] Starting >>> gennodejs [ make ]                                     
[rosmake-1] Finished <<< genpy ROS_NOBUILD in package genpy
 No Makefile in package genpy
[rosmake-3] Finished <<< gencpp ROS_NOBUILD in package gencpp
 No Makefile in package gencpp
[rosmake-2] Finished <<< geneus ROS_NOBUILD in package geneus
 No Makefile in package geneus
[rosmake-1] Starting >>> opencv3 [ make ]                                       
[rosmake-0] Finished <<< gennodejs ROS_NOBUILD in package gennodejs
 No Makefile in package gennodejs
[rosmake-3] Starting >>> cpp_common [ make ]                                    
[rosmake-2] Starting >>> rosgraph [ make ]                                      
[rosmake-0] Starting >>> message_generation [ make ]                            
[rosmake-3] Finished <<< cpp_common ROS_NOBUILD in package cpp_common
 No Makefile in package cpp_common
[rosmake-3] Starting >>> rostime [ make ]                                       
[rosmake-2] Finished <<< rosgraph ROS_NOBUILD in package rosgraph
 No Makefile in package rosgraph
[rosmake-2] Starting >>> rosparam [ make ]                                      
[rosmake-0] Finished <<< message_generation ROS_NOBUILD in package message_generation
 No Makefile in package message_generation
[rosmake-0] Starting >>> cmake_modules [ make ]                                 
[rosmake-1] Finished <<< opencv3 ROS_NOBUILD in package opencv3
 No Makefile in package opencv3
[rosmake-1] Starting >>> rosmaster [ make ]                                     
[rosmake-3] Finished <<< rostime ROS_NOBUILD in package rostime
 No Makefile in package rostime
[rosmake-3] Starting >>> roscpp_traits [ make ]                                 
[rosmake-2] Finished <<< rosparam ROS_NOBUILD in package rosparam
 No Makefile in package rosparam
[rosmake-2] Starting >>> roslang [ make ]                                       
[rosmake-0] Finished <<< cmake_modules ROS_NOBUILD in package cmake_modules
 No Makefile in package cmake_modules
[rosmake-2] Finished <<< roslang ROS_NOBUILD in package roslang
 No Makefile in package roslang
[rosmake-1] Finished <<< rosmaster ROS_NOBUILD in package rosmaster
 No Makefile in package rosmaster
[rosmake-0] Starting >>> rospack [ make ]                                       
[rosmake-2] Starting >>> xmlrpcpp [ make ]                                      
[rosmake-1] Starting >>> rosclean [ make ]                                      
[rosmake-3] Finished <<< roscpp_traits ROS_NOBUILD in package roscpp_traits
 No Makefile in package roscpp_traits
[rosmake-3] Starting >>> roscpp_serialization [ make ]                          
[rosmake-0] Finished <<< rospack ROS_NOBUILD in package rospack
 No Makefile in package rospack
[rosmake-1] Finished <<< rosclean ROS_NOBUILD in package rosclean
 No Makefile in package rosclean
[rosmake-0] Starting >>> roslib [ make ]                                        
[rosmake-2] Finished <<< xmlrpcpp ROS_NOBUILD in package xmlrpcpp
 No Makefile in package xmlrpcpp
[... 98 lines cut ...]
  Exception: quotes not allowed in description string `log10 of threshold on point's variance, in the respective keyframe's scale. `
  
  CMake Error at /opt/ros/kinetic/share/dynamic_reconfigure/cmake/cfgbuild.cmake:78 (string):
    string sub-command REPLACE requires at least four arguments.
  Call Stack (most recent call first):
    /opt/ros/kinetic/share/dynamic_reconfigure/cmake/cfgbuild.cmake:99 (gencfg_cpp)
    CMakeLists.txt:29 (include)
  
  
  -- [rosbuild] Found SSE3 extensions, using flags: -msse3 -mfpmath=sse
  -- Configuring incomplete, errors occurred!
  See also "/home/raktim/rosbuild_ws/package_dir/lsd_slam-master/lsd_slam_viewer/build/CMakeFiles/CMakeOutput.log".
  See also "/home/raktim/rosbuild_ws/package_dir/lsd_slam-master/lsd_slam_viewer/build/CMakeFiles/CMakeError.log".
  /opt/ros/kinetic/share/mk/cmake.mk:7: recipe for target 'all' failed
-------------------------------------------------------------------------------}
[ rosmake ] Output from build of package lsd_slam_viewer written to:
[... cut ...]
[rosmake-3] Finished <<< lsd_slam_viewer [FAIL] [ 5.68 seconds ]                
[ rosmake ] Halting due to failure in package lsd_slam_viewer. 
[ rosmake ] Waiting for other threads to complete.
[ rosmake ] Results:                                                            
[ rosmake ] Built 45 packages with 1 failures.                                  
[ rosmake ] Summary output to directory                                         
[ rosmake ] /home/raktim/.ros/rosmake/rosmake_output-20180404-003153            
raktim@raktim-HP-Pavilion-Notebook:~$

bespoke-code, 2018-04-04:
Hi @raktimkashyap, 

What you encounter is a problem/bug in the dynamic_reconfigure package. It has been supposedly fixed, as noted by @andyli some time ago in this same thread (I haven't tested the change to confirm).

More specifically, here is what Rosmake reports: 
`Exception: quotes not allowed in description string log10 of threshold on point's variance, in the respective keyframe's scale.`

The problem arises when one wants to use single quotation marks within a string inside a .cfg file. If you go through the .cfg files which raise an error and remove all single quotation marks (') from the strings, the code should compile. I did this and the code compiled successfully on 14.04.

A simple example for the quick fix:
master/lsd_slam_viewer/cfg/LSDSLAMViewerParams.cfg", line 20, in 
`gen.add("scaledDepthVarTH", double_t, 0, "log10 of threshold on point's variance, in the respective keyframe's scale. ", -3, -10, 1)`
would be changed to
`gen.add("scaledDepthVarTH", double_t, 0, "log10 of threshold of a point variance, in the respective keyframe scale. ", -3, -10, 1)`

I hope this can help!