tum-vision/lsd_slam #220: rosmake fault

QinZiwen, 2016-09-05:
lion@lion-Veriton-B832:~/rosbuild_ws/package_dir$ rosmake lsd_slam
[ rosmake ] rosmake starting...                                                                                                                          
[ rosmake ] Packages requested are: ['lsd_slam']                                                                                                         
[ rosmake ] Logging to directory /home/lion/.ros/rosmake/rosmake_output-20160905-215053                                                                  
[ rosmake ] Expanded args ['lsd_slam'] to:
['lsd_slam_viewer', 'lsd_slam_core']                                                                          
[rosmake-0] Starting >>> catkin [ make ]                                                                                                                 
[rosmake-0] Finished <<< catkin ROS_NOBUILD in package catkin
 No Makefile in package catkin                                                             
[rosmake-0] Starting >>> genmsg [ make ]                                                                                                                 
[rosmake-6] Starting >>> cpp_common [ make ]                                                                                                             
[rosmake-3] Starting >>> cmake_modules [ make ]                                                                                                          
[rosmake-4] Starting >>> rosgraph [ make ]                                                                                                               
[rosmake-0] Finished <<< genmsg ROS_NOBUILD in package genmsg
 No Makefile in package genmsg                                                             
[rosmake-5] Starting >>> rosclean [ make ]                                                                                                               
[rosmake-6] Finished <<< cpp_common ROS_NOBUILD in package cpp_common
 No Makefile in package cpp_common                                                 
[rosmake-3] Finished <<< cmake_modules ROS_NOBUILD in package cmake_modules
 No Makefile in package cmake_modules                                        
[rosmake-4] Finished <<< rosgraph ROS_NOBUILD in package rosgraph
 No Makefile in package rosgraph                                                       
[rosmake-7] Starting >>> genlisp [ make ]                                                                                                                
[rosmake-0] Starting >>> genpy [ make ]                                                                                                                  
[rosmake-6] Starting >>> gencpp [ make ]                                                                                                                 
[rosmake-3] Starting >>> rostime [ make ]                                                                                                                
[rosmake-4] Starting >>> rospack [ make ]                                                                                                                
[rosmake-1] Starting >>> rosparam [ make ]                                                                                                               
[rosmake-2] Starting >>> rosmaster [ make ]                                                                                                              
[rosmake-7] Finished <<< genlisp ROS_NOBUILD in package genlisp
 No Makefile in package genlisp                                                          
[rosmake-5] Finished <<< rosclean ROS_NOBUILD in package rosclean
 No Makefile in package rosclean                                                       
[rosmake-5] Starting >>> roslang [ make ]                                                                                                                
[rosmake-7] Starting >>> xmlrpcpp [ make ]                                                                                                               
[rosmake-2] Finished <<< rosmaster ROS_NOBUILD in package rosmaster
 No Makefile in package rosmaster                                                    
[rosmake-6] Finished <<< gencpp ROS_NOBUILD in package gencpp
 No Makefile in package gencpp                                                             
[rosmake-0] Finished <<< genpy ROS_NOBUILD in package genpy
 No Makefile in package genpy                                                                
[rosmake-0] Starting >>> message_generation [ make ]                                                                                                     
[rosmake-1] Finished <<< rosparam ROS_NOBUILD in package rosparam
 No Makefile in package rosparam                                                       
[rosmake-5] Finished <<< roslang ROS_NOBUILD in package roslang
 No Makefile in package roslang                                                          
[rosmake-7] Finished <<< xmlrpcpp ROS_NOBUILD in package xmlrpcpp
 No Makefile in package xmlrpcpp                                                       
[rosmake-3] Finished <<< rostime ROS_NOBUILD in package rostime
 No Makefile in package rostime                                                          
[rosmake-3] Starting >>> roscpp_traits [ make ]                                                                                                          
[rosmake-4] Finished <<< rospack ROS_NOBUILD in package rospack
 No Makefile in package rospack                                                          
[rosmake-0] Finished <<< message_generation ROS_NOBUILD in package message_generation
 No Makefile in package message_generation                         
[rosmake-7] Starting >>> roslib [ make ]                                                                                                                 
[rosmake-7] Finished <<< roslib ROS_NOBUILD in package roslib
 No Makefile in package roslib                                                             
[rosmake-3] Finished <<< roscpp_traits ROS_NOBUILD in package roscpp_traits
 No Makefile in package roscpp_traits                                        
[rosmake-3] Starting >>> roscpp_serialization [ make ]                                                                                                   
[rosmake-7] Starting >>> rosunit [ make ]                                                                                                                
[rosmake-3] Finished <<< roscpp_serialization ROS_NOBUILD in package roscpp_serialization
 No Makefile in package roscpp_serialization                   
[rosmake-1] Starting >>> message_runtime [ make ]                                                                                                        
[rosmake-7] Finished <<< rosunit ROS_NOBUILD in package rosunit
 No Makefile in package rosunit                                                          
[rosmake-4] Starting >>> roslz4 [ make ]                                                                                                                 
[rosmake-1] Finished <<< message_runtime ROS_NOBUILD in package message_runtime
 No Makefile in package message_runtime                                  
[... 85 lines cut ...]
  /opt/ros/indigo/include/sophus/sim3.hpp:339:5: error: passing ‘const RxSO3Type {aka const Sophus::RxSO3Group<double>}’ as ‘this’ argument of ‘void Sophus::RxSO3GroupBase<Derived>::setScale(const Scalar&) [with Derived = Sophus::RxSO3Group<double>; Sophus::RxSO3GroupBase<Derived>::Scalar = double]’ discards qualifiers [-fpermissive]
       rxso3().setScale(scale);
       ^
  make[3]: *** [CMakeFiles/lsdslam.dir/src/DataStructures/FramePoseStruct.cpp.o] 错误 1
  make[3]: *** [CMakeFiles/lsdslam.dir/src/DataStructures/FrameMemory.cpp.o] 错误 1
  make[3]: *** [CMakeFiles/lsdslam.dir/src/DataStructures/Frame.cpp.o] 错误 1
  make[3]: *** [CMakeFiles/lsdslam.dir/src/DepthEstimation/DepthMapPixelHypothesis.cpp.o] 错误 1
  make[3]: *** [CMakeFiles/lsdslam.dir/src/util/globalFuncs.cpp.o] 错误 1
  make[3]: *** [CMakeFiles/lsdslam.dir/src/DepthEstimation/DepthMap.cpp.o] 错误 1
  make[3]: *** [CMakeFiles/lsdslam.dir/src/SlamSystem.cpp.o] 错误 1
  make[3]:正在离开目录 `/home/lion/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
  make[2]: *** [CMakeFiles/lsdslam.dir/all] 错误 2
  make[2]:正在离开目录 `/home/lion/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
[... cut ...]
[ rosmake ] Results:                                                                                                                                     
[ rosmake ] Built 42 packages with 1 failures.                                                                                                           
[ rosmake ] Summary output to directory                                                                                                                  
[ rosmake ] /home/lion/.ros/rosmake/rosmake_output-20160905-215053

TungTNguyen, 2016-10-16:
try make in lsd slam core folder

Dorothy-2016, 2017-02-06:
Do U have solve this problem? @QinZiwen

QinZiwen, 2017-02-09:
NO, I am busy recently @Dorothy-2016