tum-vision/lsd_slam #222: lsd_slam_core and lsd_slam_viewer in Ubuntu 16.04

chiragmajithia, 2016-10-17:
Hey,
I was able to compile lsd_slam on ubuntu 16.04 with ros-kinetic. However, while running lsd_slam_viewer I get following error: 

> *** Error in `/home/chirag/vision/SLAM-ROS/package_dir/lsd_slam/lsd_slam_viewer/bin/viewer': realloc(): invalid pointer: 0x00007fda0b922820 ***
> ======= Backtrace: =========
> /lib/x86_64-linux-gnu/libc.so.6(+0x77725)[0x7fda0a30a725]
> /lib/x86_64-linux-gnu/libc.so.6(realloc+0x348)[0x7fda0a316f88]
> /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN9QListData7reallocEi+0x1f)[0x7fda056b9a5f]
> /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(_ZN9QListData6appendEi+0x81)[0x7fda056b9b31]
> /usr/lib/x86_64-linux-gnu/libQt5Core.so.5(+0x1d7018)[0x7fda05786018]
> ...

The same issue can be seen [here](https://github.com/tum-vision/lsd_slam/issues/33).

As, suggested in issue#33 -- my libqglviewer2 is already the newest version (2.6.3+dfsg1-1).

However, I think the 
`rosrun lsd_slam_core live_slam image:=/image_raw camera_info:=/camera_info`
 is running with following output:

> Received ROS Camera Calibration: fx: 254.326950, fy: 375.934387, cx: 267.381897, cy: 231.599091 @ 640x480
> RECEIVED ROS camera calibration!
> Started mapping thread!
> Started  constraint search thread!
> Started optimization thread 
> Doing Random initialization!
> started image display thread!
> Done Random initialization!
> init done
> QObject::startTimer: Timers cannot be started from another thread
> TRACKING LOST for frame 1175 (0.01% good Points, which is 100.00% of available points, DIVERGED)!
> QObject::killTimer: Timers cannot be stopped from another thread
> QObject::startTimer: Timers cannot be started from another thread
> 

There is nothing echoed from /lsd_slam/pose .. 
There are no tfs published so that I can use them to view in rviz. 
I am relatively new to using ROS  -- is it possible to get the package running on 16.04 kinetic system?

I had to change couple of lines in 
/lsd_slam_viewer/src/PointCloudViewer.cpp and /lsd_slam_viewer/src/PointCloudViewer.h

from float x,y,z to qreal x,y,z beforeqreal x,y,z;
    	  camera()->frame()->getPosition(x,y,z);
function is called

to make the package compile sucessfully.

shubhankar12, 2016-11-05:
Did u make any progress with this

chiragmajithia, 2016-11-07:
Not yet..

martinarroyo, 2016-11-27:
The issue is caused by a conflict between versions 4 and 5 of Qt. I fixed it by following [this instructions](https://github.com/tum-vision/fastfusion/issues/9). I did not uninstall Qt 5 nor did I manually compiled libQGLViewer. I just removed the symlinks to libQGLviewer.so.2.6.3 and created new ones that point to libQGLViewer-qt4.so.2.6.3.

blindnote, 2016-11-28:
I also followed [this](https://github.com/tum-vision/fastfusion/issues/9) except that i didn't setup the soft symlinks:
1.  sudo apt-get purge --auto-remove qt5-default
2.  [qglviewer](http://www.libqglviewer.com/src/libQGLViewer-2.6.3.tar.gz),  decompress and cd into the folder, then qmake & make & sudo make install
3. rebuild lsd_slam

mbennehar, 2016-12-14:
@blindnote your solution works for compilation, but when launching the viewer it is not working, nothing is shown on the window.

blindnote, 2016-12-26:
@mbennehar  is your issue the same with this[https://github.com/tum-vision/lsd_slam/issues/224] ?

mbennehar, 2016-12-26:
@blindnote yes it is the same error. I fixed it by removing some wait returns in one cpp file (sorry but I don't remember which one or which lines). This only fixed the pointcloud viewer. The window showing depth pixels never showed up. It should be noted that I'm using a catkin version of this package on Ubuntu 16.04 on ROS kinetic. On a virtual machine running Ubuntu 14.04 and ROS indigo, everything worked just fine.

jiapei100, 2017-01-28:
Same issue here...
Qt 5.8.0 under Ubuntu 16.04.1

Any solutions?

Cheers
Pei

tsuyony1982, 2017-02-20:
@jiapei100  
I installed ros-indigo to ubuntu 16.04 from src, and installed lsd_slam under ros-indigo workspace.
I seems working well.
(I have encountered a issue https://github.com/tum-vision/lsd_slam/issues/237 when compiling ros-indigo )

jeremyfix, 2017-03-11:
I experienced the same issue (on a 16.04 LTS with kinetic) and solved it by following something similar to @martinarroyo .

sudo apt remove libqglviewer-dev
sudo apt install libqglviewer-dev-qt4
cd /usr/lib/x86_64-linux-gnu
sudo ln -s libQGLViewer-qt4.so libQGLViewer.so

cd catkin_ws
catkin clean
catkin build lsd_slam_viewer

rosrun lsd_slam_viewer viewer

EduardoTayupanta, 2017-09-12:
someone solved this mistake:
QObject::startTimer: Timers cannot be started from another thread
^CQObject::~QObject: Timers cannot be stopped from another thread

Young532042725, 2018-01-20:
@jeremyfix @mbennehar  Hi, friends. so desperate!  rosmake LSD-slam, I had to change couple of lines in
/lsd_slam_viewer/src/PointCloudViewer.cpp: 326 and /lsd_slam_viewer/src/PointCloudViewer.h:135

326:from float x,y,z to double x,y,z 
camera()->frame()->getPosition(x,y,z);

135:from float x,y,z to double x,y,z 
After this,  rosmake again. some error occurs！！ may u give me some advice?  thank u very much!
[image]

johnchars, 2019-03-27:
Maybe the Eigen3 version leads to this question, I have replaced my Eigen3.3 with Eigen3.2.9, after compile, its fixed.

Cielyuy, 2019-12-23:
> I also followed [this](https://github.com/tum-vision/fastfusion/issues/9) as @martinarroyo except that i didn't setup the soft symlinks:
> 
> 1. sudo apt-get purge --auto-remove qt5-default
> 2. [qglviewer](http://www.libqglviewer.com/src/libQGLViewer-2.6.3.tar.gz),  decompress and cd into the folder, then qmake & make & sudo make install
> 3. rebuild lsd_slam

thanks for ur advice. However, when I input the first demand, bash says Package 'qt5-default' is not installed, so not removed. but I do find qt5 in PC.