tum-vision/lsd_slam #224: QObject::startTimer: Timers cannot be started from another thread

TungTNguyen, 2016-10-22:
Dear TUM Vision lab and LSD SLAM users,
Newbie here. I'm trying LSD SLAM for a visual localization problem. Here's what I got after running the command-line: rosrun lsd_slam_core live_slam image:=/image_raw camera_info:=/camera_info
The window can't be even rendered. A Google search suggests that it's an issue from Qt side. Any fixes? 
[image]

af-silva, 2016-10-24:
Hey,
What Qt version are you running?
And please, if you can execute the rosrun with --screen at the end and post here what the program writes to the console instead of the print screen... if you want to post the print screen please post one that is more clear, I can't understand what is going on in there...
If I can help I will...

blindnote, 2016-11-29:
@af-silva , i also encounter this issue, i'm using qt4 on ubuntu 16.04 with ROS kinetic, any suggestions.

jiapei100, 2017-01-28:
Same issue here...
Qt 5.8.0 under Ubuntu 16.04.1

Any solutions?

Cheers
Pei

af-silva, 2017-01-30:
@blindnote Without more information its difficult to diagnose! And I don't have the code running on Ubuntu 16.04 so it's difficult to me to pinpoint the error. Two tings I know, first this code runs without problems using Ubuntu 14.04 + Qt4 + ROS Indigo and second the ROS indigo distribution is not supported by the Ubuntu 16.04 witch recommends the ROS Kinetic that has less packages available in the repos and some times requires some troubleshooting to get things working has expected...

Ok now some possible solutions! Try to uninstall the Qt5 and install only the Qt4 in your machine and try to compile the code and run again some example. Second some functions signatures and other stuff have changed from Qt4 to Qt5 so see this [link](http://wiki.ros.org/kinetic/Migration) and dig into the code and fix the problems.

If I had some time I would like to help you guys finding a solution to this, but I'm working with other things using this code and my time is very limited. But if you want to run this code on ubuntu 16.04 using ROS Kinetic (I will assume this is the distributions of ROS you are using) you will have to migrate the code base from ROS Indigo, it's not very difficult but requires some patience and time, and some times some trial and error. Much of the problems rise from dependencies, and if you are using new API's (new versions) the probability that some stuff has changed is from the previous ones is great.

Another thing that you can do is to run the gdb toll ([link](https://www.sourceware.org/gdb/)) that will show you more debug information than the usual, and perhaps you can better pinpoint the problem or at least give more information here so some one can help you better! To run the gdb tool you will have to install it if you haven't already and then do the following:

(using the command line)
1) go to the folder `devel/lib` and search for the `lsd_slam_viewer` package and enter
2) run the following command `gdb viewer `
3) on the command window press"r" (for run)
4) the program will run and after the error happened, please press "bt" (back-trace) and the gdb tool will give you the backtrace to the error! 

The information can be overwhelming but if you read with some attention you can pinpoint the problem!

I hoe this helps! Sorry for the big text!

blindnote, 2017-02-07:
@af-silva I think the issue is related to depth image display, since when I do below changes, the PointClound Viewer is working perfect:
 lsd_slam_core/src/util/settings.cpp:  bool displayDepthMap = false;

JeroenDelcour, 2017-02-15:
Thanks to @blindnote, I also got it working on Ubuntu 16.04, albeit with the depth view disabled.

I installed `qt4-default` and `libqgtviewer2-qt4` and made a symlink so `lsd_slam_viewer` uses it instead of the qt5 version:

```
sudo apt install qt4-default libqglviewer2-qt4
sudo ln -s /usr/lib/libQGLViewer-qt4.so.2 /usr/lib/x86_64-linux-gnu/libQGLViewer.so.2
```

rnunziata, 2017-05-15:
I also getting the same error but when I tried the solution with libQGViewer it does not seems to work. Running ubuntu 16 with ros kinetic

```
richard@rjn:~$ rosrun lsd_slam_core live_slam image:=/image_raw camera_info:=/camera_info
WAITING for ROS camera calibration!
Received ROS Camera Calibration: fx: 254.326950, fy: 375.934387, cx: 267.381897, cy: 231.599091 @ 640x480
RECEIVED ROS camera calibration!
Started mapping thread!
Doing Random initialization!
Started  constraint search thread!
Started optimization thread 
started image display thread!
Done Random initialization!
init done
QObject::startTimer: Timers cannot be started from another thread
^CQObject::~QObject: Timers cannot be stopped from another thread
live_slam: /usr/include/boost/thread/pthread/condition_variable_fwd.hpp:81: boost::condition_variable::~condition_variable(): Assertion `!ret' failed.
Aborted

```

EduardoTayupanta, 2017-09-12:
someone solved this mistake:
QObject::startTimer: Timers cannot be started from another thread
^CQObject::~QObject: Timers cannot be stopped from another thread

fulkast, 2018-02-06:
This same issue has also been discussed here: [for ORB SLAM](https://github.com/raulmur/ORB_SLAM2/issues/418)
As of the time of writing this post, one of the poster suggesting linking against an OpenCV which is compiled without Qt support. This seemed reasonable because the QObject is a Qt object after all.

I have tried this approach and it has worked for my setup (16.04 ros-kinetic). However, it did require compiling opencv3 and cv-bridge from source.

mrgransky, 2018-11-05:
> displayDepthMap

This solution does not work in ROS kinetict and Ubuntu 16.04.

Cielyuy, 2019-12-23:
> Thanks to @blindnote, I also got it working on Ubuntu 16.04, albeit with the depth view disabled.
>
> I installed `qt4-default` and `libqgtviewer2-qt4` and made a symlink so `lsd_slam_viewer` uses it [...]

thank you. and I wonder when the method works for u? and if I need to remove  qt5 first before using your method.

Cielyuy, 2019-12-23:
> Thanks to @blindnote, I also got it working on Ubuntu 16.04, albeit with the depth view disabled.
>
> I installed `qt4-default` and `libqgtviewer2-qt4` and made a symlink so `lsd_slam_viewer` uses it [...]

This method does not work for me, Thanks the same.