tum-vision/lsd_slam #227: Link libg2o correctly

romdos, 2016-10-30:
Hi! I try to install lsd_slam via catkin_make in ros_workspace, but there are some errors connected with libg2o. I installed it previously and it is located in /opt/ros/kinetic/share/cmake_modules/cmake/Modules. But when I run catkin_make in catkin workspace there is next error. (I use Ubuntu 16.04 and ros kinetic)
Scanning dependencies of target lsdslam
[ 37%] Building CXX object lsd_slam/lsd_slam_viewer/CMakeFiles/viewer.dir/src/main_viewer.o
[ 38%] Building CXX object lsd_slam/lsd_slam_viewer/CMakeFiles/viewer.dir/src/PointCloudViewer.o
[ 42%] Building CXX object lsd_slam/lsd_slam_core/CMakeFiles/lsdslam.dir/src/DataStructures/Frame.cpp.o
[ 42%] Building CXX object lsd_slam/lsd_slam_core/CMakeFiles/lsdslam.dir/src/DataStructures/FramePoseStruct.cpp.o
In file included from /home/roman/ros_workspace/catkin/src/lsd_slam/lsd_slam_core/src/DataStructures/FramePoseStruct.h:23:0,
                 from /home/roman/ros_workspace/catkin/src/lsd_slam/lsd_slam_core/src/DataStructures/FramePoseStruct.cpp:22:
/home/roman/ros_workspace/catkin/src/lsd_slam/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h:24:59: fatal error: /opt/ros/kinetic/include/g2o/core/base_vertex.h: No such file or directory

af-silva, 2016-11-18:
Hi,

Put [this](https://svn.openslam.org/data/svn/vertigo/trunk/cmake-modules/FindG2O.cmake) in your cmake folder located in the lsd_slam_core package and then in the CMakeLists add this line `include(cmake/FindG2O.cmake)` and try again to compile the project.

betsyngten, 2018-10-05:
the link seems to be dead
also i found the include(cmake/FindG20.make) in the CMakeLists already maybe they patched it?

I still get the same error is there another possibility to fix this?

af-silva, 2018-10-07:
Maybe I don't know, you can find it here [link](https://github.com/RainerKuemmerle/g2o/blob/master/cmake_modules/FindG2O.cmake).
Make sure the file is there and that the main CMakeList file has it (the include). If you want, you can go to my fork and see, the code there is being compiled on Ubuntu 16.04 and with ROS Indigo and it works ok.

af-silva, 2018-10-09:
It seems it's missing a header file from the g2o library. So, try to put [this ](https://github.com/af-silva/path_detector_v2/tree/master/g2o )in the project folder and compile again. Are you using catkin?? If so try to compile the catkin branch of the repo. 

The oficial repo for the g2o is [here](https://github.com/RainerKuemmerle/g2o). I would recomend you to use Indigo instead of Kinetic maybe, but it should also work in kinetic. If everything fails, could you please try to use the LSD-SLAM port from my [project](https://github.com/af-silva/path_detector_v2/tree/master/lsd_slam), it as small changes in the code but for now it's is the only one I have sure to have worked in the ubuntu 14.04 distro under ROS Indigo (the oficial LSD-SLAM was having some troubles with catkin at the time, but the branch catking should also work). During the weekend if I have some time I can teste this, but if you could try the different approaches that I have mention above.

Good luck! ;)