tum-vision/lsd_slam #228: Error in Launch the lsd_slam viewer

Moha555, 2016-11-09:
Hi!
 I use `roscore` in a terminal and `rosrun lsd_slam_viewer viewer` in another terminal.
But I receive below errors:
`
Started QApplication thread
libGL error: pci id for fd 16: 80ee:beef, driver (null)
libGL error: core dri or dri2 extension not found
libGL error: failed to load driver: vboxvideo
Started ROS thread
[ INFO] [1478707931.856605266]: lsd_slam_viewer started
`
I follow the installation procedure as in the README and I'm beginner in Linux OS.
what might be wrong?

af-silva, 2016-11-09:
Hi,
Are you running the LSD-SLAM in a virtual machine???

Moha555, 2016-11-09:
Hi, 
yes

af-silva, 2016-11-10:
Hi, 

That is a hard problem to solved without seeing what is going on in the machine. Bu you can try to enable hardware acceleration in your Virtual Machine and make sure that you have the VirtualBox Extension Pack installed. If the problem persists try to look around  for the specific error that you are getting when trying to run the code  `libGL error: failed to load driver: vboxvideo` .

And I don't have experience running this code on a virtual machine, so I don't know how it behaves, but from my experience the LSD-SLAM uses a lot of resources from the machine and in a virtual environment can be a little slow... 

Sorry if I can't help you more than this :)

romdos, 2016-11-15:
I have the next error running lsd_slam_viewer
[image]

Moha555, 2016-11-15:
Dear @af-silva 
My problem is solved... :)
Virtual machine could not run this code
I tried to run it on the Ubunto OS , and eventually it's ran.

af-silva, 2016-11-17:
@romdos I thinks it's better to you to create a Issue related to the problem that your project instance is invoking. Second please provide more information, because just with the error is very difficult to help (e.g. are you using rosbuild or catkin? ). You could try to delete the devel and build folders and compile everything again... it's a memory problem but I don't know what is causing it.

@Moha555 Nice :) happy hacking :p 
If the problem is solved please close the issue, to maintained the issue list clean.

romdos, 2016-11-18:
@af-silva, I have created new issue https://github.com/tum-vision/lsd_slam/issues/230

closed 2016-11-19 by Moha555