tum-vision/lsd_slam #229: How to detect Loop Closures?

GrimHull, 2016-11-17:
Is there a way to get a "notification" that a loop closure has occurred?
From what I understand from the code, it seems the graph is just continuously optimized and a loop closure just "falls into place". I don't see a way to check for loop closure.

Is there a function that tests for loop closure?
I would like to know when it has reached a point seen before and optimizes over the whole map.
If there are 2 loop closures in a trajectory, I would like 2 indications.

Thanks, 
Grim

reynoldscem, 2016-11-21:
While the graph is optimised continuously in the background, explicit checking for closures occurs when a new keyframe is finalised & added to the graph.

If you don't want to dig around the code then I suggest looking at what debug flags there are, and compiling with them. LSD-SLAM can be configured to print out quite a bit of information about this kind of thing.

Otherwise, follow the code path from finalising keyframes.

I'd point you to something more specific, but I haven't worked with LSD-SLAM in a long while.

GrimHull, 2016-11-21:
Hi Reynoldscem,
Thanks for the response. Your debug flag recommendation was helpful.

What I have found is that the keyframes are finalised right after they are chosen and the depth is estimated. And the graph is continuously changing, so 'finalising a keyframe' or adding it to the graph would not mean it's pose is also finalised.

I may have found it though, in file TrackableKeyFrameSearch.cpp
>> Frame* TrackableKeyFrameSearch::findRePositionCandidate(Frame* frame, float maxScore)

This method seems to attempt to find old keyframes that are close to current keyframes and then assigns a score to the evaluation. If a good enough score is determined, then the old keyframe is returned, else nothing is returned. 
This provides a good indication of a loop closure.

If I am mistaken in any of my theory above, please correct me.

Thanks again,
Grim