tum-vision/lsd_slam #232: Is it ok to run lsd_slam on a color topic without converting it to monochrome ?

ap108, 2016-12-01:
Hi 

Thanks for the LSD-SLAM package. I've been running it on the "camera/rgb/image_color" from a kinect bag file and seems to have been working fine.

But I saw [http://answers.ros.org/question/208413/how-to-use-lsd-slam-with-a-usb-webcam/](url)  
which said you have to use  monochrome images.

Is it ok to run lsd_slam on a color topic without converting it to monochrome ?

Does it affect the results in anyway ?

Thank you

af-silva, 2016-12-16:
Hi,

You can run lsd_slam using a color topic, the lsd_slam will convert to monochrome for you. The important thing that you have to have in mind, is to give the lsd_slam the correct resolution, I suggest that you crop the image using the **video_stream_opencv** package. You can use another package if you don't trust the conversion to monochrome to do this job XD

I hope it helped.

ap108, 2016-12-20:
Cool, thanks!