tum-vision/lsd_slam #233: LSD Slam n

LettuceBongani, 2016-12-13:
Good day
I'm new to LSD SLAM and I am looking to use it on a 100 images. I am running on a Ubuntu 14.04 machine, and using ROS Ind
My operation runs as follows:
I open my 1st terminal : roscore
....2nd terminal: rosrun lsd_slam_viewer viewer: point cloud window pops up
.....3rd terminal: I go on to enter the following command:

rosrun lsd_slam_core dataset_slam _files:=home/rick/Mokolo/lsd-test/images _hz:=0_calib:=/home/rick/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/calib/OpenCV_example_calib.cfg

I then get the following output:

Reading Calibration from file /home/rick/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/calib/OpenCV_example_calib.cfg ... found!
found OpenCV camera model, building rectifier.
Input resolution: 752 480
In: 500.000000 500.000000 350.000000 240.000000 0.000000 0.000000 0.000000 0.000000
Out: Crop
Output resolution: 640 480
Started mapping thread!
Started  constraint search thread!
Started optimization thread 
could not load file list! wrong path / file?
Finalizing Graph... finding final constraints!!
Optizing Full Map!
Done optizing Full Map! Added 0 constraints.
Finalizing Graph... optimizing!!
doing final optimization iteration!
Finalizing Graph... publishing!!
Done Finalizing Graph.!!
... waiting for SlamSystem's threads to exit
Exited constraint search thread 
Exited mapping thread 
Exited optimization thread 
DONE waiting for SlamSystem's threads to exit
rick@rick-Lenovo-Y70-70-Touch:~$ rosrun lsd_slam_core dataset_slam _files:=home/rick/Mokolo/lsd-test/images _hz:=30 _calib:=/home/rick/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/calib/OpenCV_example_calib.cfg
Reading Calibration from file /home/rick/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/calib/OpenCV_example_calib.cfg ... found!
found OpenCV camera model, building rectifier.
Input resolution: 752 480
In: 500.000000 500.000000 350.000000 240.000000 0.000000 0.000000 0.000000 0.000000
Out: Crop
Output resolution: 640 480
Started mapping thread!
Started optimization thread 
Started  constraint search thread!
could not load file list! wrong path / file?
Finalizing Graph... finding final constraints!!
Optizing Full Map!
Done optizing Full Map! Added 0 constraints.
Finalizing Graph... optimizing!!
doing final optimization iteration!
Finalizing Graph... publishing!!
Done Finalizing Graph.!!
... waiting for SlamSystem's threads to exit
Exited mapping thread 
Exited constraint search thread 
Exited optimization thread 
DONE waiting for SlamSystem's threads to exit

Should I try out a different calibration setting or is the problem not that?

LettuceBongani, 2016-12-13:
Or is the problem the fact that it cannot load a file list due to a wrong path?

sophie-shin, 2017-01-25:
I also have same issue.
1st terminal : roscore  --> okay
2nd terminal: rosrun lsd_slam_viewer viewer: point cloud window pops up --> with black window
3rd terminal: I go on to enter the following command:
ubuntu@ubuntu:~/rosbuild_ws/package_dir$ rosrun lsd_slam_core live_slam image:=/image_raw camera_info:=/camera_info
--> [rosrun] Couldn't find executable named live_slam below /home/ubuntu/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core

sakshamgupta006, 2017-01-28:
@sophie-shin run the 3rd terminal command in ~/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/bin directory

Ruchikros, 2017-04-11:
@sakshamgupta006  i used catkin_workspace for the same. I wrote: roscore in one terminal,  
  
lsd_slam_viewer viewer in the second and  rosrun lsd_slam_core live_slam image:=/image_raw camera_info:=/camera_info
in the third and got the error as: [rosrun] Couldn't find executable named live_slam below /home/ruchik/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core  
no able to figure out what to do

Roshnee, 2018-07-19:
@sakshamgupta006  there is no bin directory in lsd_slam_core

Roshnee, 2018-07-19:
Did anybody here get it working coz even im having the same issue 

`[rosrun] Couldn't find executable named live_slam below /home/ubuntu/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core`

migtissera, 2018-11-04:
Yeah it's in the devel folder if you built it using catkin_make