tum-vision/lsd_slam #234: what is the IDE of LSD_slam

Moha555, 2016-12-14:
Hello...
I'm a java programmer and I usually program in Windows OS.
Recently, because of need to Slam in my thesis, I'm working on LSD-slam. I'm not familiar with linux and specially C++ in linux.
I need to edit and compile this code but I don't know what's its IDE !
Can anyone help me?

af-silva, 2016-12-14:
You can use any IDE you want that supports a c++ compiler (e.g. Eclipse, QT, etc)! My suggestion is that you use Eclipse with the CDT add-on. Then you can compile the project without any problem, follow the tutorial in the README file. And I strongly suggest that you install a linux environment on your machine (Ubuntu it's recommended), than you can start hacking the code. Another thing, I don't know what will you be developing but you should also go ahead and read some of the documentation of ROS if you want to see first the project running. You can use it as a stand-alone project, but a think it requires some tweeks. 

Please mark this as replayed!

Moha555, 2016-12-16:
@af-silva  Thank you for this complete answer :)
and Excuse for that beginning question