tum-vision/lsd_slam #237: Compilation failure - ununderstandable - Ubuntu 14.04 + ROS Indigo

pierre-alexandre-masset, 2016-12-23:
Hi everybody,

I am currently working on a SLAM integration on ROS for a school project, and I found lsd_slam to be convenient and perfectly doing what I want.

I have been struggling for days now to compile lsd_slam, and each time a new kind of error comes up. I am depressed and I need help.

I have tried the "official" way not using catkin_make, but it failed. I have found this tutorial http://visbot.blogspot.fr/2014/11/tutorial-building-of-lsd-slam-on-ros.html to be very helpful to use catkin and made everything as said. When I type catkin_make I first got an error saying CMake could not find Eigen, but the fixes proposed on the page worked.

So I did everything, including bug fixing proposed but now I have really weird compilation errors : 

(See my CMakeLists.txt and package.xml files attached
[LIST-XML.zip](https://github.com/tum-vision/lsd_slam/files/670635/LIST-XML.zip)

THANK YOU Anyone with any idea... I am just going crazy and dreaming of fixing this issue literally ! ahah

masset@masset-G75VX:~/lsd_ws$ catkin_make
Base path: /home/masset/lsd_ws
Source space: /home/masset/lsd_ws/src
Build space: /home/masset/lsd_ws/build
Devel space: /home/masset/lsd_ws/devel
Install space: /home/masset/lsd_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/masset/lsd_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/masset/lsd_ws/build"
####
[  2%] Built target lsd_slam_viewer_gencfg
[  7%] Built target lsd_slam_core_gencfg
[  9%] Building CXX object lsd_slam/lsd_slam_viewer/CMakeFiles/viewer.dir/src/KeyFrameDisplay.o
[  9%] [  9%] Built target _lsd_slam_viewer_generate_messages_check_deps_keyframeGraphMsg
Built target _lsd_slam_viewer_generate_messages_check_deps_keyframeMsg
[ 26%] [ 24%] [ 26%] Built target lsd_slam_viewer_generate_messages_lisp
Built target lsd_slam_viewer_generate_messages_cpp
Built target lsd_slam_viewer_generate_messages_py
[ 26%] Built target lsd_slam_viewer_generate_messages
[ 85%] Built target lsdslam
Linking CXX executable /home/masset/lsd_ws/devel/lib/lsd_slam_core/live_slam
Linking CXX executable /home/masset/lsd_ws/devel/lib/lsd_slam_core/dataset
/home/masset/lsd_ws/devel/lib/liblsdslam.so : référence indéfinie vers « cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&) »
/home/masset/lsd_ws/devel/lib/liblsdslam.so : référence indéfinie vers « cv::line(cv::_InputOutputArray const&, cv::Point_<int>, cv::Point_<int>, cv::Scalar_<double> const&, int, int, int) »
/home/masset/lsd_ws/devel/lib/liblsdslam.so : référence indéfinie vers « cv::String::allocate(unsigned long) »
/home/masset/lsd_ws/devel/lib/liblsdslam.so : référence indéfinie vers « cv::String::deallocate() »
/home/masset/lsd_ws/devel/lib/liblsdslam.so : référence indéfinie vers « cv::putText(cv::_InputOutputArray const&, cv::String const&, cv::Point_<int>, int, double, cv::Scalar_<double>, int, int, bool) »
/home/masset/lsd_ws/devel/lib/liblsdslam.so : référence indéfinie vers « cv::error(int, cv::String const&, char const*, char const*, int) »
/home/masset/lsd_ws/devel/lib/liblsdslam.so : référence indéfinie vers « cv::namedWindow(cv::String const&, int) »
/home/masset/lsd_ws/devel/lib/liblsdslam.so : référence indéfinie vers « cv::resizeWindow(cv::String const&, int, int) »
/home/masset/lsd_ws/devel/lib/liblsdslam.so : référence indéfinie vers « cv::imshow(cv::String const&, cv::_InputArray const&) »
collect2: error: ld returned 1 exit status
make[2]: *** [/home/masset/lsd_ws/devel/lib/lsd_slam_core/live_slam] Erreur 1
make[1]: *** [lsd_slam/lsd_slam_core/CMakeFiles/live_slam.dir/all] Erreur 2
make[1]: *** Attente des tâches non terminées....
CMakeFiles/dataset.dir/src/main_on_images.cpp.o : Dans la fonction « cvflann::anyimpl::big_any_policy<cv::String>::static_delete(void**) » :
main_on_images.cpp:(.text._ZN7cvflann7anyimpl14big_any_policyIN2cv6StringEE13static_deleteEPPv[_ZN7cvflann7anyimpl14big_any_policyIN2cv6StringEE13static_deleteEPPv]+0x15) : référence indéfinie vers « cv::String::deallocate() »
CMakeFiles/dataset.dir/src/main_on_images.cpp.o : Dans la fonction « cvflann::anyimpl::big_any_policy<cv::String>::move(void* const*, void**) » :
main_on_images.cpp:(.text._ZN7cvflann7anyimpl14big_any_policyIN2cv6StringEE4moveEPKPvPS5_[_ZN7cvflann7anyimpl14big_any_policyIN2cv6StringEE4moveEPKPvPS5_]+0x10) : référence indéfinie vers « cv::String::deallocate() »
main_on_images.cpp:(.text._ZN7cvflann7anyimpl14big_any_policyIN2cv6StringEE4moveEPKPvPS5_[_ZN7cvflann7anyimpl14big_any_policyIN2cv6StringEE4moveEPKPvPS5_]+0x24) : référence indéfinie vers « cv::String::deallocate() »
CMakeFiles/dataset.dir/src/main_on_images.cpp.o : Dans la fonction « main » :
main_on_images.cpp:(.text.startup+0x987) : référence indéfinie vers « cv::imread(cv::String const&, int) »
main_on_images.cpp:(.text.startup+0x994) : référence indéfinie vers « cv::String::deallocate() »
main_on_images.cpp:(.text.startup+0x9fc) : référence indéfinie vers « cv::String::allocate(unsigned long) »
main_on_images.cpp:(.text.startup+0xfd3) : référence indéfinie vers « cv::String::deallocate() »
/home/masset/lsd_ws/devel/lib/liblsdslam.so : référence indéfinie vers « cv::imwrite(cv::String const&, cv::_InputArray const&, std::vector<int, std::allocator<int> > const&) »
/home/masset/lsd_ws/devel/lib/liblsdslam.so : référence indéfinie vers « cv::line(cv::_InputOutputArray const&, cv::Point_<int>, cv::Point_<int>, cv::Scalar_<double> const&, int, int, int) »
/home/masset/lsd_ws/devel/lib/liblsdslam.so : référence indéfinie vers « cv::putText(cv::_InputOutputArray const&, cv::String const&, cv::Point_<int>, int, double, cv::Scalar_<double>, int, int, bool) »
/home/masset/lsd_ws/devel/lib/liblsdslam.so : référence indéfinie vers « cv::error(int, cv::String const&, char const*, char const*, int) »
/home/masset/lsd_ws/devel/lib/liblsdslam.so : référence indéfinie vers « cv::namedWindow(cv::String const&, int) »
/home/masset/lsd_ws/devel/lib/liblsdslam.so : référence indéfinie vers « cv::resizeWindow(cv::String const&, int, int) »
/home/masset/lsd_ws/devel/lib/liblsdslam.so : référence indéfinie vers « cv::imshow(cv::String const&, cv::_InputArray const&) »
collect2: error: ld returned 1 exit status
make[2]: *** [/home/masset/lsd_ws/devel/lib/lsd_slam_core/dataset] Erreur 1
make[1]: *** [lsd_slam/lsd_slam_core/CMakeFiles/dataset.dir/all] Erreur 2
In file included from /usr/local/include/opencv2/opencv.hpp:86:0,
                 from /home/masset/lsd_ws/src/lsd_slam/lsd_slam_viewer/src/KeyFrameDisplay.cpp:31:
/usr/local/include/opencv2/stitching.hpp:58:4: warning: #warning Detected X11 'Status' macro definition, it can cause build conflicts. Please, include this header before any X11 headers. [-Wcpp]
 #  warning Detected X11 'Status' macro definition, it can cause build conflicts. Please, include this header before any X11 headers.
    ^
In file included from /usr/include/GL/glx.h:30:0,
                 from /home/masset/lsd_ws/src/lsd_slam/lsd_slam_viewer/src/KeyFrameDisplay.cpp:27:
/usr/local/include/opencv2/stitching.hpp:128:10: error: expected identifier before ‘int’
     enum Status
          ^
In file included from /usr/local/include/opencv2/opencv.hpp:86:0,
                 from /home/masset/lsd_ws/src/lsd_slam/lsd_slam_viewer/src/KeyFrameDisplay.cpp:31:
/usr/local/include/opencv2/stitching.hpp:129:5: error: expected unqualified-id before ‘{’ token
     {
     ^
/home/masset/lsd_ws/src/lsd_slam/lsd_slam_viewer/src/KeyFrameDisplay.cpp: In member function ‘void KeyFrameDisplay::setFrom(lsd_slam_viewer::keyframeMsgConstPtr)’:
/home/masset/lsd_ws/src/lsd_slam/lsd_slam_viewer/src/KeyFrameDisplay.cpp:94:106: warning: format ‘%u’ expects argument of type ‘unsigned int’, but argument 3 has type ‘long unsigned int’ [-Wformat=]
      msg->pointcloud.size(), sizeof(InputPointDense), width, height, width*height*sizeof(InputPointDense));
                                                                                                          ^
/home/masset/lsd_ws/src/lsd_slam/lsd_slam_viewer/src/KeyFrameDisplay.cpp:94:106: warning: format ‘%u’ expects argument of type ‘unsigned int’, but argument 6 has type ‘long unsigned int’ [-Wformat=]
make[2]: *** [lsd_slam/lsd_slam_viewer/CMakeFiles/viewer.dir/src/KeyFrameDisplay.o] Erreur 1
make[1]: *** [lsd_slam/lsd_slam_viewer/CMakeFiles/viewer.dir/all] Erreur 2
make: *** [all] Erreur 2
Invoking "make -j8 -l8" failed

pierre-alexandre-masset, 2016-12-26:
# ** SOLUTION**

I deleted all Eigen versions from my computer (with apt-get and sudo make uninstall with all versions).
I have then installed EIgen2.3.5 here http://eigen.tuxfamily.org/index.php?title=News:Eigen_3.2.5_released!

And it worked !

FirefoxMetzger, 2017-03-24:
I think I found the root of all evil. Essentially a naming conflict between openCV (Status defined in stitching.hpp) and xlib macro Status (defines stitching as well)
It is further elaborated in: [this openCV Issue](https://github.com/opencv/opencv/issues/7113)

As of yet I am not sure how to fix this within ROS, as I don't know how to 'update' kinetic's native opencv to the latest version.

Ankita-Singh-21, 2021-03-04:
> I think I found the root of all evil. Essentially a naming conflict between openCV (Status defined in stitching.hpp) and xlib macro Status (defines stitching as well)
> It is further elaborated in: [this openCV Issue](https://github.com/opencv/opencv/issues/7113)
> 
> As of yet I am not sure how to fix this permanently, asside from updating openCV to the latest version (which would then have to be integrated into ROS).
> 
> Edit: As a quick and dirty fix comment
> 
> ```
> //#include "opencv2/opencv.hpp"
> ```
> 
> in `lsd_slam_viewer/src/KeyFrameDisplay.cpp:31` it gets included from some other file already, thus rosmake knows about OpenCV already (as it does with xlib, which is what causes this in the first place)

Thanks a lot, it works