tum-vision/lsd_slam #238: Launching lsd_slam : double free or corruption (out) ERROR

pierre-alexandre-masset, 2016-12-23:
Hi,

I have just installed lsd_slam and when I try to run it with
rosrun lsd_slam_core live_slam /image:=/bebop/image_raw /camera_info:=/bebop/camera_info

It appears that I have : 
*** Error in `/home/masset/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/bin/live_slam': double free or corruption (out): 0x0000000001589ae0 ***
Abandon (core dumped)

Anybody knows what is wrong ?

pierre-alexandre-masset, 2016-12-26:
# ** SOLUTION**

I deleted all Eigen versions from my computer (with apt-get and sudo make uninstall with all versions).
I have then installed EIgen2.3.5 here http://eigen.tuxfamily.org/index.php?title=News:Eigen_3.2.5_released!

And it worked !

DL-ljw, 2018-01-25:
I can not find 3.2.5, so I install 3.2.6. And it still has the same problem. Can you tell me how to download the 3.2.5 version. Do I need to remove the old version before install the 3.2.5?

bespoke-code, 2018-01-25:
You can 'install' (or more specifically, extract) Eigen 3.2.5 by downloading the source code for v.3.2.5 from [the official Eigen repository](https://bitbucket.org/eigen/eigen/downloads/?tab=tags). As previous answers by @pierre-alexandre-masset suggest, you probably need to remove the previous version to make it work.

lingxianv, 2018-01-26:
How to remove the previous version ?

bespoke-code, 2018-01-26:
@lingxianv , that depends on what system you're running. A possible answer on linking your code against the downloaded header files is present [in this thread](https://askubuntu.com/questions/860207/how-to-install-eigen-3-3-in-ubuntu-14-04).