tum-vision/lsd_slam #240: Compilation failure - Ubuntu 16.04 - ROS Kinetic

Kostech, 2017-01-20:
Hi guys,

Here I am uploading my build_log file. 
Has anyone managed to successfully run LSD_SLAM on Ubuntu 16.04, ROS Kinetic?
Have you faced the same situation?

[build_log.txt](https://github.com/tum-vision/lsd_slam/files/719518/build_log.txt)

alejandronespereira, 2017-01-20:
What a coincidence, I just found this software today and I am installing it on a Ubuntu 16.04 (it's a VM tho) with ROS Kinect. Totally new to ROS here. 

The fix from https://github.com/tum-vision/lsd_slam/issues/163#issuecomment-262445203 definitelly solved the build process for me, but now I am struggling with usage. 

Hope it helps!

Kostech, 2017-01-20:
@alejandronespereira Thanks man, it really solved my compilation problem too. I will see now how it works :)

Bell9, 2017-01-22:
I am having issues building LSD SLAM with ROS Kinetic. The fix mentioned above solves some errors, but I'm getting a lot more.

Could you tell me what steps you took to build the package and also what dependencies I would need?

Also, are you using catkin?

Thanks!

alejandronespereira, 2017-01-23:
I do not know about @Kostech, I just followed the instructions having ROS Kinetic built instead of the proposed ones. 

I'll do it again on another computer and let you know. I gues I am using catkin, as it's a ROS package present on my computer, but I don't recall installing it.

alejandronespereira, 2017-01-27:
Hi @Kostech, 
I managed to install it on another computer with a fresh ubuntu 16.04. However, I wasn't able to run it properly. I am sorry :(

Kostech, 2017-01-27:
@alejandronespereira Me neither. It seems there is no way to make it work on Ubuntu 16.04 :(

alejandronespereira, 2017-01-27:
If it helps, I changed to Ubuntu 14.04 and worked like a charm

lucaarcteam, 2017-03-03:
Hi,
I am also having problem. I am tryig to compile it in Debian Jessi, with Ros Kinetic via catkin. This fix did not work for me...

Young532042725, 2018-01-09:
@Kostech  Hello man, I have a question, Is there any problem that LSD-SLAM running on the Ubuntu16.04 + Kinetic ? Or even whether it cannot run on Ubuntu 16.04 +Kinetic or not ?