tum-vision/lsd_slam #241: Use without debug window

Haeryangkim, 2017-02-01:
How can I use it without the debug window ??
Is there any parameter for it? Or Should I change some code in this project?

apesIITM, 2017-04-11:
1. Open File: `lsd_slam_core/src/SlamSystem.cpp`
Line 690
`		Util::displayImage( "DebugWindow DEPTH", map->debugImageDepth, false );`
Comment it
2. Open File: `lsd_slam_core/src/Tracking/Relocalizer.cpp`
Line 89
`		Util::displayImage( "DebugWindow DEPTH", cv::Mat(currentFrame->height(), currentFrame->width(), CV_32F, currentFrame->image())*(1/255.0f), false );`
Comment it
I haven't tested, but I believe it should work.

infoagee, 2017-07-21:
@Haeryangkim yes, it works. but what's the difference with setting displayDepthImage to false in util/settings.cpp?