tum-vision/lsd_slam #246: Which is the best SLAM algorithm?

apesIITM, 2017-03-11:
Hello,
I have tried running LSD-SLAM and RTAB-SLAM, and currently reading about few more papers. Can you point out in what ways is LSD-SLAM better. (As per my experiments, RTAB-SLAM gave better map representation) [ No offence intended to anyone :) ]
Thanks.

Dhivin, 2017-03-11:
I am currently looking in moncular slam techniques? Can any suggest me the best monocular slam method?

proepkes, 2017-03-30:
ElasticFusion is also pretty good

mihaibujanca, 2017-03-30:
Elastic fusion is not monocular though

mihaibujanca, 2017-03-30:
The answer is, of course, it depends. What are you trying to achieve? Techincally SLAM - Simultaneous localisation and mapping has two parts to it:
The localisation - how well the system manages to localise itself relative to the space it's in - and subsequently relocalise in an already visited region.

The mapping - how accurately it can construct a map of the environment. The form this map takes is can massively vary from reconstructing the trajectory of the device (odometry map) to performing dense 3D reconstruction.