tum-vision/lsd_slam #248: DebugWindow DEPTH [Not Responding]

apesIITM, 2017-03-17:
Hello All,
I have compiled LSD-SLAM successfully on few systems, and with catkin as well. But after making some changes I tried to run it, and since it was not running I reinstalled ros-indigo and removed catkin_ws and rosbuild_ws, re-created rosbuild_ws and followed the same steps. There were no errors in compilation, but _no map is being generated in viewer_ and "**DebugWindow DEPTH**" is _not reponding_ with no image. 
Can you point the possible cause for such behaviour?
Thanks.
System Info:
ROS: indigo
Ubuntu: 14.04 (64-bit)
Opencv: 2.4.8 (with non-free also installed)
OpenGL support: available

=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
Console Output:

ajay@alpha24:~/rosbuild_ws/package_dir$ rosrun lsd_slam_viewer viewer  & 
[3] 28010
ajay@alpha24:~/rosbuild_ws/package_dir$ Started QApplication thread
Started ROS thread
[ INFO] [1489786144.229567227]: lsd_slam_viewer started

ajay@alpha24:~/rosbuild_ws/package_dir$ rosrun lsd_slam_core live_slam image:=/image_raw camera_info:=/camera_info & 
[4] 28101
ajay@alpha24:~/rosbuild_ws/package_dir$ [ WARN] [1489786160.168992731]: Shutdown request received.
[ WARN] [1489786160.169043023]: Reason given for shutdown: [new node registered with same name]
WAITING for ROS camera calibration!

[2]   Done                    rosrun lsd_slam_core live_slam image:=/image_raw camera_info:=/camera_info
ajay@alpha24:~/rosbuild_ws/package_dir$ 
ajay@alpha24:~/rosbuild_ws/package_dir$ rosbag play ~/datasets/DDP/SLAM/LSD-SLAM-dataset/mono/LSD_foodcourt.bag 
[ INFO] [1489786176.603981357]: Opening /home/ajay/datasets/DDP/SLAM/LSD-SLAM-dataset/mono/LSD_foodcourt.bag

Waiting 0.2 seconds after advertising topics... done.

Hit space to toggle paused, or 's' to step.
Received ROS Camera Calibration: fx: 257.164673, fy: 373.116455, cx: 290.454681, cy: 249.251343 @ 640x480
RECEIVED ROS camera calibration!
Started mapping thread!409581246.559882   Duration: 0.076515 / 709.215081     
Doing Random initialization!
Started  constraint search thread!
Started optimization thread 
started image display thread!246.561805   Duration: 0.078437 / 709.215081     
Done Random initialization!
init done   Bag Time: 1409581246.763473   Duration: 0.280106 / 709.215081     
opengl support available 
TRACKING LOST for frame 178 (0.02% good Points, which is 80.95% of available points, DIVERGED)!
Shutting down... ime: 1409581268.801458   Duration: 22.318091 / 709.215081     
Exiting ROS threadme: 1409581268.961938   Duration: 22.478571 / 709.215081     
[3]-  Exit 1                  rosrun lsd_slam_viewer viewer13 / 709.215081 
=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=
PFA Image while running:
[image]

apesIITM, 2017-03-18:
Hello @JakobEngel @icoderaven 
LSD-SLAM is compiled and run successfully. But after I make some changes and recompile, it gives above output. Can you please suggest what might be wrong.
Note: Once this error appears, I am unable to compile LSD-SLAM even from scratch. I tried using ~/catkin_ws and ~/rosbuild_ws , and deleted ~/.ros folder as well.

apesIITM, 2017-03-25:
BUG Fixed!
I recompiled OpenCV with QT OFF, and it worked!

closed 2017-03-25 by apesIITM

Gyiming, 2017-07-21:
Hi apesllTM, when I tried to recompiled OpenCV with QT OFF, another bug showed:
QXcbConnection: Could not connect to display.

How do you recompiled OpenCV with QT OFF? Maybe I did something wrong on that. What I did is when doint cmake, I did CMAKE -D WITH_QT=off

alexis-dupuis, 2018-06-08:
Irenedvs' fixed worked for me too on Ubuntu 16.04 with ros kinetic and a catkin build environment. Thanks for saving my life on this one!

shahram95, 2019-08-22:
Hi, I am facing the same issue. Could you guys tell me how to compile OPENCV with QT turned off. I am a newbie to all this.

WANGLXUE, 2019-09-18:
@shahram95 It is easy to do with cmake-gui. Before generate, there is a checkbox for QT. Just make sure it is off.

RENO216, 2019-11-25:
@alexis-dupuis  Hi, may I know the catkin version of lsd_slam you were using? I use Ubuntu 16.04 LTS and ROS kinetic but Irenedvs' solution doesn't work for me. Thanks!

Kanishk598, 2020-07-14:
> Hi, I am facing the same issue. Could you guys tell me how to compile OPENCV with QT turned off. I am a newbie to all this.

Hi @shahram95 , compilation of a catkin workspace requires the command 'catkin_make'. Like other Linux commands, you can pass parameters with this command. One such parameter to compile the workspace with QT off is also there and to pass it with the command catkin_make, you need to write in the terminal 'catkin_make -D WITH_QT=off'. 
If you want, you can see the catkin documentation [here](https://docs.ros.org/api/catkin/html/)