tum-vision/lsd_slam #249: rosmake error in building package

TheKarex, 2017-03-24:
I had already installed and ran the lsd slam, it worked perfectly without any error. But when i tried to do a clean install again i am getting the following error. Could anyone help me ????

[ rosmake ] rosmake starting...                                                 
[ rosmake ] Packages requested are: ['lsd_slam']                                
[ rosmake ] Logging to directory /home/karthi/.ros/rosmake/rosmake_output-20170324-134214
[ rosmake ] Expanded args ['lsd_slam'] to:
['lsd_slam_viewer', 'lsd_slam_core'] 
[rosmake-0] Starting >>> catkin [ make ]                                        
[rosmake-0] Finished <<< catkin ROS_NOBUILD in package catkin
 No Makefile in package catkin
[rosmake-0] Starting >>> genmsg [ make ]                                        
[rosmake-1] Starting >>> cpp_common [ make ]                                    
[rosmake-3] Starting >>> cmake_modules [ make ]                                 
[rosmake-0] Finished <<< genmsg ROS_NOBUILD in package genmsg
 No Makefile in package genmsg
[rosmake-0] Starting >>> genlisp [ make ]                                       
[rosmake-2] Starting >>> genpy [ make ]                                         
[rosmake-3] Finished <<< cmake_modules ROS_NOBUILD in package cmake_modules
 No Makefile in package cmake_modules
[rosmake-3] Starting >>> gencpp [ make ]                                        
[rosmake-1] Finished <<< cpp_common ROS_NOBUILD in package cpp_common Complete ]
 No Makefile in package cpp_common
[rosmake-0] Finished <<< genlisp ROS_NOBUILD in package genlisp
 No Makefile in package genlisp
[rosmake-2] Finished <<< genpy ROS_NOBUILD in package genpy
 No Makefile in package genpy
[rosmake-0] Starting >>> rospack [ make ]                                       
[rosmake-1] Starting >>> rostime [ make ]                                       
[rosmake-2] Starting >>> rosgraph [ make ]                                      
[rosmake-3] Finished <<< gencpp ROS_NOBUILD in package gencpp
 No Makefile in package gencpp
[rosmake-3] Starting >>> message_generation [ make ]                            
[rosmake-2] Finished <<< rosgraph ROS_NOBUILD in package rosgraph
 No Makefile in package rosgraph
[rosmake-2] Starting >>> rosparam [ make ]                                      
[rosmake-1] Finished <<< rostime ROS_NOBUILD in package rostime
 No Makefile in package rostime
[rosmake-1] Starting >>> roscpp_traits [ make ]                                 
[rosmake-0] Finished <<< rospack ROS_NOBUILD in package rospack
 No Makefile in package rospack
[rosmake-3] Finished <<< message_generation ROS_NOBUILD in package message_generation
 No Makefile in package message_generation
[rosmake-0] Starting >>> roslib [ make ]                                        
[rosmake-3] Starting >>> rosmaster [ make ]                                     
[rosmake-2] Finished <<< rosparam ROS_NOBUILD in package rosparam
 No Makefile in package rosparam
[rosmake-2] Starting >>> roslang [ make ]                                       
[rosmake-1] Finished <<< roscpp_traits ROS_NOBUILD in package roscpp_traits
 No Makefile in package roscpp_traits
[rosmake-1] Starting >>> roscpp_serialization [ make ]                          
[rosmake-0] Finished <<< roslib ROS_NOBUILD in package roslib
 No Makefile in package roslib
[rosmake-3] Finished <<< rosmaster ROS_NOBUILD in package rosmaster
 No Makefile in package rosmaster
[rosmake-2] Finished <<< roslang ROS_NOBUILD in package roslang
 No Makefile in package roslang
[rosmake-0] Starting >>> rosunit [ make ]                                       
[rosmake-3] Starting >>> xmlrpcpp [ make ]                                      
[rosmake-2] Starting >>> rosclean [ make ]                                      
[rosmake-1] Finished <<< roscpp_serialization ROS_NOBUILD in package roscpp_serialization
 No Makefile in package roscpp_serialization
[rosmake-1] Starting >>> message_runtime [ make ]                               
[rosmake-2] Finished <<< rosclean ROS_NOBUILD in package rosclean
 No Makefile in package rosclean
[rosmake-3] Finished <<< xmlrpcpp ROS_NOBUILD in package xmlrpcpp
 No Makefile in package xmlrpcpp
[rosmake-0] Finished <<< rosunit ROS_NOBUILD in package rosunit
 No Makefile in package rosunit
[rosmake-0] Starting >>> roslz4 [ make ]                                        
[rosmake-1] Finished <<< message_runtime ROS_NOBUILD in package message_runtime
[... 88 lines cut ...]
  /opt/ros/indigo/include/sophus/sim3.hpp: In instantiation of ‘void Sophus::Sim3GroupBase<Derived>::setScale(const Scalar&) const [with Derived = Sophus::Sim3Group<double>; Sophus::Sim3GroupBase<Derived>::Scalar = double]’:
  /home/karthi/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/src/util/SophusUtil.h:56:23:   required from here
  /opt/ros/indigo/include/sophus/sim3.hpp:339:5: error: passing ‘const RxSO3Type {aka const Sophus::RxSO3Group<double>}’ as ‘this’ argument of ‘void Sophus::RxSO3GroupBase<Derived>::setScale(const Scalar&) [with Derived = Sophus::RxSO3Group<double>; Sophus::RxSO3GroupBase<Derived>::Scalar = double]’ discards qualifiers [-fpermissive]
       rxso3().setScale(scale);
       ^
  make[3]: *** [CMakeFiles/lsdslam.dir/src/DataStructures/FramePoseStruct.cpp.o] Error 1
  make[3]: *** Waiting for unfinished jobs....
  make[3]: *** [CMakeFiles/lsdslam.dir/src/DataStructures/Frame.cpp.o] Error 1
  make[3]: *** [CMakeFiles/lsdslam.dir/src/DataStructures/FrameMemory.cpp.o] Error 1
  make[3]: *** [CMakeFiles/lsdslam.dir/src/SlamSystem.cpp.o] Error 1
  make[3]: Leaving directory `/home/karthi/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
  make[2]: *** [CMakeFiles/lsdslam.dir/all] Error 2
  make[2]: Leaving directory `/home/karthi/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
  make[1]: *** [all] Error 2
  make[1]: Leaving directory `/home/karthi/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
-------------------------------------------------------------------------------}
[ rosmake ] Output from build of package lsd_slam_core written to:
[ rosmake ]    /home/karthi/.ros/rosmake/rosmake_output-20170324-134214/lsd_slam_core/build_output.log
[rosmake-0] Finished <<< lsd_slam_core [FAIL] [ 55.79 seconds ]                                                                                 
[ rosmake ] Halting due to failure in package lsd_slam_core. 
[ rosmake ] Waiting for other threads to complete.                                
[ rosmake ] Results:                                                                                                                            
[ rosmake ] Built 42 packages with 1 failures.                                                                                                  
[ rosmake ] Summary output to directory                                                                                                         
[ rosmake ] /home/karthi/.ros/rosmake/rosmake_output-20170324-134214

vacky11, 2017-04-04:
I am getting the same error. @TheKarex  Were you able to fix it?

TheKarex, 2017-04-06:
@vacky11 no luck with that. I reinstalled Ubuntu and tried again, it works fine now. But no idea about the error.

vacky11, 2017-04-06:
@TheKarex Thanks for replying. 
I tried building it in catkin workspace by following the instructions given in this [link](http://visbot.blogspot.com/2014/11/tutorial-building-of-lsd-slam-on-ros.html) and it worked for me.

Hope it helps others who are not able to get rid of this error.

jreed-theneumachine, 2017-07-22:
Thanks @vacky11 I tried the rosbuild today and it failed but using the link you gave it worked.