tum-vision/lsd_slam #251: Adding Fabmap support for ROS-kinetic install

prarobo, 2017-03-24:
I was able to compile lsd_slam for ROS-kinetic in Ubuntu 16.04 without fabmap support. When I included fabmap, I got this error

>   [  8%] Building CXX object thirdparty/openFabMap/CMakeFiles/openFABMAP.dir/src/FabMap.cpp.o
  [ 14%] Built target ROSBUILD_gencfg_cpp
  make[3]: Entering directory '/home/prasanna/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
  make[3]: Leaving directory '/home/prasanna/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
  [ 14%] Built target rospack_gencfg
  make[3]: Entering directory '/home/prasanna/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
  make[3]: Entering directory '/home/prasanna/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
  In file included from /home/prasanna/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/thirdparty/openFabMap/src/ChowLiuTree.cpp:29:0:
  /home/prasanna/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/thirdparty/openFabMap/src/../include/openfabmap.hpp:38:30: fatal error: opencv2/opencv.hpp: No such file or directory
  compilation terminated.
  thirdparty/openFabMap/CMakeFiles/openFABMAP.dir/build.make:62: recipe for target 'thirdparty/openFabMap/CMakeFiles/openFABMAP.dir/src/ChowLiuTree.cpp.o' failed
  make[3]: *** [thirdparty/openFabMap/CMakeFiles/openFABMAP.dir/src/ChowLiuTree.cpp.o] Error 1
  make[3]: *** Waiting for unfinished jobs....
  make[3]: Leaving directory '/home/prasanna/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
  make[3]: Leaving directory '/home/prasanna/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
  In file included from /home/prasanna/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/thirdparty/openFabMap/src/BOWMSCTrainer.cpp:29:0:
  /home/prasanna/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/thirdparty/openFabMap/src/../include/openfabmap.hpp:38:30: fatal error: opencv2/opencv.hpp: No such file or directory
  compilation terminated.
  thirdparty/openFabMap/CMakeFiles/openFABMAP.dir/build.make:86: recipe for target 'thirdparty/openFabMap/CMakeFiles/openFABMAP.dir/src/BOWMSCTrainer.cpp.o' failed
  make[3]: *** [thirdparty/openFabMap/CMakeFiles/openFABMAP.dir/src/BOWMSCTrainer.cpp.o] Error 1
  [ 14%] Built target rospack_genmsg_libexe
  [ 14%] Built target rospack_gencfg_real
  make[3]: Entering directory '/home/prasanna/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
  In file included from /home/prasanna/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/thirdparty/openFabMap/src/FabMap.cpp:29:0:
  /home/prasanna/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/thirdparty/openFabMap/src/../include/openfabmap.hpp:38:30: fatal error: opencv2/opencv.hpp: No such file or directory
  compilation terminated.
  thirdparty/openFabMap/CMakeFiles/openFABMAP.dir/build.make:110: recipe for target 'thirdparty/openFabMap/CMakeFiles/openFABMAP.dir/src/FabMap.cpp.o' failed
  make[3]: *** [thirdparty/openFabMap/CMakeFiles/openFABMAP.dir/src/FabMap.cpp.o] Error 1
  make[3]: Leaving directory '/home/prasanna/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
  CMakeFiles/Makefile2:1032: recipe for target 'thirdparty/openFabMap/CMakeFiles/openFABMAP.dir/all' failed
  make[2]: *** [thirdparty/openFabMap/CMakeFiles/openFABMAP.dir/all] Error 2
  make[2]: *** Waiting for unfinished jobs....
  make[3]: Leaving directory '/home/prasanna/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
  [ 14%] Built target rosbuild_precompile
  make[2]: Leaving directory '/home/prasanna/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
  Makefile:127: recipe for target 'all' failed
  make[1]: *** [all] Error 2
  make[1]: Leaving directory '/home/prasanna/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/build'
  /opt/ros/kinetic/share/mk/cmake.mk:7: recipe for target 'all' failed

It appears like it is not able to find openCV, however the lsd_slam package compile without fabmap despite using opencv. All opencv includes in the CMakeLists.txt look fine. Any thoughts?

romdos, 2017-04-06:
I have another error related on FabMap, which is about nonfree usage of this library.

FirefoxMetzger, 2017-04-11:
The issue is, that openFabMap uses SURF and Star features. Both are part of the non-free libraries of openCV. Since opencv3 they have been moved into another repo ( opencv-contrib ). Also their includes and structure has been modified, thus you have to change some code around to make it work.

Further `/thirdparty/openFabMap` supplies its own `FindOpenCV.cmake` which will then find different libraries than LSD-SLAM (which gets them from ROS)

Here is how I've build mine. Make sure you have ros-kinetic-opencv3 installed. It also provides the non-free libraries.
In `lsd_slam_core/thirdparty/openFabMap`

change  FindOpenCV.cmake:19 to:

    FIND_PATH( OPENCV2_INCLUDE_PATH opencv.hpp
	/opt/ros/kinetic/include/opencv-3.2.0-dev/opencv2
	)

-- that should fix @prarobo 's issue, that the headers can't be found

in CMakeLists.txt:64 add

    # should not be needed, if ROS is sourced libraries are added to $LD_LIBRARY_PATH
    link_directories(/opt/ros/kinetic/lib)

in CMakeLists.txt:64 change to

    TARGET_LINK_LIBRARIES(openFABMAP opencv_highgui opencv_core opencv_features2d opencv_imgproc opencv_xfeatures2d3)

compare `samples\openFABMAPcli.cpp` with [my version](https://github.com/FirefoxMetzger/openfabmap/blob/master/samples/openFABMAPcli.cpp). There are a few changes to comply with the changed syntax from opencv2.4 -> opencv3.2

in `src\ChowLiuTree.cpp:55` add:

    #include <map>

now you *should* be able to compile openFabMap with openCV3.2, however you still have to adjust LSD-slam to work with that.

Go to `lsd_slam_core/src/GlobalMapping`

In `FabMap.h:24` add

    #include <opencv2/features2d.hpp>

in `FabMap.h:29` remove

        // namespace cv {
	// 	class FeatureDetector;
	// 	class BOWImgDescriptorExtractor;
	// }

In `FabMap.cpp:25` change to:

    #include <opencv2/features2d.hpp>
    #include <opencv2/xfeatures2d.hpp>

In `FamMap.cpp:95` change to:

    detector = cv::xfeatures2d::StarDetector::create(32, 10, 18, 18, 20);
    cv::Ptr<cv::DescriptorExtractor> extractor = cv::xfeatures2d::SURF::create(1000, 4, 2, false, true); // new cv::SIFT();

If I didn't miss any change I did it should be compiling now. Hope that helps.

abdur4373, 2019-02-20:
Hello @FirefoxMetzger  @prarobo 

After making the changes according to @FirefoxMetzger  comment i was getting this error:- 
`/usr/bin/ld: cannot find -lopencv_xfeatures2d3
collect2: error: ld returned 1 exit status
src/lsd_slam/lsd_slam_core/thirdparty/openFabMap/CMakeFiles/openFABMAP.dir/build.make:153: recipe for target '/home/maq/catkin_ws/devel/lib/libopenFABMAP.so' failed
make[2]: *** [/home/maq/catkin_ws/devel/lib/libopenFABMAP.so] Error 1
CMakeFiles/Makefile2:2502: recipe for target 'src/lsd_slam/lsd_slam_core/thirdparty/openFabMap/CMakeFiles/openFABMAP.dir/all' failed
make[1]: *** [src/lsd_slam/lsd_slam_core/thirdparty/openFabMap/CMakeFiles/openFABMAP.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 30%] Built target lsd_slam_viewer_generate_messages
[ 41%] Built target viewer
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j16 -l16" failed
`

But when i removed 'opencv_xfeatures2d3' from:-
`TARGET_LINK_LIBRARIES(openFABMAP opencv_highgui opencv_core opencv_features2d opencv_imgproc opencv_xfeatures2d3)` in CMakeLists.txt:64. catkin_make was good and LSD SLAM is running as well.

I have cloned LSD SLAM from this [respository](https://github.com/kevin-george/lsd_slam). Kindly comment what was the use of 'opencv_xfeatures2d3' library and how to know that loop closure is running with openFabMap or not. As result of PointCloudViewer is not much different from before when i was running it without loop closures.