tum-vision/lsd_slam #252: Compiling lsd_slam_viewer under ubuntu 16.04

bdludwig, 2017-03-31:
Hi,

I have installed ros kinetic. and libqglviewer-dev-qt4.

When I run rosmake lsd_slam, I get the following error:

/opt/ros/kinetic/share/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.h: In member function ‘std::__cxx11::string AnimationObject::toString()’:
/opt/ros/kinetic/share/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.h:136:26: e
rror: invalid initialization of non-const reference of type ‘qreal& {aka double&
}’ from an rvalue of type ‘qreal {aka double}’
   frame.getPosition(x,y,z);
                          ^
In file included from /usr/include/QGLViewer/keyFrameInterpolator.h:31:0,
                 from /usr/include/QGLViewer/camera.h:26,
                 from /usr/include/QGLViewer/qglviewer.h:26,
                 from /opt/ros/kinetic/share/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.h:26,
                 from /opt/ros/kinetic/share/lsd_slam/lsd_slam_viewer/src/main_viewer.cpp:25:

Does anybody know if there is an update of the PointCloudViewer or which other qglviewer lib I have to install?

Thanks 
Bernd

romdos, 2017-04-06:
I recommend you to install it on Ubuntu 14.04. Even if you could install it sucessfully on 16.04, there will be errors with its running related on different libqglviewer versions.

extr15, 2017-04-11:
you may solve this by editing this file: /lsd_slam_viewer/src/PointCloudViewer.h:136, modify "float x, y, z" to "double x,y,z" and try rosmake.

Young532042725, 2018-01-20:
@bdludwig @extr15 ,  Hi, my friends,  Did you solved the problem?  I also confront with it, now. may you give me some advice? thank u very much!!!!

MrMinemeet, 2019-07-12:
@extr15 solution helped for me