tum-vision/lsd_slam #255: IMU with LSD_SLAM

apesIITM, 2017-04-11:
Hello,
I am trying to fuse IMU with LSD_SLAM.
I am unable to get `pose` values from LSD_SLAM and modify it in `lsd_slam_core` after fusing it from IMU sensors.
Any hint towards this  will be really helpful.
Thanks

ObiKwiet, 2017-06-28:
You can't do that. The pose values from LSD_SLAM are not in any consistent real world scale. You have to correlate them with real world sensor values first, and propagate the covariance of that data before you can fuse it in an EKF. I'm working on a method to do this, but as yet I don't really understand the lsd-slam algorithm well enough to use the internal pose covariance matrix. 

You can get the raw pose data from the /lsd_slam/pose topic.

vkpankov, 2018-03-13:
@ObiKwiet, Hello! Is it possible to fuse IMU with /lsd_slam/pose to estimate real world scale?

ObiKwiet, 2018-03-15:
Yup. I've attached my thesis on the subject. Now, what I did is more of a proof of concept for a more niche application, and probably not the best approach if you just want to get something working. For that I'd find an IMU with an internal gravity estimator filter that's been finessed by some company into working really well off the shelf. 

[Tucker Thesis Draft 2 rev C.pdf](https://github.com/tum-vision/lsd_slam/files/1816366/Tucker.Thesis.Draft.2.rev.C.pdf)

vkpankov, 2018-03-16:
@ObiKwiet, 
Thank you so much! wonderful job, what I've been looking for so long:-)
Is it possible to get the source code?  I would be very grateful.
For use in the bachelor thesis, of course with reference
vkpankov@gmail.com

vkpankov, 2018-03-22:
@ObiKwiet, I can pay for it ($$). Or get the source is impossible in any case?
I try to implement on my own, but it turns out with difficulty

ObiKwiet, 2018-03-23:
Sorry, I've been dealing with some crap and forgot about this thread. I'm out of town now, so shoot me an email Monday or Tuesday if I forget.  seth.c.tucker@gmail.com

I'll shoot you the code, but unfortunately it really sucks. I'm not really a software guy, so it works more or less with my solution. And like I say, my exact approach probably won't get you where you want. You probably want a standard Kalman Filter fusion with an IMU that has had all the kinks ironed out.  But my code should at least give you and idea of how to get pose info out of lsd_slam. Let me know what you come up with!

-Seth

vkpankov, 2018-03-29:
@ObiKwiet, 
I understand that the code may be too specific. But I'm sure it will be very useful in preprocessing LSD SLAM results

DigitalGabriele, 2019-02-22:
@vkpankov have you managed to fuse IMU data with LSD_SLAM?