tum-vision/lsd_slam #258: debian/minibian support

antonays, 2017-04-18:
I read that the system was tested on ubuntu 12.04 and ubuntu 14.04, and also for me, compiling with ros-indigo on ubuntu 14.04 works perfectly with no dependency issues.
My question would be, how difficult would it be to make the system work on a debian platform?, specifically, on a minibian deployed on a raspberry pi?
I did try, with the help of some other issues ([1](https://github.com/tum-vision/lsd_slam/issues/202), [2](https://github.com/tum-vision/lsd_slam/issues/237) and others ), had many dependency issues and was not able to solve them all.

FirefoxMetzger, 2017-04-18:
Definitly doable. The amount of time you are able to put in is the only limiting factor.

I've got mine to work on 16.04 with OpenCV3 and Kinetic. Took me about 3 weeks though to patch everything...
What kind of issues are you getting?

antonays, 2017-04-19:
Thanks @FirefoxMetzger for your reply. 

I currently abandoned the efforts to make the system work on the rPi, and turned to a distributed approach where the rPi is a client transmitting a topic of visual data to an ubuntu host on which the lsd_slam is running, it's fine and a nice cloud based approach, but i have many latency and QoS issues.

The question was more a theoretical one, trying to figure out if someone had success with debian.

Anyway, the point being, currently i don't have the specific errors that i encountered when trying to compile.
There were mostly dependency issues when compiling with catkin, fixing the immediate ones by installing the additional libraries and sourcing them to the environment did fix it, but created additional dependency problems.

I will dedicate some time to try again in the next couple of weeks and re-post here.