tum-vision/lsd_slam #259: TRACKING LOST error while using dataset SLAM

vacky11, 2017-04-18:
I am using Iphone 6s camera to take a video and then extract each frame fro the video to form the dataset. My calibration file is as follows.

0.8609 1.1474 0.4999 0.4966 0.062
1920 1080
crop
640 480

After running the following coomand:
rosrun lsd_slam_core dataset _files:=/home/vk/Downloads/my_data/images2 _hz:=30 _calib:=/home/vk/catkin_ws/src/lsd_slam/lsd_slam_core/calib/iphone_6s_calib.cfg

I get the following error:
Reading Calibration from file /home/vk/catkin_ws/src/lsd_slam/lsd_slam_core/calib/iphone_6s_calib.cfg ... found!
found ATAN camera model, building rectifier.
Input resolution: 1920 1080
In: 0.860900 1.147400 0.499900 0.496600 0.062000
Out: Crop
Output resolution: 640 480
new K: 550.914551 550.794739 319.769287 238.145020
old K: 1652.927979 1239.192017 959.308044 535.828003
Prepped Warp matrices
Started mapping thread!
Started  constraint search thread!
Started optimization thread 
found 2131 image files in folder /home/vk/Downloads/my_table/images2!
Doing Random initialization!
started image display thread!
Done Random initialization!
warning: reciprocal tracking on new frame failed badly, added odometry edge (Hacky).
warning: reciprocal tracking on new frame failed badly, added odometry edge (Hacky).
TRACKING LOST for frame 302 (5.27% good Points, which is 29.07% of available points, NOT DIVERGED)!

Can someone please tell me where am I going wrong?

Thanks!

FirefoxMetzger, 2017-04-18:
It's hard to tell from this.
TRACKING LOST indecates, that there aren't enough keypoints visible in the active frame (only ~5%). Meaning that LSD-SLAM can't tell where the camera is with respect to the active keyframe.
Usually this should start the relocalizer, but it keeps failing for me, too.

You could try minimizing running into this issue by being even more mindfull of rotations and translations, thats at least what worked for me.

vacky11, 2017-04-18:
@FirefoxMetzger  
Thanks a lot for replying. [Here](https://youtu.be/PxUEOzEhGoo) is my test video. Can you please suggest me any improvements on it?

sinha-kirti, 2017-04-19:
Hi, I believe the problem is due to  the translation motion in initial part
of the video. I was facing the same problem because lsd-slam uses rotation
for initial approximations. Try with video having only rotational motion in
the beginning of the video for about 30 -40 frames. Also, you can try with
prerectified images instead of directly feeding the images captured in
video.

>
> @FirefoxMetzger
> Thanks a lot for replying. Here is my test video. Can you please suggest
me any improvements on it?

FirefoxMetzger, 2017-04-19:
I don't see anything obviously wrong with it.
Assuming 30Hz you diverge around 10 sec, which mainly shows the laptop.

While the laptop has a pretty slick design, LSD-SLAM works on image gradients (corners and edges). Looking at it from that perspective, the laptop might be a poor choice as there is no potential for keypoints on the display and the keyboard area is pretty smooth as well (esp. at 640 x 480). You could try changing the object or show some image on the screen to give LSD-SLAM more change to find keypoints on it.

Also you are doing offline SLAM. Try setting the framerate option to 0 Hz and increase the output resolution. This will not be real time anymore, but should give you more information about potential problems,