tum-vision/lsd_slam #265: EuRoC dataset execution and evaluation

Toumi0812, 2017-07-11:
Hi everyone, 
Do you know how to set up the configuration file for EuRoC dataset sequences for LSD_SLAM?
and how to compare results with the ground truth?

Thanks

MichaelGrupp, 2017-10-30:
I didn't run it on EuRoC yet, but the easiest way is probably to play the ROS bag files from their [website](http://projects.asl.ethz.ch/datasets/doku.php?id=kmavvisualinertialdatasets). Then you need to give the correct ROS topic names as parameters to the LSD SLAM node together with a calibration parameter file for rectification, as written in the README.

Maybe useful for you: I wrote an [evaluation tool](https://github.com/raulmur/ORB_SLAM2/issues/454) that supports different datasets, also EuRoC. With this, you can run metrics with the groundtruth .csv from the EuRoC dataset (`<seq>/mav0/state_groundtruth_estimate0/data.csv` in the folder version of the dataset) and the estimated trajectory from LSD SLAM in the TUM benchmark format. Keep in mind that you need to consider scale issues of monocular SLAM, as I mentioned [here](https://github.com/raulmur/ORB_SLAM2/issues/121).

jwangjie, 2019-01-21:
I was wondering has anyone successfully evaluated LSD SLAM with EuRoC dataset? Could you specify a little bit how you implement it? Thanks a lot.