tum-vision/lsd_slam #270: rosmake lsd_slam error

SherlockMoriaty, 2017-08-03:
I compile the lsd_slam by typing: rosmake lsd_slam. But I get some failures, which shows that "[ rosmake Built 41 packages with 1 failures]". Can you kindly help me solve the problem? Below are the whole messages.

[ rosmake ] Last 40 linesd_slam_viewer: 4.7 sec ]    [ 1 Active 40/42 Complete ]
{-------------------------------------------------------------------------------
  -- Looking for Q_WS_MAC
  -- Looking for Q_WS_MAC - not found
  -- Found Qt4: /usr/bin/qmake (found version "4.8.6") 
  -- Found QGLVIEWER: /usr/include/QGLViewer  
  MSG: gencfg_cpp on:LSDSLAMViewerParams.cfg
  ERROR [gendeps] 1 Finding dependencies for /root/rosbuild_ws/package_dir/lsd_slam-master/lsd_slam_viewer/cfg/LSDSLAMViewerParams.cfg
  ***********************************************************************************
  load_module did not return. Unable to determine dependencies for file listed above.
  ***********************************************************************************
  Traceback (most recent call last):
    File "/opt/ros/indigo/share/dynamic_reconfigure/cmake/gendeps", line 66, in <module>
      imp.load_module("__main__", f, srcfile, ('.cfg', 'U', 1))
    File "/root/rosbuild_ws/package_dir/lsd_slam-master/lsd_slam_viewer/cfg/LSDSLAMViewerParams.cfg", line 20, in <module>
      gen.add("scaledDepthVarTH", double_t, 0, "log10 of threshold on point's variance, in the respective keyframe's scale. ", -3, -10, 1)
    File "/opt/ros/indigo/lib/python2.7/dist-packages/dynamic_reconfigure/parameter_generator.py", line 273, in add
      self.group.add(name, paramtype, level, description, default, min, max, edit_method) 
    File "/opt/ros/indigo/lib/python2.7/dist-packages/dynamic_reconfigure/parameter_generator.py", line 143, in add
      check_description(description)
    File "/opt/ros/indigo/lib/python2.7/dist-packages/dynamic_reconfigure/parameter_generator.py", line 68, in check_description
      raise Exception(r"""quotes not allowed in description string `%s`""" % description)
  Exception: quotes not allowed in description string `log10 of threshold on point's variance, in the respective keyframe's scale. `
  
  CMake Error at /opt/ros/indigo/share/dynamic_reconfigure/cmake/cfgbuild.cmake:78 (string):
    string sub-command REPLACE requires at least four arguments.
  Call Stack (most recent call first):
    /opt/ros/indigo/share/dynamic_reconfigure/cmake/cfgbuild.cmake:99 (gencfg_cpp)
    CMakeLists.txt:29 (include)
  
  
  -- Performing Test HAS_SSE3_EXTENSIONS
  -- Performing Test HAS_SSE3_EXTENSIONS - Success
  -- Performing Test HAS_SSE2_EXTENSIONS
  -- Performing Test HAS_SSE2_EXTENSIONS - Success
  -- Performing Test HAS_SSE_EXTENSIONS
  -- Performing Test HAS_SSE_EXTENSIONS - Success
  -- [rosbuild] Found SSE3 extensions, using flags: -msse3 -mfpmath=sse
  -- Configuring incomplete, errors occurred!
  See also "/root/rosbuild_ws/package_dir/lsd_slam-master/lsd_slam_viewer/build/CMakeFiles/CMakeOutput.log".
  See also "/root/rosbuild_ws/package_dir/lsd_slam-master/lsd_slam_viewer/build/CMakeFiles/CMakeError.log".
-------------------------------------------------------------------------------}
[ rosmake ] Output from build of package lsd_slam_viewer written to:
[ rosmake ]    /root/.ros/rosmake/rosmake_output-20170802-101550/lsd_slam_viewer/build_output.log
[rosmake-6] Finished <<< lsd_slam_viewer [FAIL] [ 4.75 seconds ]                
[ rosmake ] Halting due to failure in package lsd_slam_viewer. 
[ rosmake ] Waiting for other threads to complete.
[ rosmake ] Results:                                                            
[ rosmake ] Built 41 packages with 1 failures.                                  
[ rosmake ] Summary output to directory

EnsekiTT, 2017-08-10:
I also experienced the same problem.
I solved the problem by replacing "'" in description string in cfg files with "`".

like this:
Edit `lsd_slam/lsd_slam_viewer/cfg/LSDSLAMViewerParams.cfg`
I rewrote line 20 
from
`"log10 of threshold on point's variance, in the respective keyframe's scale. "`
to
`"log10 of threshold on point`s variance, in the respective keyframe`s scale. "`
Make the same changes for lines 21 and 30.

Edit `lsd_slam/lsd_slam_core/cfg/LSDDebugParams.cfg`
Make the same changes for lines 11, 12 and 48.

jonnydark, 2017-08-16:
There's a fix for this in pull request #263

Young532042725, 2018-01-20:
@SherlockMoriaty  Hi man, had you solved the problem? i also meet it . can you help me ?  thank u !  I do something according to @EnsekiTT ,but other problem occurs!  so desperate!!!

SherlockMoriaty, 2018-01-20:
Thank u for your reply and discussion, but I have started to do other projects, and the problem has not been solved.sorry.

bespoke-code, 2018-01-20:
@Young532042725 @SherlockMoriaty You can get around this problem by removing all the single quotation marks currently present in some of the strings in `LSDSLAMViewerParams.cfg` and any other .cfg file.

As an example, what I did to fix this was change this
`gen.add("scaledDepthVarTH", double_t, 0, "log10 of threshold on point's variance, in the respective keyframe's scale. ", -3, -10, 1)` 
to this:
`gen.add("scaledDepthVarTH", double_t, 0, "log10 of threshold on the variance of a point, in the respective scale of the keyframe. ", -3, -10, 1)`

This 'single quotation mark removal' was applied to all the strings in all files which caused errors at compile time. If you don't give a damn about grammar, just remove the single quotation marks from the strings and voila :grin: