tum-vision/lsd_slam #274: Working fork of LSD SLAM?

kevin-george, 2017-09-01:
Does anyone know of a working fork of this repo compatible with ROS & OpenCV?
The authors have obviously stopped working on/maintaining this. 

I would be willing to help out as much as I can if someone decides to do it :)

sathishkumark27, 2017-09-02:
Hi,

I have tried to work with it but could not able to make the setup, I could
use same help to work on it.

Thanks & Regards
Sathish

OAkyildiz, 2017-09-04:
Hello, there is a ros-kinetic fork, if you look at the timeline. I would link it but I am on mobile right now. I did not try it and the instructions are not updated, please let us know if you end up trying it.

kevin-george, 2017-09-12:
Hey @OAkyildiz, Unfortunately there were way too many ros-kinetic forks. I haven't been check them all out, please let me know if you can find it!

kevin-george, 2017-09-13:
So, after a fair amount of debugging and going through fixes in the pull requests I got it to work with Ubuntu 16.04 + ROS Kinetic + Latest version of G2O. If anyone wants to give it a try it's available at 

https://github.com/kevin-george/lsd_slam.

Most of the work was already done in [269](https://github.com/tum-vision/lsd_slam/pull/269) but the latest G2O required me to make other changes. I will write a brief guide on the wiki soon :)

Young532042725, 2018-01-20:
@kevin-george , Hi, my friend, I have run the LSD-slam with16.04 +ROS kinetic according to the instruction on Github (just the installation for Ubuntu 14.04+ROS indigo), but some error occurs! can you give me some advice?  so desperate!

bespoke-code, 2018-01-20:
@Young532042725 there's a discussion about this problem in Issue #270. Remove all single quotation marks from all strings in each of the problematic *.cfg* files and you should be good to go.

Young532042725, 2018-01-20:
@bespoke-code  Oh , my friend.  i removed the single quotation marks according to grammar in each .cfg file. after   rosmake....   another error!!  desperate!  
[image]

kevin-george, 2018-01-20:
Hello everyone, please go to [my fork ](https://github.com/kevin-george/lsd_slam)of the project and follow the instructions on the [wiki](https://github.com/kevin-george/lsd_slam/wiki/LSD-SLAM-with-ROS-and-Ubuntu-16.04). 
I have made all the fixes required for Ubuntu 16.04+ROS Kinetic+OpenCV3.

I have built and tested it as of today and things look peachy!

Young532042725, 2018-01-21:
@kevin-george thank u, my friend, so appreciate it!  I will try.

oneTimePad, 2018-02-12:
@kevin-george  I tried compiling this with ROS Kinetic on the Jetson TX2 using your fork, but I receive a lot of compilation errors regarding the code itself. Such as missing semicolons and variables not declared. Could this be an issue with building it on arm?

kevin-george, 2018-02-12:
I couldn't say much unless I see what the errors are. Missing semicolons sounds pretty serious :(

oneTimePad, 2018-02-12:
Yeah didn't want to past it since it's a long print out. Here is some of it:
```
lsd_slam/lsd_slam_core/CMakeFiles/lsdslam.dir/build.make:62: recipe for target 'lsd_slam/lsd_slam_core/CMakeFiles/lsdslam.dir/src/DataStructures/Frame.cpp.o' failed
make[2]: *** [lsd_slam/lsd_slam_core/CMakeFiles/lsdslam.dir/src/DataStructures/Frame.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/nvidia/ros_workspace/src/lsd_slam/lsd_slam_core/src/Tracking/Sim3Tracker.h:26:0,
                 from /home/nvidia/ros_workspace/src/lsd_slam/lsd_slam_core/src/Tracking/Sim3Tracker.cpp:21:
/home/nvidia/ros_workspace/src/lsd_slam/lsd_slam_core/src/Tracking/LGSX.h:129:11: error: ‘__m128’ does not name a type
     const __m128 &J1,const __m128 &J2,const __m128 &J3,const __m128 &J4,
           ^
/home/nvidia/ros_workspace/src/lsd_slam/lsd_slam_core/src/Tracking/LGSX.h:129:28: error: ‘__m128’ does not name a type
     const __m128 &J1,const __m128 &J2,const __m128 &J3,const __m128 &J4,
                            ^
/home/nvidia/ros_workspace/src/lsd_slam/lsd_slam_core/src/Tracking/LGSX.h:129:45: error: ‘__m128’ does not name a type
     const __m128 &J1,const __m128 &J2,const __m128 &J3,const __m128 &J4,
                                             ^
/home/nvidia/ros_workspace/src/lsd_slam/lsd_slam_core/src/Tracking/LGSX.h:129:62: error: ‘__m128’ does not name a type
     const __m128 &J1,const __m128 &J2,const __m128 &J3,const __m128 &J4,
                                                              ^
/home/nvidia/ros_workspace/src/lsd_slam/lsd_slam_core/src/Tracking/LGSX.h:130:11: error: ‘__m128’ does not name a type
     const __m128& res, const __m128& weight)
           ^
/home/nvidia/ros_workspace/src/lsd_slam/lsd_slam_core/src/Tracking/LGSX.h:130:30: error: ‘__m128’ does not name a type
     const __m128& res, const __m128& weight)
                              ^
/home/nvidia/ros_workspace/src/lsd_slam/lsd_slam_core/src/Tracking/LGSX.h: In member function ‘void lsd_slam::LGS4::finishNoDivide()’:
/home/nvidia/ros_workspace/src/lsd_slam/lsd_slam_core/src/Tracking/LGSX.h:68:4: error: ‘__m128’ was not declared in this scope
    __m128 a;
    ^
/home/nvidia/ros_workspace/src/lsd_slam/lsd_slam_core/src/Tracking/LGSX.h:70:4: error: ‘a’ was not declared in this scope
    a = _mm_load_ps(SSEData+4*0);
    ^
/home/nvidia/ros_workspace/src/lsd_slam/lsd_slam_core/src/Tracking/LGSX.h:70:31: error: ‘_mm_load_ps’ was not declared in this scope
    a = _mm_load_ps(SSEData+4*0);
                               ^
/home/nvidia/ros_workspace/src/lsd_slam/lsd_slam_core/src/Tracking/LGSX.h: In member function ‘void lsd_slam::LGS4::updateSSE(const int&, const int&, const int&, const int&, const int&, const int&)’:
/home/nvidia/ros_workspace/src/lsd_slam/lsd_slam_core/src/Tracking/LGSX.h:133:4: error: ‘__m128’ was not declared in this scope
    __m128 J1w = _mm_mul_ps(J1,weight);
```

oneTimePad, 2018-02-12:
So I got it to build by removing `add_definitions("-DENABLED_SSE")` in the core CMakeLists.txt file.

bespoke-code, 2018-02-13:
@oneTimePad just a quick suggestion - you could try to incorporate [SSE2NEON](https://github.com/jratcliff63367/sse2neon) in the code if you're building on ARM (nVidia Jetson) and you'd like to get more performance out of LSD-SLAM. SSE2NEON is a C++ header which converts SSE (x64) to NEON (ARM) instructions. This is what TUM Vision did in [their newer DSO project](https://github.com/JakobEngel/dso) - have a look!

oneTimePad, 2018-02-13:
@bespoke-code  Thank you! I'll definitely take a look!

LazerVater, 2018-09-05:
Hi,
Im running Ubuntu 16.04 and ROS Kinetic, and have compiled lsd slam from the @kevin-george fork, (many thanks !). Get quite a few  warnings that ‘Eigen::AlignedBit’ is depriciated, but no errors.
But when testing the build I am running into some problems.
 Here is what I am doing:
`roscore`
then 
`rosrun lsd_slam_viewer viewer`
so far all seems to work, a small window pops up, but then when trying to run the lsd slam core by :
`rosrun lsd_slam_core live_slam image:=/image_raw camera_info:=/camera_info`
this error appears. 
```

*** Error in `/home/peter/catkin_ws/devel/lib/lsd_slam_core/live_slam': double free or corruption (out): 0x000000000248f620 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x777e5)[0x7f98373c97e5]
/lib/x86_64-linux-gnu/libc.so.6(+0x8037a)[0x7f98373d237a]
/lib/x86_64-linux-gnu/libc.so.6(cfree+0x4c)[0x7f98373d653c]
/opt/ros/kinetic/lib/libg2o_types_sba.so(_ZN3g2o26EdgeProjectP2MC_IntrinsicsD0Ev+0x2d)[0x7f9834dcddad]
/opt/ros/kinetic/lib/libg2o_core.so(_ZN3g2o7Factory12registerTypeERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEEPNS_32AbstractHyperGraphElementCreatorE+0x33f)[0x7f983524df8f]
/opt/ros/kinetic/lib/libg2o_types_sba.so(+0x1db2a)[0x7f9834dc7b2a]
/lib64/ld-linux-x86-64.so.2(+0x106ba)[0x7f98393236ba]
/lib64/ld-linux-x86-64.so.2(+0x107cb)[0x7f98393237cb]
/lib64/ld-linux-x86-64.so.2(+0xc6a)[0x7f9839313c6a]

Aborted (core dumped)
```

Any idea what the problem might be?

20chix, 2019-01-20:
> So, after a fair amount of debugging and going through fixes in the pull requests I got it to work with Ubuntu 16.04 + ROS Kinetic + Latest version of G2O. If anyone wants to give it a try it's [...]

life saver :)

slinshady89, 2019-05-27:
> Hi,
> Im running Ubuntu 16.04 and ROS Kinetic, and have compiled lsd slam from the @kevin-george fork, (many thanks !). Get quite a few warnings that ‘Eigen::AlignedBit’ is depriciated, but no errors. [...]

Hey @LazerVater, 
I get the same errors this way. I guess it's due to his changes in the fork. If you start via the [roslaunch](https://github.com/kevin-george/lsd_slam/wiki/LSD-SLAM-with-ROS-and-Ubuntu-16.04) command he provides the viz isn't working too for me. 

He suggests to use rostopic echo /lsg_slam/pose to get the pose which unfortunately for me presents completly incorrect data since I'm not moving inside a miniature world

LazerVater, 2019-05-28:
If I remember correctly the issue was resolved when i rolled my version of Eigen back to version 3.2.

IshitaTakeshi, 2019-05-30:
I created the one which runs on Docker 

https://github.com/IshitaTakeshi/lsd_slam_noros

slinshady89, 2019-05-30:
thanks for that @IshitaTakeshi. Got your solution running with that debug window. However, I can't locate the pointcloud of the map afterwards.

IshitaTakeshi, 2019-05-31:
@slinshady89 That's the problem I'm facing to. tum-vision's implementation has a visualizer but it is removed in the ROS independent one. I'm not familiar with OpenGL so what I'm trying now is to export point cloud to ply file instead of showing in the window.

IshitaTakeshi, 2019-05-31:
@slinshady89 I added lines to export pointcloud to a ply file. The code is available on the `refactor` branch but is not stable like `master` so may require you to handle instability.

slinshady89, 2019-06-01:
@IshitaTakeshi cool thanks a lot! I will test tomorrow. If I'm able to fix something I will let you know!

IshitaTakeshi, 2019-06-03:
@slinshady89 Possibly it can be calculated from `se3FromSim3(frame->getScaledCamToWorld())` ?

slinshady89, 2019-06-03:
@IshitaTakeshi since I can't do an issue in your repo. I dont know why but now I can't start it any more even though I started the container with the script because it couldn't open a debug window
```root@0ba17b59bb25:~/workspace/lsd_slam_noros# ./bin/main_on_images data/sequence_$SEQUENCEReading Calibration from file data/sequence_30/camera.txt ... found!
found ATAN camera model, building rectifier.
Input resolution: 1280 1024
In: 0.535719 0.669567 0.493249 0.500409 0.897966
Out: Crop
Output resolution: 640 480
new K: 277.523987 291.680328 312.474304 240.035583
old K: 685.720764 685.636475 630.858154 511.918457
Prepped Warp matrices
Started mapping thread!
Started  constraint search thread!
Started optimization thread 
found 1800 image files in folder data/sequence_30/images/!
reading data/sequence_30/images/00000.jpg
Doing Random initialization!
No protocol specified

(DebugWindow DEPTH:323): Gtk-WARNING **: 10:30:32.805: cannot open display: :0```

master nor refactored branch :(

IshitaTakeshi, 2019-06-03:
@slinshady89 I just got to know issues cannot be opened on a forked repository by default so I just enabled the issue function. 
That problem (the window cannot be opened) happens frequently in my environment as well but I still cannot figure out why. Trying execution multiple times may solve it.

slinshady89, 2019-06-03:
okay that issue... 
"have you tried to turn it off and on again" 
I gonna do that. On the other hand I'll test the SVO-Stereo-SLAM because I'm no longer fixed to mono

Abrkoura, 2020-02-15:
Hi I'm trying to install and work with LSD slam on Ros melodic ubuntu 18. but whe I catkin_make this error appears:
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
QGLVIEWER_INCLUDE_DIR

closed 2020-02-15 by kevin-george