tum-vision/lsd_slam #276: LSD_Slam g2o

francoisCat, 2017-09-21:
Hello,

I am trying to install LSD-SLAM on Ubuntu 16 + kinetic (for compatibility reason as I need to also control a robot). I fixed some compatibility issues but I am a bit lost with the following compilation error:

[ 13%] Building CXX object CMakeFiles/lsdslam.dir/src/GlobalMapping/KeyFrameGraph.cpp.o
  In file included from /home/XXX/Documents/lsdslam/g2o-master/g2o/core/block_solver.h:190:0,
                   from /home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/src/GlobalMapping/KeyFrameGraph.cpp:27:
  /home/XXX/Documents/lsdslam/g2o-master/g2o/core/block_solver.hpp: In instantiation of ‘bool g2o::BlockSolver<Traits>::solve() [with Traits = g2o::BlockSolverTraits<7, 3>]’:
  /home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/src/GlobalMapping/KeyFrameGraph.cpp:376:1:   required from here
  /home/XXX/Documents/lsdslam/g2o-master/g2o/core/block_solver.hpp:374:3: error: no matching function for call to ‘g2o::SparseBlockMatrix<Eigen::Matrix<double, 7, 7> >::add(g2o::SparseBlockMatrix<Eigen::Matrix<double, 7, 7> >*&)’
     _Hpp->add(_Hschur);
     ^
  In file included from /home/XXX/Documents/lsdslam/g2o-master/g2o/core/sparse_block_matrix.h:240:0,
                   from /home/XXX/Documents/lsdslam/g2o-master/g2o/core/sparse_optimizer.h:33,
                   from /home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/src/GlobalMapping/KeyFrameGraph.h:27,
                   from /home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/src/GlobalMapping/KeyFrameGraph.cpp:21:
  /home/XXX/Documents/lsdslam/g2o-master/g2o/core/sparse_block_matrix.hpp:199:8: note: candidate: bool g2o::SparseBlockMatrix<MatrixType>::add(g2o::SparseBlockMatrix<MatrixType>&) const [with MatrixType = Eigen::Matrix<double, 7, 7>]
     bool SparseBlockMatrix<MatrixType>::add(SparseBlockMatrix<MatrixType>& dest) const
          ^
  /home/XXX/Documents/lsdslam/g2o-master/g2o/core/sparse_block_matrix.hpp:199:8: note:   no known conversion for argument 1 from ‘g2o::SparseBlockMatrix<Eigen::Matrix<double, 7, 7> >*’ to ‘g2o::SparseBlockMatrix<Eigen::Matrix<double, 7, 7> >&’
  CMakeFiles/lsdslam.dir/build.make:470: recipe for target 'CMakeFiles/lsdslam.dir/src/GlobalMapping/KeyFrameGraph.cpp.o' failed
  make[3]: *** [CMakeFiles/lsdslam.dir/src/GlobalMapping/KeyFrameGraph.cpp.o] Error 1
  make[3]: Leaving directory '/home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/build'
  CMakeFiles/Makefile2:105: recipe for target 'CMakeFiles/lsdslam.dir/all' failed
  make[2]: *** [CMakeFiles/lsdslam.dir/all] Error 2
  make[2]: Leaving directory '/home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/build'
  Makefile:127: recipe for target 'all' failed
  make[1]: *** [all] Error 2
  make[1]: Leaving directory '/home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/build'
  /opt/ros/kinetic/share/mk/cmake.mk:7: recipe for target 'all' failed

If anybody already experienced this problem which is related to g2o as far as I understand, I will really appreciate your help because I am not really fluent in c++ .

Thank you very much

kevin-george, 2017-09-21:
Did you install the latest g2o from source?

francoisCat, 2017-09-21:
I actually tried with two versions of g2o (which explain why you see the absolute path in the error log). The last version of g2o that I downloaded from github and the one provided with orb slam (as suggested in a previous issue). I don't know which version is the most appropriate but I agree with you that to me it sounds like a problem of version as it is an error in the definition of a function (or the call to this function).
I also try installing the last version of LSD SLAM (github) as well as your version (thx for sharing and helping btw) ut I still end up with g2o errors.

kevin-george, 2017-09-22:
If you built g2o from source, you should un-comment these lines [68-78](https://github.com/kevin-george/lsd_slam/blob/7f040edb95b4841e7bdf8dddeb0b5cf358000dde/lsd_slam_core/src/GlobalMapping/KeyFrameGraph.cpp#L68) and comment [80-86](https://github.com/kevin-george/lsd_slam/blob/7f040edb95b4841e7bdf8dddeb0b5cf358000dde/lsd_slam_core/src/GlobalMapping/KeyFrameGraph.cpp#L80)

If you used ros-kinetic-libg2o, you should be able to compile the source as-is.

I have tried both, so I'm pretty sure either one should work :)

francoisCat, 2017-09-22:
Thanks a lot I ll try asap and let you know which solution fixed this problem. Thank you again for your time and your help.

francoisCat, 2017-09-22:
Hello

I commented the lines however it generated new compilation errors. I will have a look to see what is going on however if you have any idea it will be greatly appreciate:

from /home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/src/GlobalMapping/KeyFrameGraph.h:27,
                   from /home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/src/GlobalMapping/KeyFrameGraph.cpp:21:
  /home/XXX/Documents/lsdslam/g2o-master/g2o/stuff/misc.h: In instantiation of ‘std::unique_ptr<_Codecvt> g2o::make_unique(ArgTs&& ...) [with T = g2o::BlockSolver<g2o::BlockSolverTraits<7, 3> >; ArgTs = {std::unique_ptr<g2o::LinearSolver<Eigen::Matrix<double, 7, 7, 0, 7, 7> >, std::default_delete<g2o::LinearSolver<Eigen::Matrix<double, 7, 7, 0, 7, 7> > > >}]’:
  /home/george/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/src/GlobalMapping/KeyFrameGraph.cpp:77:69:   required from here
  /home/george/Documents/lsdslam/g2o-master/g2o/stuff/misc.h:55:64: error: no matching function for call to ‘g2o::BlockSolver<g2o::BlockSolverTraits<7, 3> >::BlockSolver(std::unique_ptr<g2o::LinearSolver<Eigen::Matrix<double, 7, 7> > >)’
     return std::unique_ptr<T>(new T(std::forward<ArgTs>(args)...));
                                                                  ^
  In file included from /home/XXX/Documents/lsdslam/g2o-master/g2o/core/block_solver.h:190:0,
                   from /home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/src/GlobalMapping/KeyFrameGraph.cpp:27:
  /home/george/Documents/lsdslam/g2o-master/g2o/core/block_solver.hpp:42:1: note: candidate: g2o::BlockSolver<Traits>::BlockSolver(g2o::BlockSolver<Traits>::LinearSolverType*) [with Traits = g2o::BlockSolverTraits<7, 3>; g2o::BlockSolver<Traits>::LinearSolverType = g2o::LinearSolver<Eigen::Matrix<double, 7, 7> >]
   BlockSolver<Traits>::BlockSolver(LinearSolverType* linearSolver) :
   ^
  /home/XXX/Documents/lsdslam/g2o-master/g2o/core/block_solver.hpp:42:1: note:   no known conversion for argument 1 from ‘std::unique_ptr<g2o::LinearSolver<Eigen::Matrix<double, 7, 7> > >’ to ‘g2o::BlockSolver<g2o::BlockSolverTraits<7, 3> >::LinearSolverType* {aka g2o::LinearSolver<Eigen::Matrix<double, 7, 7> >*}’
  In file included from /home/XXX/Documents/lsdslam/g2o-master/g2o/core/block_solver.h:190:0,
                   from /home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/src/GlobalMapping/KeyFrameGraph.cpp:27:
  /home/XXX/Documents/lsdslam/g2o-master/g2o/core/block_solver.hpp: In instantiation of ‘bool g2o::BlockSolver<Traits>::solve() [with Traits = g2o::BlockSolverTraits<7, 3>]’:
  /home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/src/GlobalMapping/KeyFrameGraph.cpp:388:1:   required from here
  /home/XXX/Documents/lsdslam/g2o-master/g2o/core/block_solver.hpp:374:3: error: no matching function for call to ‘g2o::SparseBlockMatrix<Eigen::Matrix<double, 7, 7> >::add(g2o::SparseBlockMatrix<Eigen::Matrix<double, 7, 7> >*&)’
     _Hpp->add(_Hschur);
     ^
  In file included from /home/XXX/Documents/lsdslam/g2o-master/g2o/core/sparse_block_matrix.h:240:0,
                   from /home/XXX/Documents/lsdslam/g2o-master/g2o/core/sparse_optimizer.h:33,
                   from /home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/src/GlobalMapping/KeyFrameGraph.h:27,
                   from /home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/src/GlobalMapping/KeyFrameGraph.cpp:21:
  /home/XXX/Documents/lsdslam/g2o-master/g2o/core/sparse_block_matrix.hpp:199:8: note: candidate: bool g2o::SparseBlockMatrix<MatrixType>::add(g2o::SparseBlockMatrix<MatrixType>&) const [with MatrixType = Eigen::Matrix<double, 7, 7>]
     bool SparseBlockMatrix<MatrixType>::add(SparseBlockMatrix<MatrixType>& dest) const
          ^
  /home/XXX/Documents/lsdslam/g2o-master/g2o/core/sparse_block_matrix.hpp:199:8: note:   no known conversion for argument 1 from ‘g2o::SparseBlockMatrix<Eigen::Matrix<double, 7, 7> >*’ to ‘g2o::SparseBlockMatrix<Eigen::Matrix<double, 7, 7> >&’
  CMakeFiles/lsdslam.dir/build.make:470: recipe for target 'CMakeFiles/lsdslam.dir/src/GlobalMapping/KeyFrameGraph.cpp.o' failed
  make[3]: *** [CMakeFiles/lsdslam.dir/src/GlobalMapping/KeyFrameGraph.cpp.o] Error 1
  make[3]: *** Waiting for unfinished jobs....
  make[3]: Leaving directory '/home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/build'
  CMakeFiles/Makefile2:105: recipe for target 'CMakeFiles/lsdslam.dir/all' failed
  make[2]: *** [CMakeFiles/lsdslam.dir/all] Error 2
  make[2]: Leaving directory '/home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/build'
  Makefile:127: recipe for target 'all' failed
  make[1]: *** [all] Error 2
  make[1]: Leaving directory '/home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/build'
  /opt/ros/kinetic/share/mk/cmake.mk:7: recipe for target 'all' failed

Thank you for your help

francoisCat, 2017-09-22:
To follow up on my last comment, I also tried to use ros kinetic libg2o but still get a compilation error (I changed the keyframe.cpp file accordingly):

In file included from /opt/ros/kinetic/include/g2o/core/block_solver.h:189:0,
                   from /home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/src/GlobalMapping/KeyFrameGraph.cpp:27:
  /opt/ros/kinetic/include/g2o/core/block_solver.hpp: In instantiation of ‘bool g2o::BlockSolver<Traits>::solve() [with Traits = g2o::BlockSolverTraits<7, 3>]’:
  /home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/src/GlobalMapping/KeyFrameGraph.cpp:388:1:   required from here
  /opt/ros/kinetic/include/g2o/core/block_solver.hpp:353:3: error: no matching function for call to ‘g2o::SparseBlockMatrix<Eigen::Matrix<double, 7, 7> >::add(g2o::SparseBlockMatrix<Eigen::Matrix<double, 7, 7> >*&)’
     _Hpp->add(_Hschur);
     ^
  In file included from /home/XXX/Documents/lsdslam/g2o-master/g2o/core/sparse_block_matrix.h:240:0,
                   from /home/XXX/Documents/lsdslam/g2o-master/g2o/core/sparse_optimizer.h:33,
                   from /home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/src/GlobalMapping/KeyFrameGraph.h:27,
                   from /home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/src/GlobalMapping/KeyFrameGraph.cpp:21:
  /home/XXX/Documents/lsdslam/g2o-master/g2o/core/sparse_block_matrix.hpp:199:8: note: candidate: bool g2o::SparseBlockMatrix<MatrixType>::add(g2o::SparseBlockMatrix<MatrixType>&) const [with MatrixType = Eigen::Matrix<double, 7, 7>]
     bool SparseBlockMatrix<MatrixType>::add(SparseBlockMatrix<MatrixType>& dest) const
          ^
  /home/XXX/Documents/lsdslam/g2o-master/g2o/core/sparse_block_matrix.hpp:199:8: note:   no known conversion for argument 1 from ‘g2o::SparseBlockMatrix<Eigen::Matrix<double, 7, 7> >*’ to ‘g2o::SparseBlockMatrix<Eigen::Matrix<double, 7, 7> >&’
  CMakeFiles/lsdslam.dir/build.make:470: recipe for target 'CMakeFiles/lsdslam.dir/src/GlobalMapping/KeyFrameGraph.cpp.o' failed
  make[3]: *** [CMakeFiles/lsdslam.dir/src/GlobalMapping/KeyFrameGraph.cpp.o] Error 1
  make[3]: *** Waiting for unfinished jobs....
  make[3]: Leaving directory '/home/george/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/build'
  CMakeFiles/Makefile2:105: recipe for target 'CMakeFiles/lsdslam.dir/all' failed
  make[2]: *** [CMakeFiles/lsdslam.dir/all] Error 2
  make[2]: Leaving directory '/home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/build'
  Makefile:127: recipe for target 'all' failed
  make[1]: *** [all] Error 2
  make[1]: Leaving directory '/home/XXX/rosbuild_ws/lsdSlamP/lsd_slam/lsd_slam_core/build'
  /opt/ros/kinetic/share/mk/cmake.mk:7: recipe for target 'all' failed

francoisCat, 2017-09-22:
FIREWORKS it compiled ... sorry for the last two messages ... and thanks a lot @kevin-george (I owe you a beer if at some points you plan to visit London let me know ;) ) !!!

So basically it compiled using ros-kinetic-libg2o I just messed up with an include. Using this version of libg2o as mentioned by @kevin-george you can just use the keyframe.cpp file as it is.

Thanks again

closed 2017-09-22 by francoisCat

9friday, 2018-05-09:
Hey! I know this issue is closed, but I too am having issues compiling LSD-SLAM and the issue is related to g2o. I am compiling it to be used without the ROS dependencies. 
I have been trying out these repos:

1.  https://github.com/aivijay/lsd_slam_noros
2.  https://github.com/mp3guy/lsd_slam
3.  https://github.com/jserv/lsd_slam
The `cmake ..` command is executed successfully but on executing the `make` command, I recieve the following error:

> [  3%] Building CXX object CMakeFiles/lsdslam.dir/DataStructures/Frame.cpp.o
> In file included from /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/DataStructures/FramePoseStruct.h:23:0,
>                  from /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/DataStructures/Frame.h:26,
>                  from /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/DataStructures/Frame.cpp:21:
> /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h: In member function ‘virtual bool lsd_slam::EdgeSim3::setMeasurementData(const double*)’:
> /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h:96:20: error: ISO C++ forbids declaration of ‘type name’ with no type [-fpermissive]
>    Eigen::Map<const g2o::Vector7d> v(m);
>                     ^~~
> /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h:96:33: error: template argument 1 is invalid
>    Eigen::Map<const g2o::Vector7d> v(m);
>                                  ^
> /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h:96:38: error: invalid conversion from ‘const double*’ to ‘int’ [-fpermissive]
>    Eigen::Map<const g2o::Vector7d> v(m);
>                                       ^
> /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h:97:38: error: no matching function for call to ‘Sophus::Sim3Group<double>::exp(int&)’
>    setMeasurement(Sophus::Sim3d::exp(v));
>                                       ^
> In file included from /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/util/SophusUtil.h:23:0,
>                  from /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/DataStructures/Frame.h:22,
>                  from /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/DataStructures/Frame.cpp:21:
> /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/thirdparty/Sophus/sophus/sim3.hpp:418:27: note: candidate: static const Sophus::Sim3Group<typename Eigen::internal::traits<T>::Scalar> Sophus::Sim3GroupBase<Derived>::exp(const Tangent&) [with Derived = Sophus::Sim3Group<double>; typename Eigen::internal::traits<T>::Scalar = double; Sophus::Sim3GroupBase<Derived>::Tangent = Eigen::Matrix<double, 7, 1>]
>    const Sim3Group<Scalar> exp(const Tangent & a) {
>                            ^~~
> /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/thirdparty/Sophus/sophus/sim3.hpp:418:27: note:   no known conversion for argument 1 from ‘int’ to ‘const Tangent& {aka const Eigen::Matrix<double, 7, 1>&}’
> CMakeFiles/lsdslam.dir/build.make:62: recipe for target 'CMakeFiles/lsdslam.dir/DataStructures/Frame.cpp.o' failed
> make[2]: *** [CMakeFiles/lsdslam.dir/DataStructures/Frame.cpp.o] Error 1
> CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/lsdslam.dir/all' failed
> make[1]: *** [CMakeFiles/lsdslam.dir/all] Error 2
> 

While 2 and 3 give the exact same error, 1 is a bit different:

> 
> [  4%] Building CXX object CMakeFiles/lsd_slam.dir/depth_estimation/depth_map.cc.o
> In file included from /home/username/Downloads/lsd_slam_noros-master/lsd_slam/depth_estimation/depth_map.h:28:0,
>                  from /home/username/Downloads/lsd_slam_noros-master/lsd_slam/depth_estimation/depth_map.cc:21:
> /home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:72:31: error: ‘SE3Group’ is not a class template
>  extern template class Sophus::SE3Group<float, 0>;
>                                ^~~~~~~~
> /home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:72:31: error: ‘SE3Group’ in namespace ‘Sophus’ does not name a type
> /home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:73:31: error: ‘SE3Group’ is not a class template
>  extern template class Sophus::SE3Group<double, 0>;
>                                ^~~~~~~~
> /home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:73:31: error: ‘SE3Group’ in namespace ‘Sophus’ does not name a type
> /home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:75:31: error: ‘Sim3Group’ is not a class template
>  extern template class Sophus::Sim3Group<float, 0>;
>                                ^~~~~~~~~
> /home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:75:31: error: ‘Sim3Group’ in namespace ‘Sophus’ does not name a type
> /home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:76:31: error: ‘Sim3Group’ is not a class template
>  extern template class Sophus::Sim3Group<double, 0>;
>                                ^~~~~~~~~
[... cut ...]
[`Eigen::Map<const g2o::Vector7d> v(m);`](https://github.com/tum-vision/lsd_slam/blob/d1e6f0e1a027889985d2e6b4c0fe7a90b0c75067/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h#L96)

All three repos already have the changes suggested by @kevin-george in this [comment](https://github.com/tum-vision/lsd_slam/issues/276#issuecomment-331314318).

I would really be grateful if anybody has any suggestions as to how I should proceed. I have also opened an issue [here](https://github.com/tum-vision/lsd_slam/issues/303) regarding the same.

Thank You :+1: