tum-vision/lsd_slam #283: Compile error....unable to set Sophus directory

rnunziata, 2017-10-02:
I am compiling lsd_slam and it keeps taking the versio of Sophus released in kinetic and not the thirdparty folder even though the CMakeLists.txt has 

> :include_directories(  
> ${PROJECT_SOURCE_DIR}/thirdparty/Sophus
> ${EIGEN3_INCLUDE_DIR}  
> ${PROJECT_SOURCE_DIR}/src  
> ${CSPARSE_INCLUDE_DIR} #Has been set
> by SuiteParse   ${CHOLMOD_INCLUDE_DIR}
> #Has been set by SuiteParse )

The error is as follows , notice the libs used.

>  
> CMakeFiles/lsdslam.dir/build.make:134:
> recipe for target
> 'CMakeFiles/lsdslam.dir/src/SlamSystem.cpp.o'
> failed   make[3]: ***
> [CMakeFiles/lsdslam.dir/src/SlamSystem.cpp.o]
> Error 1   In file included from
> /opt/ros/kinetic/include/sophus/rxso3.hpp:26:0,
>                    from /opt/ros/kinetic/include/sophus/sim3.hpp:26,
>                    from /home/rjn/catkin_ws/src_nstd/package_dir/lsd_slam/lsd_slam_core/src/util/SophusUtil.h:23,
>                    from /home/rjn/catkin_ws/src_nstd/package_dir/lsd_slam/lsd_slam_core/src/LiveSLAMWrapper.h:30,
>                    from /home/rjn/catkin_ws/src_nstd/package_dir/lsd_slam/lsd_slam_core/src/LiveSLAMWrapper.cpp:21:   >/opt/ros/kinetic/include/sophus/sophus.hpp:50:23:
> warning: type qualifiers ignored on
> function return type
> [-Wignored-qualifiers]
>      const float epsilon() {
>                          ^   /opt/ros/kinetic/include/sophus/sophus.hpp:55:18:
> warning: type qualifiers ignored on
> function return type
> [-Wignored-qualifiers]
>      const float pi() {
>
>  In file included from /home/rjn/catkin_ws/src_nstd/package_dir/lsd_slam/lsd_slam_core/src/util/SophusUtil.h:23:0,
>                   from /home/rjn/catkin_ws/src_nstd/package_dir/lsd_slam/lsd_slam_core/src/LiveSLAMWrapper.h:30,
>                  from /home/rjn/catkin_ws/src_nstd/package_dir/lsd_slam/lsd_slam_core/src/LiveSLAMWrapper.cpp:21:
>  /opt/ros/kinetic/include/sophus/sim3.hpp: In instantiation of ‘void Sophus::Sim3GroupBase<Derived>::setScale(const  
> Scalar&) const [with Derived = Sophus::Sim3Group<double>; Sophus::Sim3GroupBase<Derived>::Scalar = double]’:
>  /home/rjn/catkin_ws/src_nstd/package_dir/lsd_slam/lsd_slam_core/src/util/SophusUtil.h:56:23:   required from here>
>  /opt/ros/kinetic/include/sophus/sim3.hpp:339:5: error: passing ‘const RxSO3Type {aka const >Sophus::RxSO3Group<double>}’ as ‘this’ argument discards qualifiers [-fpermissive]
>       rxso3().setScale(scale);

rnunziata, 2017-10-11:
I got around this by hand editing the references in the generated make files

closed 2017-10-11 by rnunziata

johnchars, 2019-03-18:
```
CMakeFiles/lsdslam.dir/build.make:86: recipe for target 'CMakeFiles/lsdslam.dir/src/DataStructures/FramePoseStruct.cpp.o' failed
  make[3]: *** [CMakeFiles/lsdslam.dir/src/DataStructures/FramePoseStruct.cpp.o] Error 1
```
My error seems like your compile error. Have you fixed it?