tum-vision/lsd_slam #293: Failed to read camera calibration (invalid format)?

ConradSelig, 2018-01-26:
I'm trying to run live_slam with a calibration file I made using camera_calibration.

The camera I'm using is being used through a usb_cam_node and is called /usb_cam/ under 'rostopic list'

Everything seems fine, If I don't use a calibration file live_slam runs (very slowly but that might just be the old computer I'm using).

I want to use the calibration mode, as I know that I'll get a better model, and thats what I need for the project I'm working on, however, lsd_slam cannot seem to read my camera calibration.

Here is the command I'm using:

**rosrun lsd_slam_core live_slam /image:=/usb_cam/image_raw/ _calib:=head_camera.yml**

and here is the error that is produced.

          Reading Calibration from file head_camera.yml ... not found!

           Trying /home/sdsmtuav/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/cali/head_camera.yml ... found!

          found ATAN camera model, building rectifier.

          Failed to read camera calibration (invalid format?)

          Calibration file: /home/sdsmtuav/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/calib/head_camera.yml

          Out: Failed to Read Output pars... not rectifying.

          Out: Failed to Read Output resolution... not rectifying.

          Not Rectifying

          Failed to read camera calibration from file... wrong syntax?

And then it goes back to the command line.

I've tried all the following formats, converting them with camera_calibration_parsers.

.yaml
.yml
.ini
 
Trying to convert to .cfg gives the following error:
          [ERROR] [1516986393.919639145]: Failed to save camera model to file head_camera.cfg

Why can lsd_slam not read my config file? What extension do I need to convert it to? And what command would I use to do that?

Thank you in advance!

bespoke-code, 2018-01-26:
The LSD-SLAM [README](https://github.com/tum-vision/lsd_slam/blob/master/README.md#313-camera-calibration) explains all supported calibration file formats in Section 3.1.3. You might have to do some calculations to get to the right format if you have a yaml file, but inside you'll certainly have all necessary parameters (fx, fy, cx, cy etc.). Please refer to the README for more details. 

I saved my calibration file as a .txt file. I had no problems running LSD-SLAM with it specified on the command line via the _calib:=PATH flag.

You can have a look at [Kevin George's calibration file](https://github.com/kevin-george/lsd_slam/blob/master/calibration_params) if you need an example calibration file. LSD-SLAM also includes a pinhole calibration example among others [in the calib folder inside lsd_slam_core](https://github.com/tum-vision/lsd_slam/tree/master/lsd_slam_core/calib).

I hope this will clarify the calibration file format and solve the issue.

ConradSelig, 2018-01-26:
I see what you mean with the .txt, and did get a calibration file with the .txt extension to run, but I'm not sure which numbers to use .

Kevin George's calibration file's first line has 8 params, looking at my .yaml file, there are three lists of data points that look like that might match the formatting for the numbers that Kevin George has, however, **none of these lists are 8 elements long**

Those three lists are:
          camera_matrix (9 elements)
          distortion_coefficients (5 elements)
          projection_matrix (12 elements)

which list should I use? and which data points should I leave out?

I've already tried using all three lists without removing any elements.
distortion_coefficients was most successful (it was actually the only one that loaded more than a blank screen), but the resulting map is totally flat.

bespoke-code, 2018-01-26:
Kevin George's file uses the OpenCV Calibration model:
```
fx fy cx cy k1 k2 p1 p2
inputWidth inputHeight
"crop" / "full" / "none" / "e1 e2 e3 e4 0"
outputWidth outputHeight
```
To get this, you'll need information from the camera matrix and the distortion coefficients. If you take a look at the [yaml calibration file in this tutorial](http://wiki.ros.org/camera_calibration/Tutorials/MonocularCalibration), you can see the **camera matrix** and **distortion** sections.
The camera matrix is given in the following format:
```
fx  0  cx
0  fy  cy
0   0   1
```
You can use these parameters to calculate the calibration coefficients for a FOV camera model calibration file. They're also present in the OpenCV camera calibration model as seen in K.George's file.

The distortion coefficients (k1, k2, p1 and p2 in the first line of the calibration file format above) are the first four numbers in the **distortion** section of the yaml file, so use those.

The second row holds the input size of your image topic / direct camera stream / etc. For example, if you have calibrated your camera at 640x480px resolution, use that resolution as an input resolution for your camera and enter `640 480` in the second row. These numbers are also present in the yaml file.

I'd go for a 'crop' or 'full' in the third row, just speaking from previous experience. Try both to see what works best for your camera. Or try everything and see for yourself.

Go for 640 480 in the fourth row as an output resolution. It's what the authors suggest, but beware if your camera doesn't have the same height/width ratio. For example, I've tried a camera with an input resolution of 1280x1024 - I used 640x512 in the last row (smaller output image for LSD-SLAM -> faster processing of each frame).

I hope this helps!

ConradSelig, 2018-01-26:
This is awesome! I'm convinced that my calibration file looks like what it should. That matrix you provided was super helpful :+1: 

I've just got (hopefully) one more issue. When I try to run live_slam it never seems to finish setting up. Here is the console output:

```
sdsmtuav@sdsmtuav-VGN-FW390J:~$ rosrun lsd_slam_core live_slam /image:=/usb_camera/image_raw _calib:=head_camera.txt
Reading Calibration from file head_camera.txt ... not found!
 Trying /home/sdsmtuav/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/calib/head_camera.txt ... found!
found OpenCV camera model, building rectifier.
Input resolution: 640 480
In: 660.701660 615.359314 317.788483 234.521469 0.040 -0.081868 -0.001957 -0.001337
Out: Full
Output resolution: 640 480
Started mapping thread!
Started optimization thread 
Started  constraint search thread!
```

I let the console be for almost an hour and it never changed.

The first thing I did was make sure my camera was working with
`rostopic hz /usb_cam/image_raw`
And didn't see any red flags - everything looked fine.

I decided to go scurrying the other issue forums to see if I could find a solution.
All I could come across that seemed at all useful was a [post by kilmarnock](https://github.com/tum-vision/lsd_slam/issues/115) near the bottom of that page.

Kilmarnock talked about remapping the image topic, and when I went to look for the Subscribers of my camera topic, I found that there aren't any! However, I can't seem to figure out how to add /live_slam to the subscribers list of my camera topic, and I'm not even sure if that would help my problem.

I know this is a very open ended problem, so let me know if there is any more information that would be useful!

bespoke-code, 2018-01-26:
I believe you should replace `rosrun lsd_slam_core live_slam /image:=/usb_camera/image_raw _calib:=head_camera.txt` with something like `rosrun lsd_slam_core live_slam _image:=/usb_cam/image_raw _calib:=head_camera.txt` if the topic you expect to see images from is /usb_cam/image_raw.

Mind the underscore before the image topic specified on the command line! I've had situations where I'd simply forget the underscore before the calib parameter and the whole thing doesn't want to start.

ConradSelig, 2018-01-27:
Nice eye! It did have to do with invalid syntax in the command, here is the one that worked:

`rosrun lsd_slam_core live_slam /image:=/usb_cam/image_raw _calib:=head_camera.txt
`

Looks great! Thank you again!

closed 2018-01-27 by ConradSelig