tum-vision/lsd_slam #295: CNN-SLAM with LSD-SLAM

oneTimePad, 2018-02-13:
Currently, there doesn't seem to be an implementation of [CNN-SLAM](https://arxiv.org/abs/1704.03489). However, it also uses Key-frame and Graph-based SLAM like LSD-SLAM, and a similar method for fusing depth maps (although there are differences). 

If I have a network that outputs depth maps is it possible to pass them into LSD-SLAM for a better reconstruction (possibly to absolute scale?) as opposed to using a randomly initialized depth map that LSD-SLAM uses. These would be single channel images representing depth.

srikar8, 2018-07-06:
hi @oneTimePad  
any update on absolute scale when using lsd_slam and RGB_D ?
thanks in advance.

UsamaMaq, 2019-01-10:
Hello @oneTimePad  @srikar8 
I am also interested in your question. I want to implement CNN SLAM in future. So planning to start with implementation of LSD SLAM and CNN Depth prediction. So did you find answer to your question or any progress in your quest?