tum-vision/lsd_slam #298: Getting malloc():memory corruption error while running live_slam

kungfrank, 2018-03-11:
Hi! I have Ubuntu 14.04 and ROS indigo. I get memory corruption error while running live_slam on ardrone.
`rosrun lsd_slam_core live_slam image:=/ardrone/image_raw camera_info:=/ardrone/camera_info`

And then I get error as follow

`malloc(): memory corruption: 0x0000000001b163c0 ***Aborted (core dumped)`

Anyone face same problem? Or figure out how to fix this error ?

thanks for help.

calm0815, 2018-05-01:
I'm facing the same bugs.

you already fixed it ?

kungfrank, 2018-05-01:
Hi, I fixed it by change my camera pixel to multiple of 16 EX: 640x480 .
Hope this help you.

vyi, 2018-05-30:
[This post ](https://github.com/tum-vision/lsd_slam/issues/122)gives a solution. Basically we can modify the resolution settings inside "video.h" include file inside ardrone_autonomy directory.

kungfrank, 2018-05-30:
@vyi  Yes, you are right.
Recently, I had finish the ARDrone indoor navigation project used LSD_SLAM and TUM_ARDrone.
There are more information(tutorial) at the follow link.
https://github.com/Hypha-ROS/hypharos_ardrone_navigation

closed 2018-05-30 by kungfrank

zackerwong5505, 2019-04-18:
@kungfrank  hi , can i know more detail how to change camera pixel to multiple of 16 EX: 640x480 . 
i have facing the same problem with you , but i change the value of video.h and  ardrone_front.yaml , im still getting the same problem.