tum-vision/lsd_slam #302: [rospack] Error: could not find python module 'rosdep2.rospack'. is rosdep up-to-date (at least 0.10.4)?

mgarbade, 2018-04-19:
During compilation I get a couple of errors which I could all fix by sequentially running

    pip install rospkg
    pip install catkin_pkg

but this last error seems to be unfixable using pip. Here is the full output: 

```
(py27_ros) garbade@cvg15> rosmake --pre-clean lsd_slam
[ rosmake ] rosmake starting...                                                                       
[ rosmake ] Packages requested are: ['lsd_slam']                                                      
[ rosmake ] Logging to directory /home/garbade/.ros/rosmake/rosmake_output-20180419-165055            
[ rosmake ] Expanded args ['lsd_slam'] to:
['lsd_slam_core', 'lsd_slam_viewer']                       
[rosmake-0] Starting >>> cv_bridge [ make clean ]                                                     
[rosmake-1] Starting >>> dynamic_reconfigure [ make clean ]                                           
[rosmake-2] Starting >>> sensor_msgs [ make clean ]                                                   
[rosmake-3] Starting >>> roslib [ make clean ]                                                        
[rosmake-0] Finished <<< cv_bridge ROS_NOBUILD in package cv_bridgegure: ... [ 4 Active 0/7 Complete ]
 No Makefile in package cv_bridge 
[rosmake-1] Finished <<< dynamic_reconfigure ROS_NOBUILD in package dynamic_reconfigure
 No Makefile in package dynamic_reconfigure
[rosmake-0] Starting >>> rosbag [ make clean ]                                                        
[rosmake-1] Starting >>> lsd_slam_viewer [ make clean ]                                               
[rosmake-2] Finished <<< sensor_msgs ROS_NOBUILD in package sensor_msgs
 No Makefile in package sensor_msgs
[rosmake-2] Starting >>> lsd_slam_core [ make clean ]                                                 
[rosmake-0] Finished <<< rosbag ROS_NOBUILD in package rosbag
 No Makefile in package rosbag          
[rosmake-3] Finished <<< roslib ROS_NOBUILD in package roslib
 No Makefile in package roslib          
[rosmake-1] Finished <<< lsd_slam_viewer [PASS] [ 0.06 seconds ]                                      
[rosmake-2] Finished <<< lsd_slam_core [PASS] [ 0.04 seconds ]                                        
[rosmake-0] Starting >>> cv_bridge [ make ]                                                           
[rosmake-3] Starting >>> roslib [ make ]                                                              
[rosmake-2] Starting >>> sensor_msgs [ make ]                                                         
[rosmake-1] Starting >>> dynamic_reconfigure [ make ]                                                 
[rosmake-2] Finished <<< sensor_msgs ROS_NOBUILD in package sensor_msgs
 No Makefile in package sensor_msgs
[rosmake-3] Finished <<< roslib ROS_NOBUILD in package roslib
 No Makefile in package roslib          
[rosmake-1] Finished <<< dynamic_reconfigure ROS_NOBUILD in package dynamic_reconfigure
 No Makefile in package dynamic_reconfigure
[rosmake-0] Finished <<< cv_bridge ROS_NOBUILD in package cv_bridge
 No Makefile in package cv_bridge 
[rosmake-2] Starting >>> rosbag [ make ]                                                              
[rosmake-2] Finished <<< rosbag ROS_NOBUILD in package rosbag
 No Makefile in package rosbag          
[rosmake-6] Starting >>> lsd_slam_viewer [ make ]                                                     
[ rosmake ] Last 40 linesd_slam_viewer: 0.5 sec ]                            [ 1 Active 5/7 Complete ]
{-------------------------------------------------------------------------------
  -- Detecting CXX compiler ABI info - done
  -- Found PythonInterp: /home/garbade/anaconda2/envs/py27_ros/bin/python (found version "2.7.14") 
  [rosbuild] Building package lsd_slam_viewer
  Failed to invoke /opt/ros/indigo/bin/rospack deps-manifests lsd_slam_viewer
  Traceback (most recent call last):
    File "/home/garbade/anaconda2/envs/py27_ros/lib/python2.7/site-packages/rosdep2/__init__.py", line 40, in <module>
      from .installers import InstallerContext, Installer, \
    File "/home/garbade/anaconda2/envs/py27_ros/lib/python2.7/site-packages/rosdep2/installers.py", line 36, in <module>
      from rospkg.os_detect import OsDetect
    File "/home/garbade/anaconda2/envs/py27_ros/lib/python2.7/site-packages/rospkg/__init__.py", line 41, in <module>
      from .manifest import InvalidManifest, Manifest, parse_manifest_file
    File "/home/garbade/anaconda2/envs/py27_ros/lib/python2.7/site-packages/rospkg/manifest.py", line 40, in <module>
      import xml.dom.minidom as dom
    File "/home/garbade/anaconda2/envs/py27_ros/lib/python2.7/xml/dom/minidom.py", line 22, in <module>
      from xml.dom.xmlbuilder import DOMImplementationLS, DocumentLS
    File "/home/garbade/anaconda2/envs/py27_ros/lib/python2.7/xml/dom/xmlbuilder.py", line 3, in <module>
      import copy
    File "/home/garbade/anaconda2/envs/py27_ros/lib/python2.7/copy.py", line 52, in <module>
      import weakref
    File "/home/garbade/anaconda2/envs/py27_ros/lib/python2.7/weakref.py", line 14, in <module>
      from _weakref import (
  ImportError: cannot import name _remove_dead_weakref
  [rospack] Error: could not find python module 'rosdep2.rospack'. is rosdep up-to-date (at least 0.10.4)?
  
  
  CMake Error at /opt/ros/indigo/share/ros/core/rosbuild/public.cmake:129 (message):
    
  
    Failed to invoke rospack to get compile flags for package
    'lsd_slam_viewer'.  Look above for errors from rospack itself.  Aborting.
    Please fix the broken dependency!
  
  Call Stack (most recent call first):
    /opt/ros/indigo/share/ros/core/rosbuild/public.cmake:207 (rosbuild_invoke_rospack)
    CMakeLists.txt:13 (rosbuild_init)
  
  
  -- Configuring incomplete, errors occurred!
  See also "/home/garbade/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build/CMakeFiles/CMakeOutput.log".
-------------------------------------------------------------------------------}
[ rosmake ] Output from build of package lsd_slam_viewer written to:
[ rosmake ]    /home/garbade/.ros/rosmake/rosmake_output-20180419-165055/lsd_slam_viewer/build_output.log
[rosmake-6] Finished <<< lsd_slam_viewer [FAIL] [ 0.63 seconds ]                                      
[ rosmake ] Halting due to failure in package lsd_slam_viewer. 
[ rosmake ] Waiting for other threads to complete.
[ rosmake ] Results:                                                                                  
[ rosmake ] Cleaned 7 packages.                                                                       
[ rosmake ] Built 6 packages with 1 failures.                                                         
[ rosmake ] Summary output to directory                                                               
[ rosmake ] /home/garbade/.ros/rosmake/rosmake_output-20180419-165055 
```

rosdep version seems to be fine:

    (py27_ros) garbade@cvg15> rosdep --version
    0.12.2

Any idea how to fix this?

I'm using:
Ubuntu 14.04, ROS indigo, anaconda2  with pyhton 2.7

IdaChenCCL, 2019-08-21:
Hi there, how is it solved?

hipforth, 2019-11-23:
Hi, I also have this problem.

`File "/usr/lib/python2.7/dist-packages/rosdep2/__init__.py", line 40, in <module>
      from .installers import InstallerContext, Installer, \
    File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 33, in <module>
      import subprocess
    File "/usr/lib/python2.7/subprocess.py", line 427, in <module>
      import select
  ImportError: No module named select
  [rospack] Error: could not find python module 'rosdep2.rospack'. is rosdep up-to-date (at least 0.10.4)?
  
  
  CMake Error at /opt/ros/indigo/share/ros/core/rosbuild/public.cmake:129 (message):
    
  
    Failed to invoke rospack to get compile flags for package 'dso_ros'.  Look
    above for errors from rospack itself.  Aborting.  Please fix the broken
    dependency!
  
  Call Stack (most recent call first):
    /opt/ros/indigo/share/ros/core/rosbuild/public.cmake:207 (rosbuild_invoke_rospack)
    CMakeLists.txt:13 (rosbuild_init)
  
  
  -- Configuring incomplete, errors occurred!
  See also "/root/catkin_ws/src/dso_ros/build/CMakeFiles/CMakeOutput.log".
-------------------------------------------------------------------------------}
[ rosmake ] Output from build of package dso_ros written to:
[ rosmake ]    /root/.ros/rosmake/rosmake_output-20191123-144057/dso_ros/build_output.log
[rosmake-6] Finished <<< dso_ros [FAIL] [ 0.16 seconds ]                                                                
[ rosmake ] Halting due to failure in package dso_ros. 
[ rosmake ] Waiting for other threads to complete.              
[ rosmake ] Results:                                                                                                    
[ rosmake ] Built 34 packages with 1 failures.                                                                          
[ rosmake ] Summary output to directory                                                                                 
[ rosmake ] /root/.ros/rosmake/rosmake_output-20191123-144057                    `

I'm using: Ubuntu 14.04, ROS indigo in docker

withnoerr, 2024-04-27:
我找到了问题原因, 是由于电脑中同时存在多个版本的python, 所以需要将其他版本python删除掉, 重新运行make, 即可解决此问题
ps: 我重刷了系统, 在没有其他python版本的情况下重新进行编译,成功了