tum-vision/lsd_slam #303: Error while 'making' without ROS dependency (Eigen::Map<const g2o::Vector7d> v(m);)

9friday, 2018-04-28:
I have a 64-bit laptop with Ubuntu 17.10. I am trying to run LSD-SLAM without the ROS dependencies. So far, I have found three repos which contain LSD-SLAM without the ROS dependency. They are all modified versions of the original LSD-SLAM.
But on building them with cmake and then executing 'sudo make', I get a common error across all three repos.

    [ 3%] Building CXX object CMakeFiles/lsdslam.dir/DataStructures/Frame.cpp.o
    In file included from /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/DataStructures/FramePoseStruct.h:23:0,
    from /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/DataStructures/Frame.h:26,
    from /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/DataStructures/Frame.cpp:21:
    /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h: In member function ‘virtual bool lsd_slam::EdgeSim3::setMeasurementData(const double*)’:
    /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h:96:20: error: ISO C++ forbids declaration of ‘type name’ with no type [-fpermissive]
    Eigen::Map v(m);
    ^~~
    /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h:96:33: error: template argument 1 is invalid
    Eigen::Map v(m);
    ^
    /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h:96:38: error: invalid conversion from ‘const double*’ to ‘int’ [-fpermissive]
    Eigen::Map v(m);
    ^
    /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h:97:38: error: no matching function for call to ‘Sophus::Sim3Group::exp(int&)’
    setMeasurement(Sophus::Sim3d::exp(v));
    ^
    In file included from /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/util/SophusUtil.h:23:0,
    from /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/DataStructures/Frame.h:22,
    from /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/DataStructures/Frame.cpp:21:
    /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/thirdparty/Sophus/sophus/sim3.hpp:418:27: note: candidate: static const Sophus::Sim3Group<typename Eigen::internal::traits::Scalar> Sophus::Sim3GroupBase::exp(const Tangent&) [with Derived = Sophus::Sim3Group; typename Eigen::internal::traits::Scalar = double; Sophus::Sim3GroupBase::Tangent = Eigen::Matrix<double, 7, 1>]
    const Sim3Group exp(const Tangent & a) {
    ^~~
    /home/username/Downloads/lsd_slam-master-mp3guy/lsd_slam-master/lsd_slam_core/src/thirdparty/Sophus/sophus/sim3.hpp:418:27: note: no known conversion for argument 1 from ‘int’ to ‘const Tangent& {aka const Eigen::Matrix<double, 7, 1>&}’
    CMakeFiles/lsdslam.dir/build.make:62: recipe for target 'CMakeFiles/lsdslam.dir/DataStructures/Frame.cpp.o' failed
    make[2]: *** [CMakeFiles/lsdslam.dir/DataStructures/Frame.cpp.o] Error 1
    CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/lsdslam.dir/all' failed
    make[1]: *** [CMakeFiles/lsdslam.dir/all] Error 2

These are the links to the repos:

    https://github.com/aivijay/lsd_slam_noros
    https://github.com/mp3guy/lsd_slam
    https://github.com/jserv/lsd_slam

While 2 and 3 give the exact same error, 1 is a bit different:

    [ 4%] Building CXX object CMakeFiles/lsd_slam.dir/depth_estimation/depth_map.cc.o
    In file included from /home/username/Downloads/lsd_slam_noros-master/lsd_slam/depth_estimation/depth_map.h:28:0,
    from /home/username/Downloads/lsd_slam_noros-master/lsd_slam/depth_estimation/depth_map.cc:21:
    /home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:72:31: error: ‘SE3Group’ is not a class template
    extern template class Sophus::SE3Group<float, 0>;
    ^~~~~~~~
    /home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:72:31: error: ‘SE3Group’ in namespace ‘Sophus’ does not name a type
    /home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:73:31: error: ‘SE3Group’ is not a class template
    extern template class Sophus::SE3Group<double, 0>;
    ^~~~~~~~
    /home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:73:31: error: ‘SE3Group’ in namespace ‘Sophus’ does not name a type
    /home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:75:31: error: ‘Sim3Group’ is not a class template
    extern template class Sophus::Sim3Group<float, 0>;
    ^~~~~~~~~
    /home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:75:31: error: ‘Sim3Group’ in namespace ‘Sophus’ does not name a type
    /home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:76:31: error: ‘Sim3Group’ is not a class template
    extern template class Sophus::Sim3Group<double, 0>;
    ^~~~~~~~~
    /home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:76:31: error: ‘Sim3Group’ in namespace ‘Sophus’ does not name a type
    In file included from /home/username/Downloads/lsd_slam_noros-master/lsd_slam/model/frame_pose_struct.h:23:0,
    from /home/username/Downloads/lsd_slam_noros-master/lsd_slam/model/frame.h:26,
    from /home/username/Downloads/lsd_slam_noros-master/lsd_slam/depth_estimation/depth_map.cc:31:
    /home/username/Downloads/lsd_slam_noros-master/lsd_slam/global_mapping/g2o_type_sim3_sophus.h: In member function ‘virtual bool lsd_slam::EdgeSim3::setMeasurementData(const double*)’:
    /home/username/Downloads/lsd_slam_noros-master/lsd_slam/global_mapping/g2o_type_sim3_sophus.h:96:20: error: ISO C++ forbids declaration of ‘type name’ with no type [-fpermissive]
    Eigen::Map v(m);
    ^~~
    /home/username/Downloads/lsd_slam_noros-master/lsd_slam/global_mapping/g2o_type_sim3_sophus.h:96:33: error: template argument 1 is invalid
    Eigen::Map v(m);
    ^
    /home/username/Downloads/lsd_slam_noros-master/lsd_slam/global_mapping/g2o_type_sim3_sophus.h:96:38: error: invalid conversion from ‘const double*’ to ‘int’ [-fpermissive]
    Eigen::Map v(m);
    ^
    /home/username/Downloads/lsd_slam_noros-master/lsd_slam/global_mapping/g2o_type_sim3_sophus.h:97:38: error: no matching function for call to ‘Sophus::Sim3::exp(int&)’
    setMeasurement(Sophus::Sim3d::exp(v));
    ^
    In file included from /home/username/Downloads/lsd_slam_noros-master/lsd_slam/util/sophus_util.h:23:0,
    from /home/username/Downloads/lsd_slam_noros-master/lsd_slam/depth_estimation/depth_map.h:28,
    from /home/username/Downloads/lsd_slam_noros-master/lsd_slam/depth_estimation/depth_map.cc:21:
    /usr/local/include/sophus/sim3.hpp:439:35: note: candidate: static Sophus::Sim3<Scalar_> Sophus::Sim3<Scalar_, Options>::exp(const Tangent&) [with Scalar_ = double; int Options = 0; Sophus::Sim3<Scalar_, Options>::Tangent = Eigen::Matrix<double, 7, 1, 0, 7, 1>]
    SOPHUS_FUNC static Sim3 exp(Tangent const& a) {
    ^~~
    /usr/local/include/sophus/sim3.hpp:439:35: note: no known conversion for argument 1 from ‘int’ to ‘const Tangent& {aka const Eigen::Matrix<double, 7, 1, 0, 7, 1>&}’
    CMakeFiles/lsd_slam.dir/build.make:62: recipe for target 'CMakeFiles/lsd_slam.dir/depth_estimation/depth_map.cc.o' failed
    make[2]: *** [CMakeFiles/lsd_slam.dir/depth_estimation/depth_map.cc.o] Error 1
    CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/lsd_slam.dir/all' failed
    make[1]: *** [CMakeFiles/lsd_slam.dir/all] Error 2
    Makefile:83: recipe for target 'all' failed
    make: *** [all] Error 2

But even this has that error related to this line :
Eigen::Map<const g2o::Vector7d> v(m);

I would be really grateful if you guys could help me out or just point me in the right direction.

Than You +1

9friday, 2018-05-04:
@OAkyildiz , thanks for the advice. I tried that but I am getting the same error:

> [  3%] Building CXX object CMakeFiles/lsdslam.dir/DataStructures/Frame.cpp.o
> In file included from /home/username/Downloads/lsd_slam_jserv/lsd_slam-master/lsd_slam_core/src/DataStructures/FramePoseStruct.h:23:0,
>                  from /home/username/Downloads/lsd_slam_jserv/lsd_slam-master/lsd_slam_core/src/DataStructures/Frame.h:26,
>                  from /home/username/Downloads/lsd_slam_jserv/lsd_slam-master/lsd_slam_core/src/DataStructures/Frame.cpp:21:
> /home/username/Downloads/lsd_slam_jserv/lsd_slam-master/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h: In member function ‘virtual bool lsd_slam::EdgeSim3::setMeasurementData(const double*)’:
> /home/username/Downloads/lsd_slam_jserv/lsd_slam-master/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h:98:20: error: ISO C++ forbids declaration of ‘type name’ with no type [-fpermissive]
>    Eigen::Map<const g2o::Vector7d> v(m);
>                     ^~~
> /home/username/Downloads/lsd_slam_jserv/lsd_slam-master/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h:98:33: error: template argument 1 is invalid
>    Eigen::Map<const g2o::Vector7d> v(m);
>                                  ^
> /home/username/Downloads/lsd_slam_jserv/lsd_slam-master/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h:98:38: error: invalid conversion from ‘const double*’ to ‘int’ [-fpermissive]
>    Eigen::Map<const g2o::Vector7d> v(m);
>                                       ^
> /home/username/Downloads/lsd_slam_jserv/lsd_slam-master/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h:99:38: error: no matching function for call to ‘Sophus::Sim3Group<double>::exp(int&)’
>    setMeasurement(Sophus::Sim3d::exp(v));
>                                       ^
> In file included from /home/username/Downloads/lsd_slam_jserv/lsd_slam-master/lsd_slam_core/src/util/SophusUtil.h:23:0,
>                  from /home/username/Downloads/lsd_slam_jserv/lsd_slam-master/lsd_slam_core/src/DataStructures/Frame.h:22,
>                  from /home/username/Downloads/lsd_slam_jserv/lsd_slam-master/lsd_slam_core/src/DataStructures/Frame.cpp:21:
> /home/username/Downloads/lsd_slam_jserv/lsd_slam-master/lsd_slam_core/src/thirdparty/Sophus/sophus/sim3.hpp:418:27: note: candidate: static const Sophus::Sim3Group<typename Eigen::internal::traits<T>::Scalar> Sophus::Sim3GroupBase<Derived>::exp(const Tangent&) [with Derived = Sophus::Sim3Group<double>; typename Eigen::internal::traits<T>::Scalar = double; Sophus::Sim3GroupBase<Derived>::Tangent = Eigen::Matrix<double, 7, 1>]
>    const Sim3Group<Scalar> exp(const Tangent & a) {
>                            ^~~
> /home/username/Downloads/lsd_slam_jserv/lsd_slam-master/lsd_slam_core/src/thirdparty/Sophus/sophus/sim3.hpp:418:27: note:   no known conversion for argument 1 from ‘int’ to ‘const Tangent& {aka const Eigen::Matrix<double, 7, 1>&}’
> CMakeFiles/lsdslam.dir/build.make:62: recipe for target 'CMakeFiles/lsdslam.dir/DataStructures/Frame.cpp.o' failed
> make[2]: *** [CMakeFiles/lsdslam.dir/DataStructures/Frame.cpp.o] Error 1
> CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/lsdslam.dir/all' failed
> make[1]: *** [CMakeFiles/lsdslam.dir/all] Error 2
> Makefile:83: recipe for target 'all' failed
> make: *** [all] Error 2

I tried the changes on the [https://github.com/jserv/lsd_slam](https://github.com/jserv/lsd_slam) repo.

Thanks for helping. I will keep trying to solve this.

Cheers  :+1:

sadidaa, 2018-06-18:
Were you able to solve this?

9friday, 2018-06-18:
No, i haven't been able to solve this. What are the steps you took for
installing this?

sadidaa, 2018-06-18:
Even I haven't been able to

9friday, 2018-06-18:
What are the exact steps you followed? Can you mention them here?

9friday, 2018-06-18:
Even if you haven't been able to.

city22, 2018-11-28:
g2oTypeSim3Sophus.h line 96

`Eigen::Map<const g2o::Vector7d> v(m);`

change to

`Eigen::Map<const Eigen::Matrix<double, 7 ,1> > v(m);`

xlla, 2019-09-24:
> g2oTypeSim3Sophus.h line 96
> 
> `Eigen::Map<const g2o::Vector7d> v(m);`
> 
> change to
> 
> `Eigen::Map<const Eigen::Matrix<double, 7 ,1> > v(m);`

I think that maybe a misstype of Sophus::Vector7d