tum-vision/lsd_slam #307: rosmake lsd_slam error

srikar8, 2018-05-30:
Ubuntu version - 17.10
ros - melodic

main_stitchVideos.cpp:(.text._ZN2cv3MatD2Ev[_ZN2cv3MatD5Ev]+0x89): undefined reference to `cv::Mat::deallocate()'
  main_stitchVideos.cpp:(.text._ZN2cv3MatD2Ev[_ZN2cv3MatD5Ev]+0x76): undefined reference to `cv::fastFree(void*)'
  collect2: error: ld returned 1 exit status
  CMakeFiles/videoStitch.dir/build.make:132: recipe for target '../bin/videoStitch' failed
  make[3]: *** [../bin/videoStitch] Error 1
  make[3]: Leaving directory '/home/iq4/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
  CMakeFiles/Makefile2:205: recipe for target 'CMakeFiles/videoStitch.dir/all' failed
  make[2]: *** [CMakeFiles/videoStitch.dir/all] Error 2
  make[2]: *** Waiting for unfinished jobs....
  In file included from /home/iq4/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.cpp:39:0:
  /home/iq4/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/KeyFrameDisplay.h:24:10: fatal error: Eigen/Core: No such file or directory
   #include <Eigen/Core>
            ^~~~~~~~~~~~
  compilation terminated.
  CMakeFiles/viewer.dir/build.make:162: recipe for target 'CMakeFiles/viewer.dir/src/PointCloudViewer.cpp.o' failed
  make[3]: *** [CMakeFiles/viewer.dir/src/PointCloudViewer.cpp.o] Error 1
  In file included from /home/iq4/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/main_viewer.cpp:25:0:
  /home/iq4/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.h: In member function ‘std::__cxx11::string AnimationObject::toString()’:
  /home/iq4/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.h:136:26: error: cannot bind non-const lvalue reference of type ‘qreal& {aka double&}’ to an rvalue of type ‘qreal {aka double}’
     frame.getPosition(x,y,z);
                            ^
  In file included from /usr/include/QGLViewer/keyFrameInterpolator.h:31:0,
                   from /usr/include/QGLViewer/camera.h:26,
                   from /usr/include/QGLViewer/qglviewer.h:26,
                   from /home/iq4/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/PointCloudViewer.h:26,
                   from /home/iq4/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/src/main_viewer.cpp:25:
  /usr/include/QGLViewer/frame.h:188:7: note:   initializing argument 1 of ‘void qglviewer::Frame::getPosition(qreal&, qreal&, qreal&) const’
    void getPosition(qreal& x, qreal& y, qreal& z) const;
         ^~~~~~~~~~~
  CMakeFiles/viewer.dir/build.make:100: recipe for target 'CMakeFiles/viewer.dir/src/main_viewer.cpp.o' failed
  make[3]: *** [CMakeFiles/viewer.dir/src/main_viewer.cpp.o] Error 1
  make[3]: Leaving directory '/home/iq4/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
  CMakeFiles/Makefile2:402: recipe for target 'CMakeFiles/viewer.dir/all' failed
  make[2]: *** [CMakeFiles/viewer.dir/all] Error 2
  make[2]: Leaving directory '/home/iq4/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
  Makefile:129: recipe for target 'all' failed
  make[1]: *** [all] Error 2
  make[1]: Leaving directory '/home/iq4/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build'
  /opt/ros/melodic/share/mk/cmake.mk:7: recipe for target 'all' failed

srikar8, 2018-06-11:
qreal

closed 2018-06-11 by srikar8

omarabid59, 2018-08-25:
How did you solve this error?
\

iker-lluvia, 2023-03-22:
> How did you solve this error? \

Defining x, y and z variables as _qreal_ instead of _float_ solves the error.