tum-vision/lsd_slam #313: error: ISO C++ forbids declaration of ‘type name’ with no type [-fpermissive] Eigen::Map<const g2o::Vector7d> v(m);

TengFeiHan0, 2018-07-12:
when running LSD-slam on my laptop, I received these unexpected errors

```
GlobalMapping/g2oTypeSim3Sophus.h:96:20: error: ISO C++ forbids declaration of ‘type name’ with no type [-fpermissive]
   Eigen::Map<const g2o::Vector7d> v(m);
                    ^~~
/home/htf/catkin_ws/src/third_parties/lsd_slam/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h:96:33: error: template argument 1 is invalid
   Eigen::Map<const g2o::Vector7d> v(m);
                                 ^
/home/htf/catkin_ws/src/third_parties/lsd_slam/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h:96:38: error: invalid conversion from ‘const double*’ to ‘int’ [-fpermissive]
   Eigen::Map<const g2o::Vector7d> v(m);
                                      ^
/home/htf/catkin_ws/src/third_parties/lsd_slam/lsd_slam_core/src/GlobalMapping/g2oTypeSim3Sophus.h:97:38: error: no matching function for call to ‘Sophus::Sim3Group<double>::exp(int&)’
 setMeasurement(Sophus::Sim3d::exp(v));
```
does anyone has encountered this problem? hope to get some clues, Thank you in advance.

smmmmi, 2018-07-25:
+1 
Same problem with #303 and #276
No solution so far.

MrMinemeet, 2019-07-15:
@city22 's solution worked for me