tum-vision/lsd_slam #320: Any way to initialize with rgb-d camera's depth?

ozakiryota, 2018-10-03:
I use LSD-SLAM with rgb-d camera RealSense.
Is there any way to initialize the depth map with rgb-d sensor's values instead of the random initialization?
I would like to improve the trajectory at beginning and get absolute scales, not relative.
Thanks.

bespoke-code, 2018-10-08:
If you want to use a depth camera to map the environment, you should take some serious look at TUM-Vision's [Volumetric 3D Mapping in Real Time paper and repository](https://github.com/tum-vision/fastfusion). 

You could potentially initialize the depth map using the depth sensor, but then, what's the point of using a SLAM algorithm for mapping? What's more, it may involve a lot of trial and error to make it work :) 

I hope this helps!

ozakiryota, 2018-10-10:
Thank you for your reply and help.
I’ll take look at it.