tum-vision/lsd_slam #321: terminate called after throwing an instance of 'std::runtime_error'   what():  Duration is out of dual 32-bit range

Mrxiaofei, 2018-10-03:
When i run the code:rosrun lsd_slam_core dataset_slam _files:=/home/lab/rosbuild_ws/LSD_room/images _hz:=0 _calib:=/home/lab/rosbuild_ws/LSD_room/cameraCalibration.cfg
It wiil report the error:terminate called after throwing an instance of 'std::runtime_error'
  what():  Duration is out of dual 32-bit range
my machine is ubuntu16.04、ROS Kinetic

gispda, 2018-11-05:
I have resolve the problem,There is a bug in the rostime module,rate.cpp
Rate::Rate(double frequency)
: start_(Time::now())
, expected_cycle_time_()
, actual_cycle_time_(0.0)
{ 
   if (frequency!=0)
  	expected_cycle_time_=ros::Duration(1/frequency);
}
And then make librostime.so, can cp the so file to /opt/ros/kinetic/lib
And run okay.

Mrxiaofei, 2018-11-14:
I solved the problem by setting images _hz:=1(not equal to zero is ok) I think the time is equal to 1/frequency ,so 1/0 is gigantic

Mrxiaofei, 2018-11-14:
> I have resolve the problem,There is a bug in the rostime module,rate.cpp
> Rate::Rate(double frequency)
> : start_(Time::now())
> , expected_cycle_time_()
> , actual_cycle_time_(0.0)
> {
> if (frequency!=0) [...]
I solved the problem by setting images _hz:=1(not equal to zero is ok) I think the time is equal to 1/frequency ,so 1/0 is gigantic

Mrxiaofei, 2018-11-15:
Hi,where is the rate.cpp

karthikv123, 2018-11-15:
Hi, i am also having the same error while running "rosrun rviz rviz". Please anybody help me to solve this issue,
thanks in advance...

karthikv123, 2018-11-15:
Hi, i am also having the same error while running "rosrun rviz rviz".

$rosrun rviz rviz 
[ INFO] [1542287615.441010909]: rviz version 1.12.16
[ INFO] [1542287615.446363938]: compiled against Qt version 5.5.1
[ INFO] [1542287615.446392781]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1542287615.786892573]: Stereo is NOT SUPPORTED
[ INFO] [1542287615.788349467]: OpenGl version: 3 (GLSL 1.3).
terminate called after throwing an instance of 'std::runtime_error'
  what():  Duration is out of dual 32-bit range
Aborted (core dumped)

 Please anybody help me to solve this issue,
thanks in advance...

karthikv123, 2018-11-15:
the above issue is solved..

This is a link on how to update the Mesa 3D graphics on Ubuntu 16.04: http://ubuntuhandbook.org/index.php/2018/01/how-to-install-mesa-17-3-3-in-ubuntu-16-04-17-10/

I hope this helps because I lost days troubleshooting this as a beginner in ROS and Ubuntu myself.

Wayne-xixi, 2019-11-11:
> the above issue is solved..
> 
> This is a link on how to update the Mesa 3D graphics on Ubuntu 16.04: http://ubuntuhandbook.org/index.php/2018/01/how-to-install-mesa-17-3-3-in-ubuntu-16-04-17-10/
> 
> I hope this helps because I lost days troubleshooting this as a beginner in ROS and Ubuntu myself.

thank you very much! solved!

PavanproJack, 2020-06-20:
> the above issue is solved..
> 
> This is a link on how to update the Mesa 3D graphics on Ubuntu 16.04: http://ubuntuhandbook.org/index.php/2018/01/how-to-install-mesa-17-3-3-in-ubuntu-16-04-17-10/
> 
> I hope this helps because I lost days troubleshooting this as a beginner in ROS and Ubuntu myself.

This Worked

justinlim812, 2021-12-19:
> the above issue is solved..
> 
> This is a link on how to update the Mesa 3D graphics on Ubuntu 16.04: http://ubuntuhandbook.org/index.php/2018/01/how-to-install-mesa-17-3-3-in-ubuntu-16-04-17-10/
> 
> I hope this helps because I lost days troubleshooting this as a beginner in ROS and Ubuntu myself.

This worked for this exact same issue I'm facing. Thanks!