tum-vision/lsd_slam #325: liblsdslam.so: undefined reference to `cv::_OutputArray::_OutputArray(cv::Mat&)'

boonflies, 2018-11-09:
I am trying to build lsd_slam for ros kinetic in ubuntu 16.04, when i get the following error,
home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::_OutputArray::_OutputArray(cv::Mat&)'
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::line(cv::Mat&, cv::Point_<int>, cv::Point_<int>, cv::Scalar_<double> const&, int, int, int)'
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::_InputArray::_InputArray(cv::Mat const&)'
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::getOptimalNewCameraMatrix(cv::_InputArray const&, cv::_InputArray const&, cv::Size_<int>, double, cv::Size_<int>, cv::Rect_<int>*, bool)'
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::waitKey(int)'
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::namedWindow(cv::String const&, int)'
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::destroyAllWindows()'
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::resizeWindow(cv::String const&, int, int)'
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::imshow(cv::String const&, cv::_InputArray const&)'
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::putText(cv::Mat&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::Point_<int>, int, double, cv::Scalar_<double>, int, int, bool)'
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::_OutputArray::_OutputArray(cv::Mat&)'
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::line(cv::Mat&, cv::Point_<int>, cv::Point_<int>, cv::Scalar_<double> const&, int, int, int)'
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::_InputArray::_InputArray(cv::Mat const&)'
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::getOptimalNewCameraMatrix(cv::_InputArray const&, cv::_InputArray const&, cv::Size_<int>, double, cv::Size_<int>, cv::Rect_<int>*, bool)'
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::waitKey(int)'
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::namedWindow(cv::String const&, int)'
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::destroyAllWindows()'
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::resizeWindow(cv::String const&, int, int)'
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::imshow(cv::String const&, cv::_InputArray const&)' 
/home/dyana/catkin_make/devel/lib/liblsdslam.so: undefined reference to `cv::putText(cv::Mat&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, cv::Point_<int>, int, double, cv::Scalar_<double>, int, int, bool)'

How to resolve this issue?

bespoke-code, 2018-11-09:
@boonflies could you please check if you can use [Kevin George's fork of LSD-SLAM for Ubuntu 16](https://github.com/kevin-george/lsd_slam)? From my knowledge, it's the best fork of LSD-SLAM specifically for use on Ubuntu 16 + ROS Kinetic. It saves a ton of time debugging!

I hope this helps! :)

aniketgadwe, 2019-05-03:
@boonflies LSD-SLAM Kevin George's fork of LSD-SLAM for Ubuntu 16 works like a charm. Thanks

bespoke-code, 2019-05-03:
You're welcome, @aniketgadwe! Happy coding :)