tum-vision/lsd_slam #326: QObject::startTimer: Timers cannot be started from another thread

boonflies, 2018-11-16:
I migrated to ros kinetic lsd slam. 

When I launch lsd_slam_core,
I get the error,
**QObject::startTimer: Timers cannot be started from another thread**
[image]

lsd slam viewer is working fine.

In the lsd_slam issues, conflict between qt4 and qt5 has been given as reason.

johnchars, 2019-03-27:
I get same error, after gdb live_slam, the detail is

```
Thread 1 "live_slam" received signal SIGINT, Interrupt.
0x00007ffff5eef30d in nanosleep () at ../sysdeps/unix/syscall-template.S:84
84	../sysdeps/unix/syscall-template.S: No such file or directory.
(gdb) bt
#0  0x00007ffff5eef30d in nanosleep () at ../sysdeps/unix/syscall-template.S:84
#1  0x00007ffff5f20d94 in usleep (useconds=<optimized out>) at ../sysdeps/posix/usleep.c:32
#2  0x00007ffff2e5aef6 in cvWaitKey ()
   from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_highgui3.so.3.3
#3  0x00007ffff2e54f43 in cv::waitKeyEx(int) ()
   from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_highgui3.so.3.3
#4  0x00007ffff2e54fd1 in cv::waitKey(int) ()
   from /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_highgui3.so.3.3
#5  0x00007ffff6d66e66 in lsd_slam::Relocalizer::updateCurrentFrame(std::shared_ptr<lsd_slam::Frame>) () from /home/cwj/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/lib/liblsdslam.so
#6  0x00007ffff6d21a98 in lsd_slam::SlamSystem::trackFrame(unsigned char*, unsigned int, bool, double) () from /home/cwj/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/lib/liblsdslam.so
#7  0x00007ffff6d3f9d6 in lsd_slam::LiveSLAMWrapper::newImageCallback(cv::Mat const&, lsd_slam::Timestamp) ()
   from /home/cwj/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/lib/liblsdslam.so
#8  0x00007ffff6d40ad8 in lsd_slam::LiveSLAMWrapper::Loop() ()
   from /home/cwj/rosbuild_ws/package_dir/lsd_slam/lsd_slam_core/lib/liblsdslam.so
#9  0x000000000041bc59 in main ()
```
Is the OpenCV library compiling leads to this error? 
Do you fix it now?

kapil97, 2019-12-08:
Did anyone find a fix for this?

Cielyuy, 2019-12-23:
Have u solved the problem? I encounter the same question with u.