tum-vision/lsd_slam #329: EuRoC dataset evaluations

jwangjie, 2019-01-22:
Hello,

I was wondering has anyone successfully evaluated the LSD SLAM with the EuRoC dataset? I tested by `rosrun lsd_slam_core dataset _files:=/EuRoC/V1_01_easy/mav0/cam0/data _hz:=20 _calib:=/euroc_cam0.cfg` and `rosbag play /EuRoC/V1_01_easy/V1_01_easy.bag` using `euroc_cam0.cfg` as the calibration file. `euroc_cam0.cfg` follows the format specified in [3.1.3 Camera Calibration](https://github.com/tum-vision/lsd_slam#313-camera-calibration) as:
```
458.654/752 457.296/480 367.215/752 248.375/480 0
752 480
none
752 480
```
The tracking lost soon after a few seconds for all EuRoC dataset (MH, V1, V2). Any suggestion? 

Regards,
Jie

pisuto, 2019-02-09:
hi,

do you solve this problem? I have met the same issue.

jwangjie, 2019-02-11:
> hi,
> 
> do you solve this problem? I have met the same issue.

@pisuto 
Please try the following one, it works. 
```
0.50355332 0.894045767 0.494061013 0.509863 0
1280 720
crop
400 288
```
I will post a comprehensive document talking about this later. Good luck.

pisuto, 2019-02-12:
Hi, @jwangjie 
I use your file, but the ratio may be wrong.
`400 288`
this is warning:
[image]

pisuto, 2019-02-12:
> Hi, @jwangjie
> I use your file, but the ratio may be wrong.
> `400 288`
> this is warning:
> [image]

I set `1280 720` and `400 288` to `752 480`

pisuto, 2019-02-12:
I think this is what we need:
`0.6099122 0.95061667 0.4883178 0.51722791 -0.28340811 0.07395907 0.00019359 1.76187114e-05
752 480
crop
640 480`

glc12125, 2020-08-12:
Hey @pisuto , where did you get the 0.95061667 from? Following the README, we should put fy/height, which is 0.9527, as the second parameter. But 0.9527 gives worse result. Appreciate any insight.