tum-vision/lsd_slam #330: ERROR [gendeps] 1 Finding dependencies for /home/username/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/cfg/LSDSLAMViewerParams.cfg

swiss-knight, 2019-02-18:
I am having this error message while running `rosmake lsd_slam` fro the `~/rosbuild_ws/package_dir` directory where I cloned the lsd_slam repository :

```
$ rosmake lsd_slam
[ rosmake ] rosmake starting...                                                                                                                                
[ rosmake ] Packages requested are: ['lsd_slam']                                                                                                               
[ rosmake ] Logging to directory /home/username/.ros/rosmake/rosmake_output-20190218-162754                                                                     
[ rosmake ] Expanded args ['lsd_slam'] to:
['lsd_slam_core', 'lsd_slam_viewer']                                                                                
[rosmake-0] Starting >>> catkin [ make ]                                                                                                                       
[rosmake-0] Finished <<< catkin ROS_NOBUILD in package catkin
 No Makefile in package catkin                                                                   
[rosmake-0] Starting >>> genmsg [ make ]                                                                                                                       
[rosmake-0] Finished <<< genmsg ROS_NOBUILD in package genmsg
 No Makefile in package genmsg                                                                   
[rosmake-0] Starting >>> genpy [ make ]                                                                                                                        
[rosmake-2] Starting >>> cpp_common [ make ]                                                                                                                   
[rosmake-1] Starting >>> genlisp [ make ]                                                                                                                      
[rosmake-3] Starting >>> gencpp [ make ]                                                                                                                       
[rosmake-0] Finished <<< genpy ROS_NOBUILD in package genpy
 No Makefile in package genpy                                                                      
[rosmake-0] Starting >>> geneus [ make ]                                                                                                                       
[rosmake-1] Finished <<< genlisp ROS_NOBUILD in package genlisp
 No Makefile in package genlisp                                                                
[rosmake-1] Starting >>> gennodejs [ make ]                                                                                                                    
[rosmake-3] Finished <<< gencpp ROS_NOBUILD in package gencpp
 No Makefile in package gencpp                                                                   
[rosmake-3] Starting >>> ros_environment [ make ]                                                                                                              
[rosmake-2] Finished <<< cpp_common ROS_NOBUILD in package cpp_common
 No Makefile in package cpp_common                                                       
[rosmake-0] Finished <<< geneus ROS_NOBUILD in package geneus
 No Makefile in package geneus                                                                   
[rosmake-2] Starting >>> rostime [ make ]                                                                                                                      
[rosmake-0] Starting >>> cmake_modules [ make ]                                                                                                                
[rosmake-1] Finished <<< gennodejs ROS_NOBUILD in package gennodejs
 No Makefile in package gennodejs                                                          
[rosmake-3] Finished <<< ros_environment ROS_NOBUILD in package ros_environment
 No Makefile in package ros_environment                                        
[rosmake-1] Starting >>> message_generation [ make ]                                                                                                           
[rosmake-0] Finished <<< cmake_modules ROS_NOBUILD in package cmake_modules
 No Makefile in package cmake_modules                                              
[rosmake-3] Starting >>> roslang [ make ]                                                                                                                      
[rosmake-0] Starting >>> rospack [ make ]                                                                                                                      
[rosmake-1] Finished <<< message_generation ROS_NOBUILD in package message_generation
 No Makefile in package message_generation                               
[rosmake-2] Finished <<< rostime ROS_NOBUILD in package rostime
 No Makefile in package rostime                                                                
[rosmake-2] Starting >>> roscpp_traits [ make ]                                                                                                                
[rosmake-1] Starting >>> xmlrpcpp [ make ]                                                                                                                     
[rosmake-3] Finished <<< roslang ROS_NOBUILD in package roslang
 No Makefile in package roslang                                                                
[rosmake-3] Starting >>> rosgraph [ make ]                                                                                                                     
[rosmake-1] Finished <<< xmlrpcpp ROS_NOBUILD in package xmlrpcpp
 No Makefile in package xmlrpcpp                                                             
[rosmake-0] Finished <<< rospack ROS_NOBUILD in package rospack
 No Makefile in package rospack                                                                
[rosmake-1] Starting >>> rosclean [ make ]                                                                                                                     
[rosmake-0] Starting >>> roslib [ make ]                                                                                                                       
[rosmake-3] Finished <<< rosgraph ROS_NOBUILD in package rosgraph
 No Makefile in package rosgraph                                                             
[rosmake-2] Finished <<< roscpp_traits ROS_NOBUILD in package roscpp_traits
 No Makefile in package roscpp_traits                                              
[rosmake-3] Starting >>> rosparam [ make ]                                                                                                                     
[rosmake-2] Starting >>> roscpp_serialization [ make ]                                                                                                         
[rosmake-3] Finished <<< rosparam ROS_NOBUILD in package rosparam
 No Makefile in package rosparam                                                             
[rosmake-3] Starting >>> rosmaster [ make ]                                                                                                                    
[rosmake-0] Finished <<< roslib ROS_NOBUILD in package roslib
 No Makefile in package roslib                                                                   
[rosmake-0] Starting >>> rosunit [ make ]                                                                                                                      
[... 110 lines cut ...]
    /opt/ros/melodic/share/dynamic_reconfigure/cmake/cfgbuild.cmake:99 (gencfg_cpp)
    CMakeLists.txt:29 (include)
  
  
  -- [rosbuild] Found SSE3 extensions, using flags: -msse3 -mfpmath=sse
  -- Configuring incomplete, errors occurred!
  See also "/home/username/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build/CMakeFiles/CMakeOutput.log".
  See also "/home/username/rosbuild_ws/package_dir/lsd_slam/lsd_slam_viewer/build/CMakeFiles/CMakeError.log".
  /opt/ros/melodic/share/mk/cmake.mk:7: recipe for target 'all' failed
-------------------------------------------------------------------------------}
[ rosmake ] Output from build of package lsd_slam_viewer written to:
[ rosmake ]    /home/username/.ros/rosmake/rosmake_output-20190218-162754/lsd_slam_viewer/build_output.log
[... cut ...]
[ rosmake ] Results:                                                                                                                                           
[ rosmake ] Built 47 packages with 1 failures.                                                                                                                 
[ rosmake ] Summary output to directory                                                                                                                        
[ rosmake ] /home/username/.ros/rosmake/rosmake_output-20190218-162754
```

How to fix it ?

I'm using ubuntu 18.04 64 bit with ros-melodic. 
Thanks

toonnii, 2019-05-14:
Hello 
Do you fix it?

aayushi363, 2020-07-07:
Facing same issue please help!!

aachamehul021, 2020-11-26:
still looking for an answer

Marc-3d, 2025-07-29:
Hi, I found the solution 🔍 

Edit the file `/lsd_slam/lsd_slam_viewer/cfg/LSDSLAMViewerParams.cfg` and remove all 

> 's

within strings variables. 
<br>

For example, in line 20 of `LSDSLAMViewerParams.cfg`, I changed "point's" to "point s", and "keyframe's" to "keyframe s": 

BEFORE: `gen.add("scaledDepthVarTH", double_t, 0, "log10 of threshold on point's variance, in the respective keyframe's scale. ", -3, -10, 1)`
AFTER: `gen.add("scaledDepthVarTH", double_t, 0, "log10 of threshold on point s variance, in the respective keyframe s scale. ", -3, -10, 1)`

In total, you have to remove `'s` from a handful of lines in `/lsd_slam/lsd_slam_viewer/cfg/LSDSLAMViewerParams.cfg`, and from a handful of lines in `/lsd_slam/lsd_slam_core/cfg/LSDDebugParams.cfg`.